jgyates/genmon #862: add-on  remote access assistance please

jcherney, 2023-02-25:
I am away from my home and both power and wifi are (were) off. I know my generator is running only because my daughter came by and let me know. WiFi came back on just now, but I cant access my system to tell if the power is on or not.

Does anyone have a (relatively) clean solution to access my system (or or other means) remotely (with or without wifi), or somehow tell if the generator is running or not?

skipfire, 2023-02-25:
If you lose internet you won't be able to do anything remotely.  There are options like cellular HATs/USBs or a hotspot that could provide remote access even when your primary internet is out, but without that already in place you are out of luck for this time.  

Many internet providers don't have their intermediate equipment on backups so as soon as the power goes out it is not uncommon to lose internet access, though that is getting better as when internet companies are placing new field infrastructure they are adding some amount of power backup, although my local cable company seems to be using a battery backup system and it lasts right at 2 hours.

You said WiFi is back, what about internet?  Having power in the home is all you need for WiFi, but without internet connected as well you only have local access then.

Do you have a UPS for your router and wifi equipment?  If not I would recommend adding that so you don't have the equipment power off in the 15 seconds before the generator is providing power, and then having to wait a couple of minutes for your connection to hopefully re-establish.

jcherney, 2023-02-25:
Sorry, when I said Wifi, I meant internet. And it is working- I know that because my Genmon sent me an email and text.  Yes, I know I'm out of luck this time. For some reason my internet provider doesn't have any backup and it dies when the power dies. I do have a UPS for the router. The generator provides power to everything when it's running. 

Can you expound on the options for HATs/USB /hotspots? Thanks.

skipfire, 2023-02-25:
I have never used a cellular HAT or USB, but a quick google search came up with an example https://sixfab.com/product/raspberry-pi-base-hat-3g-4g-lte-minipcie-cards/

As for hotspots, that's just putting your phone in hotspot mode or getting a hotspot device as another line from your cell carrier.

jgyates, 2023-02-25:
At plug in for a cellular HAT (add on card for the pi) which allows for SMS to be sent directly from the pi already exist.

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gensms_modempy-optional

skipfire, 2023-02-25:
@jgyates with that solution does the HAT also provide cell data to the Pi?  If so ngrok or similar, or getting really fancy and setting up an auto VPN connection, would allow remote access to the GUI too.

jgyates, 2023-02-25:
No, this only provides SMS. 

It support cellular data, the cell card would need to act as a network adapter, in addition to either dynamic DNS and VPN solution.

jcherney, 2023-02-25:
This looks like an interesting option. I found a $5/mo SMS only plan that would work nicely for this. 

Thank you!

closed 2023-02-25 by jgyates