jgyates/genmon #864: Reset Alarms not present with serial bridge

skipfire, 2023-03-05:
### Expected Behavior

Reset alarm button should be present when directly connected or using serial bridge

### Actual Behavior

Button is present only when directly connected.

### Steps to Reproduce (including precondition)

Get a generator with the reset alarm button working when connected directly, switch to a serial bridge and the button is not present though status and commands work.

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: 22kw Air Cooled Evo2
- Genmon Logs: Submitted 11:17 AM EST
- Genmon Version: 1.18.05

skipfire, 2023-03-05:
More findings.  I had setup a new GenMon install to use with the serial bridge, that's the one that did not have the Reset Alarm button, my previous install that was directly connected did have the button.  Once my serial bridge was stable for a few hours, I swapped my old install to use the bridge instead of direct connection and it still has the Reset Alarm button.  I'm guessing it's related to model detection.

jgyates, 2023-03-06:
A couple of questions:

1) Are you actually using version 1.18.05 or are you using the latest version?
2) Are you showing any communication errors when using serial bridge?
3) If you refresh your browser and wait for genmon to fully initialize do you see the reset button?

The reset alarm button is present when an Evo Liquid Cooled or and Evo2 is detected and no other times. If there is delay in the comms or any communication errors that prevent the software from identifying the controller before the web app ask genmon for the controller type, it will not show up as the software does not know enough at that time to correctly ID the controller. 

The web interface starts up by asking for some initial data from genmon. Part of this data is the controller type. If you restart genmon or start the web interface before genmon has fully initialized the web interface could ask for the controller type before genmon has determined the actual controller. 

Once genmon has finished initializing (it should say "OK" on the monitor page under "Generator Monitor Stats -> Monitor Health " once all data has been read from the controller at last once, otherwise it will state that it is initializing), then refresh your browser. If you still do not see the reset alarms button after this then you likely have comms issues with the serial bridge so look on the Monitor page under communication stats.

closed 2023-03-06 by skipfire