jgyates/genmon #869: tank utility monitor showing wrong level

grsthegreat, 2023-03-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

My tank utility monitor is showing wrong level. when i log into tank utility app on ipad or desktop i get the same reading as the tank dial indicates. but on the genmon page its about 35% higher. 

i stopped the tankutility add on and reinserted all sign on info. it still reads wrong?

grsthegreat, 2023-03-12:
i submitted logs. thanks

jgyates, 2023-03-12:
I would wait 24 to 48 hours and see if the value is accurate. TankUtility reading via their API are delayed so if you have used any fuel in the last two days that would account for this then I would guess that this is the likely reason.

Your logs for tankutiliy add on do not show any recent errors. There were some network connectivity errors in the log but they were from last month.

If waiting does not resolve the issue then send secreen shots of your data (add on and tankutiliy web site) and your tank details.

grsthegreat, 2023-03-12:
Ok, thanks

closed 2023-03-12 by grsthegreat