jgyates/genmon #870: Evolution 1 Control Panel Replacement

dennis48755, 2023-03-16:
Hello,

I've ordered an Evolution 1 Control Panel which I anticipate replacing in the next couple days.

- Generator Model: 64622 16KW
- Firmware: 1.17

How will this affect my years of GenMon data?
Will I lose my service logs?
It will probably come with current firmware which I understand will have the encrypted Modbus data, will this be a problem?
Is there anything else that I should be concerned about?

Thanks for any suggestions or alerts.
Dennis

jgyates, 2023-03-16:
The only thing that should happen is that your serial number will be blank. If you know your old serial number you can enter it on the advanced page, but that is really just informational. The service journal (not the maintenance lon) will remain as that is external to the controller however the Evolution maintenance lon, start stop log and run log will be replaced as that information is held in the controller and genmon just displays the data. Evo1 is not encrypted so no worries there.

I don't know of a way to set the maintenance timers manually so you may have to keep up with that on your own until your maintenance is due, then reset it on the controller as you normally would.

Let me know if you have any questions or issues.

jgyates, 2023-03-16:
no problem. I am going to close this issue but if you have any question feel free to post to this thread.

closed 2023-03-16 by jgyates