jgyates/genmon #873: Help Please: Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:223

basshook, 2023-03-22:
Looks like a fantastic project, thanks. 

I do pass the loop back test at the end of the cable. I am using the pintsize me HAT & cable which connects the pi ground to the controller.

Alias's
lrwxrwxrwx 1 root root 7 Mar 21 18:17 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Mar 21 18:17 /dev/serial1 -> ttyS0

Timeout Percent Errors : 99.24%
Modbus Exceptions : 0

 Environment
- Generator Model: {48kw liquid cooled}
- pi3b with PintSize Me HAT & cable:  https://pintsize.me/store/ols/products/generator-hats-and-phat
- Genmon Logs: Submit via the About page
- Genmon Version: {Version V1.18.16}

jgyates, 2023-03-22:
Have you setup your serial port?

did you follow the steps in this link: https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

skipfire, 2023-03-22:
Did you stop genmon before trying the loopback test?  If not genmon would have the port in use already.

basshook, 2023-03-22:
Yes I ran the serialconfig.py and all items showed "OK". The loopback test was done with the Pi remove and genmon stopped, it passed. The error generated was while genmon was running, not the loopback test.

basshook, 2023-03-22:
Enable UART: Success
Disable BT: Success
Disable serial console: Success
Disable serial console service: Success
Disable BT service service: Success

Still no traffic.......

jgyates, 2023-03-22:
Are you using the cables included with the pintsize.me hardware or are you making your own?

Does the timestamp for the error you mentioned in the myserial.log correspond to you starting genmon and does it repeat if genmon is restarted or is this an older error that occurred once?

Have you restarted the pi since you ran serialconfig.py?

Is your controller Nexus or Evolution? Some Nexus models have some issues powering the pi via the molex connector.

Do you see any Frequency Throttling, Frequency Cap or Undervoltage Errors reported on the Monitor page of genmon in the Platform Stats section?

Has this controller ever had genmon, or mobile link connected before (i.e. is the serial port working on the controller?)

jgyates, 2023-03-22:
Frequency Throttling, Frequency Cap or Undervoltage Errors are raspberry pi errors that help determine if the power supply is working as expected and are independant of serial comms.

Please send photos of the firmware version screen from the controller LCD. If what you are saying is accurate I have never seen this combination before (Evolution 2.0 Liquid cooled firmware V1.11).  To date all Evolution Liquid Cooled models use Evolution 1.0. If you do have a Evo2 Liquid cooled the approach to troubleshooting is very different compared to Evo1.

Regarding power you mentioned: "Using the pintsize.me cables. Power delivered using the POE you referenced in the doc. "

There are multiple sources for PoE adapters for the Pi mentioned in the genmon docs. Send a link on specifically what you are using.

Photos of your pi, power and cable connections may help as there are limited data points to go on.

basshook, 2023-03-22:
I am not sure that the controller is an evo 2 or evo 1, I do know the firmware is 1.11. I will recheck.... Yes, I see the Platform Status you mention (Frequency Throttling/Cap). URL for the POE splitter: https://www.amazon.com/dp/B01MDLUSE7?psc=1&ref=ppx_yo2ov_dt_b_product_details

The pintsize.me HAT is a new version, wondering if there are some issues with that? It did pass the loopback. I try to scope the signal outputs and take a look at the voltage levels.......

liltux, 2023-03-22:
@jgyates it is not EVO2, however the EVO Liquid cooled with the 4.5 is a different controller.

jgyates, 2023-03-22:
The firmware version should be under the dealer menu ->  Display. 

This link describes how to access the dealer menu:

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#access-the-dealer-menu

There was an issue with the 48kW 4.5L units serial comms that has been resolved with a firmware upgrade. The firmware should be version 1.4.39366 or newer, otherwise you will need a firmware update.

basshook, 2023-03-22:
Well I think this could be a common grounding issue. I think I wrongfully assumed that a newer 48k controller would have active power pins. I reconfigured the HAT to pick up power through the cable/controller rather than the POE splitter. Plugged the Pi back into the generator and there was no power to the Pi/HAT. 

Looking at the mobile link wifi cable, I noticed that it was picking up its power from the main battery. I reconnected the wifi (same accessory port) and mobile link works fine. It's raining here, when it stops, I'll connect the battery ground into the Pi.

I did confirm that the controller firmware is 1.11.

skipfire, 2023-03-22:
You say you have a new version of the HAT, which version do you have?

The HAT is built with a common ground and the serial implementation on the HATs has not changed over the last several revisions of the HATs and the basic serial configuration you already confirmed as good with the loopback, though the loopback does not rule out a ground issue and connecting the battery ground to the Pi is a good way to confirm if it is or is not a common ground issue.  If you have a multi-meter you may also want to check the two fuses on the board.  Both fuses are labeled with a 'B' over '300'.  One is on the right side below the large serial chip, the other is on the left between the red GENPWR switch and the 5-pin header.  Every board is tested after assembly to ensure that 12v power, ground, and both serial pins are connected correctly in the Molex, but it is possible that something could have had a bad intermittent connection or that a fuse blows during installation to the generator.  One other thing that could be checked is the Molex cable itself, I buy them off the shelf and I had a single instance (out of a few hundred) where the cable may have been mis-pinned.  If you look at the cable, one end should be a left-right mirror of the other.
Bottom row: Y, O, R, B and B, R, O, Y
Top row: G, R, B, W and W, B, R, G

The Pi not having power when plugged into the generator with just the Molex cable is not surprising, as I understand only the Evo2 controller supports the 12v output which is what the HATs use.  Also, I would appreciate it if you could send me a message through the PintSize.Me contact form so I can associate your order in case this doesn't get worked out and a warranty replacement is needed.

basshook, 2023-03-22:
Hi Jon, thanks for jumping in to help. I have the V2.2 standard HAT. The HAT seems to be working and the cable is wired as you described. The two pins on the cable (generator side) coinside with the same RS-232 pins used for the mobile link wifi; and those 2 pins on the generator side of the cable pass the loopback test. Also the mobile link is working so the accessory port is OK. 

Question, if I have the cable plugged into the HAT as shown in the attached, which color wire can I tap into for my battery ground?

[image]

skipfire, 2023-03-22:
Can you post a picture of the WiFi cable?  The MobileLink WiFi I have seen uses 2 different pins, if it is in fact using RS232 that may explain the problem as RS232 is a 2-device system.  In the picture you posted the green and ?brown? wires (top row, right 2) are the RS232 pins.

Minor thing, I'd suggest flipping the GENPWR and BUCK red switches to the OFF position since you are not powering through the HAT.  If you have a heatsink you may want to add one as well, the inside of the generator gets rather hot in the summer when it isn't running, scalding hot after it's been running for several hours/days, but it cools down while running as that area is in the air intake.

For adding another ground, if you look at the plug on the HAT the ground is on the bottom row, second from the left, which I believe is the orange wire with the orientation you have there, the cable is reversible though so if you turned the cable around  and had a controller with 12v output there you would fry the controller, since there is no 12v power coming out I have no clue what would happen if you reversed it so be sure you never do (maybe tape a label on both ends).  To just test things without modifying cables, you could use a small gauge wire tied to the negative terminal of the battery and plug into the GND on either the 5-pin header or the 3-pin screw terminal.

basshook, 2023-03-22:
Jon, the mobile link and genmon do indeed seem to want the same RS232, see pic. As you say they can't run concurrently and I have not tried to do so (I am a EE). Yeah, I have a fan for board...... Understood about flipping the cable once the GND is connected.
[image]

skipfire, 2023-03-22:
Thanks for the pic, I hadn't seen an RS232 on the WiFi modules, good to know it exists.

If you try connecting the ground to one of the easy 3 and it starts working then I'd say it's either the fuse on the right side being tripped for some reason, maybe a bad joint on the Molex plug that worked well enough for testing but gave up after that, or the controller maybe has that ground pin floating.  On other controllers there are 2 ground pins with one on the far right (for the HAT end) being floating and the 2nd pin being always ground.

basshook, 2023-03-22:
It's kind of interesting that the wifi cable runs both the supply and GND back to the battery. I am thinking that the pin inside the controller you are picking up for ground is floating.

basshook, 2023-03-23:
Tried connecting the GND to the battery as discussed, no change, no communication. Perhaps its as @jgyates said, an updated firmware to 1.4.39366 is required. Anyway putting this to bed for now, kinda disappointed. Thanks to those that tried to help.

basshook, 2023-03-25:
It's bugging me that I can't get this project to work. I was looking at a recent update to the generac site on firmware rteleases and it states the v1.11 is the latest version for my model. Jon, has this V2.2 HAT been used by anyone else yet? Any possibility that RX & TX are swapped? Thanks!

skipfire, 2023-03-25:
@basshook the serial circuitry has not changed since the beginning, every unit is tested with a unidirectional tester, and you got one of the last v2.2 boards, so no they can't be swapped.  A board with a defect does happen on rare occasions though.

basshook, 2023-03-25:
Just a thought since the swap wouldn't necessarily show up on the loop test...... So, we are connected to pins 7 & 8 on the controller, the same pins that the EVO1 uses for RS232 (no RS485) wifi, the cable passes loop test, the grounds are common. What else is there to check? @jgyates is there any way to check if the Pi/HAT is trying to signal the controller?  Suggestions?

skipfire, 2023-03-25:
@basshook your title says "Attempting to use a port that is not open", but I've not seen anything about that error in the rest of these.  Where was that error, and is it still happening?  If the port is failing to open that would also explain all the problems.

jgyates, 2023-03-25:
If the problem was trying to use a port that is not open, then you would get that specific error in the log every time genmon stared, and from what you describe that does not appear to be the case (you only got that error once early on). From what you describe so far the issue sounds like the pi is attempting to transmit but the controller does not respond. It could be a cable / connector issue (bad pin on the controller?), bad RS-232 interface on the controller, bad ground connection (pi RS232 and controller RS232 are not on a common ground).

If you outbound email is configured you can send me your logs via the About page and I can look at them to see if anything jumps out. 

There is not any indication from the controller that it is attempting to communicate with the pi other than a return packet.

basshook, 2023-03-26:
Jon, there is at least one issue with the V2.2 HAT and possibly two.
First the RX/TX does indeed appear to be reversed as I previously suggested. I scoped the outputs of both the Pi TXD and the HAT TXD (see below), pin 7 & 8 are swapped on the board Molex. I reversed the connections on the cable, and now get lots of CRC errors whereas before I had no communication.

The bench setup shown below has channel 1 on Pi header pin 8 (TXD), channel 2 is on the HAT Molex pin 7 (which should be RXD). I get no output on Molex pin 8...... I am running Genmon triggering on the Pi TXD (Channel 1) and the output waveform looks pretty good. Channel 2 (RS232 HAT) follows Pi inverted as expected, but the logic "1" level is barely -5V and noisy. I suspect that's what is causing the CRC errors? I am not sure if the POE splitter voltage is in part causing the problem. Jon, rather than me continuing to try and debug this, can you send me one of your known good older version boards?

[image]
[image]

skipfire, 2023-03-27:
@basshook the board is not backwards and is known to be good (tested in my own generator as I do every single version and on my test bench that is directional), it's more likely the cable was misassembled as none of the other users of v2.2 have reported such problems.  Can you add a picture of both ends of the cable so I can see how it is now wired?

If you send a message through the contact form on the PintSize.Me website so I know who you are (and can have the log in my system) I can ship out one of the older boards.

skipfire, 2023-03-27:
Also it looks like you are connecting to both sides of the ttl chip so 5v is about what I would expect since you are comparing the 8v external signal to the 3.3v internal signal.  Were you intending to reference ground?

basshook, 2023-03-27:
Jon, I agree, to take this discussion directly. I will send you an email. However, just to answer your questions. If you take a look at probe you will see that it is on pin 7 of the board molex, there is no TXD signal on pin 8. I get the same results on the cable. Probe 2 ground is the on the GND mini connector.

skipfire, 2023-03-30:
In case anyone reads this later, basshook tested with another HAT and got the same results, it appears to not be an issue with the HAT, possibly controller or firmware version.

basshook, 2023-04-06:
Jason, I am still trying to get genmon to work on my generator. I sent you the logs & registers, would appreciate it if you could take a look. I have done the following for reference: checked cable multiple times (tried two different cables), tested with two different HAT's (Jon was kind enough to send me an older version), performed wrap test sucesssfully on both HATs, replaced SD card, reloaded all software, succesfully ran loop test at cable ends for both HATs, checked the serial logs (all I get is time outs & one non repeatable failure to open port message, ensured the Pi/HAT was grounded to the batt and then the chassis (the cable does have gnd to acces port as well). The only errors I ever get are "no communication" errors. Mobile Link wifi works without issue

Jon, thanks again for the test HAT, I will return soon. Jason, thanks for taking a look.....

-Ron

jgyates, 2023-04-06:
Sure, can you send the logs again? Once a few weeks pass it is hard to tell what logs belong to which thread.

jgyates, 2023-04-06:
I have not received any today. Is your outbound email working? If it is you should get an email when genmon starts (or restarts). 

Alternatively you can download your logs locally to your computer by entering this command in your browser:

  http://192.168.1.10:8000/cmd/get_logs

The assumes your pi's IP address is 192.168.1.10 and your are not using HTTPS. This will download a file named genmon_logs.tar.gz and you can email that file to generatormonitor.software@gmail.com

basshook, 2023-04-06:
I sent the logs again as well as attaching to an email. I do get genmon email notifications.

Thanks.

jgyates, 2023-04-06:
I received your logs. 

When you run the loopback test with genmon disabled, are you running it on /dev/serial0 or /dev/serial1?

Here is the contents of your /var/log/myserial.log file:

2023-04-05 15:47:26,969 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:223
2023-04-06 09:20:08,031 : Error on open serial port /dev/serial1: Could not configure port: (5, 'Input/output error')
2023-04-06 09:20:08,032 : Error in init: Error on open serial port /dev/serial1: Could not configure port: (5, 'Input/output error') : myserial.py:125
2023-04-06 09:20:08,032 : Error on serial port init!
2023-04-06 09:24:17,661 : Error on open serial port /dev/serial1: Could not configure port: (5, 'Input/output error')
2023-04-06 09:24:17,661 : Error in init: Error on open serial port /dev/serial1: Could not configure port: (5, 'Input/output error') : myserial.py:125
2023-04-06 09:24:17,662 : Error on serial port init!

This shows that both /dev/serail0 and /dev/serial1 have been used. You system should map /dev/ttyAMA0 to /dev/serial0 and according to your first entry in this log it looks like it is, but just make sure that your genmon settings use the serial port /dev/serial0 not /dev/serial1

Ideally you should not get any errors in your myserial.log file when you start genmon or while it is running. If you are transmitting, but your controller is not responding then your should get timeout errors (which would be logged in /var/log/mymodbus.log). Your log does contain these timeout errors, but you need to make sure you are using the correct port also. The title of this thread indicates that you were using /dev/serial0 at one point, so I am guessing that you changed in at some point for testing / troubleshooting.

basshook, 2023-04-06:
Yes, I am using serial0 and that's what the loop test was running. I switched the settings to serial1, to see if it had any effect, it did not. Mapping is correct.....
lrwxrwxrwx 1 root root 7 Apr  6 12:39 /dev/serial0 -> ttyAMA0
Yes, lots of timeout errors within log mymodbus.log with serial0 port.

It's as if the cable was open, but it's not, it is connected to pins 7 & 8 on the gen controller (same as mobile link), and the loop test works.
Does the controller need an identifier or specific handshaking to begin communication, or can you just send it commands and queries? I would expect to get some communication or ack back?

jgyates, 2023-04-06:
on the Evolution controllers the modbus address (ID) is always the same (hex 9d). On the Evolution 1 controller you only get responses back for valid packets (Evo2 and Industrial controllers send modbus exception packets on errors). 

It sounds like you have tried two different RS-232 interfaces. It is possible that you have a bad driver or receiver on your controller. 

Have you ever seen any data sent back with your o-scope? You should see something come back.

Has this ever worked with MobileLink?  That is the only other thing that I know that will communicate with the controller.

basshook, 2023-04-07:
Yes it works fine with Moble Link, I was using it today, before I reconnected the Pi, which is why I ruled out the bad driver/receiver in the controller scenario. I suppose I could scope the Molex pins while connected to the controller. I suspect I'll see the HAT outputs toggle, as I do on the bench, and no waveforms back from the controller. 

You mentioned on Evo1 (which I believe is my controller) the controller only responds back for valid packets. What constitutes a valid packet? Is it possible that generac made some mods to the packet ID?

jgyates, 2023-04-07:
I guess it is possible. You can test with with the modbusdump.py utility in the /genmon/OtherApps folder. This is the syntax:

To display all the command line options type:

    python3 modbusdump.py -?

This should get you some data on an Evo1 (assuming the default modbus address of 9d), this command would dump modbus register 0005:

    sudo python3 modbusdump.py -r 9600 -p /dev/serial0 -a 9d -s 5 -e 6

This command should probe all addresses:

    sudo python3 modbusdump.py -r 9600 -p /dev/serial0 -a all -s 5 -e 6

That may take a bit of time as each transaction would timeout but you can let it run. 

You can save the output to a file named output.txt (it will not display on the screen if you do this) with this command:

    sudo python3 modbusdump.py -r 9600 -p /dev/serial0 -a all -s 5 -e 6 > output.txt

If you specifiyou could change the 9d to any value between 00 and ff hex but that is 256 different values. You could also play around with the data rate

basshook, 2023-04-07:
The commnad to dump reg 5, modbus address 9d, timed out. Probing all addresses now...... will send when complete.

basshook, 2023-04-07:
Dang, it timed out for all 256 addresses. Any other suggestions?

jgyates, 2023-04-07:
At this point I would be curious to see the o-scope output from the controller. You can use modbusdump.py to send a packet and see if there is any movement on the Rx lines from the controller.

Another quick thing to try is instead of 9600 try each of these values: 110, 300, 600, 1200, 2400, 4800, 9600, 14400, 19200, 38400, 57600, 115200. This is a long shot but easy enough to try.

basshook, 2023-04-07:
Well, well. I got a modbus response with a baud rate of 115200.

kate@raspberrypi:~/genmon/OtherApps $ sudo python3 modbusdump.py -r 115200 -p /dev/serial0 -a 9d -s 5 -e 6
Baud Rate : 115200
Port is : /dev/serial0
Address is : 9d
Start Register : 5
End Register : 6
0005:0b13

basshook, 2023-04-07:
Jason, where can I make the baud rate edit?

jgyates, 2023-04-07:
Let's validate that data (0b13). What time is your exercise set to occur. This would indicate that it is 11:13 AM. Is that correct?

Also look in the OtherApps folder for any log files and show their contents, specifically myserial.log and/or mymodbus.log

also try this command with the controller in auto mode:

sudo python3 modbusdump.py -r 115200 -p /dev/serial0 -a 9d -s 0 -e 2

and try it again in the off position.

Currently you can not change the rate for Evolution as that was never a thing that I thought was needed. I can add that but let's validate a few data points.

basshook, 2023-04-07:
Yes, that is the exercise time. I do not have myserial.log in OtherApps but I do have the mymodbus.log, it's full of time out errors (perhaps from prior polling all addresses?). Not sure how to put the controller in auto/off mode, but just running the command shows:

kate@raspberrypi:~/genmon/OtherApps $ sudo python3 modbusdump.py -r 115200 -p /dev/serial0 -a 9d -s 0        -e 2
Baud Rate : 115200
Port is : /dev/serial0
Address is : 9d
Start Register : 0
End Register : 2
0000:0016
0001:0000
Program Complete

jgyates, 2023-04-07:
I am sorry I gave you the wrong data for the "s" and "e" parameter, run this test next:

 sudo python3 modbusdump.py -r 115200 -p /dev/serial0 -a 9d -s 1 -e 3

Run this test with the button on the controller set to Auto, then run it again with the controller set to off and send me the results.

basshook, 2023-04-07:
Right now the controller is in Auto, if you really want me to go outside to turn to Off, I will :-)

kate@raspberrypi:~/genmon/OtherApps $ kate@raspberrypi:~/genmon/OtherApps $ sudo python3 modbusdump.py -r 115200 -p /dev/serial0 -a 9d -s 1 -e 3
Baud Rate : 115200
Port is : /dev/serial0
Address is : 9d
Start Register : 1
End Register : 3
0001:0000
0002:0000
Program Complete

jgyates, 2023-04-07:
lol, OK, the data looks good so far. I will check something in shortly for you to test.

basshook, 2023-04-07:
Thanks! Just FYI this controller is about 9 months old.....

jgyates, 2023-04-07:
add this entry in your /etc/genmon/genmon.conf file

serial_rate = 115200

then update the software from the About page. Let me know how it goes. If it works then send me your logs on the About page.

basshook, 2023-04-07:
added the serial_rate = 115200 to the bottom of the conf file and made the update to 18.17. Did a full reboot......
Still not receiving data...... all time outs

jgyates, 2023-04-07:
Send your log files.

jgyates, 2023-04-07:
you can just send them on the about page. THe last time I got both sets (downloaded file and via the about page).

basshook, 2023-04-07:
sent

jgyates, 2023-04-07:
based on your logs I don't think you got the full update. I updated to V1.18.17 earlier today. You got that update. your serial stats do not include the new parameter I added to the very last update. Try updating again and see if there is any difference. 

On the Monitor page on the web interface under serial stats, you should see someting like this:

Communication Stats :
    Packet Count : M: 233, S: 0
    CRC Errors : 0
    CRC Percent Errors : 0.00%
    Timeout Errors : 232
    Timeout Percent Errors : 99.57%
    Modbus Exceptions : 0
    Validation Errors : 0
    Sync Errors : 0
    Invalid Data : 0
    Discarded Bytes : 0
    Comm Restarts : 0
    Packets Per Second : 0.33
    Modbus Transport: Serial
   Serial Data Rate: 115200

The last 2 entries are new.

jgyates, 2023-04-07:
sometimes github is slow rolling out the update to all regions.

basshook, 2023-04-07:
looks like it hasn't come across yet, still version 17, updated and no Date Rate shown.

Time for bed anyway, will try in the morning. Thanks again...... TTUL

jgyates, 2023-04-07:
yeah, it is strange. I have two systems with genmon (one connected to the generator and one I use for simulations / testing) one has the update, the other one does not. Let me know how it goes in the AM.

basshook, 2023-04-07:
Yes! I am getting data. No timeouts, no CRC errors, no modbus errors. I haven't had a chance to fully review , the controller type is ID'd as Air Cooled, should be liquid cooled. Sent you the logs.

Monitor :

    Generator Monitor Stats :
        Monitor Health : OK
        Controller : Evolution, Air Cooled
        Run time : Generator Monitor running for 0:05:37.
        Power log file size : 0.00 MB of 15.00 MB
        Generator Monitor Version : V1.18.17
        Update Available : No

    Communication Stats :
        Packet Count : M: 31454, S: 31453
        CRC Errors : 0
        CRC Percent Errors : 0.00%
        Timeout Errors : 0
        Timeout Percent Errors : 0.00%
        Modbus Exceptions : 0
        Validation Errors : 0
        Sync Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 186.54
        Average Transaction Time : 0.0103 sec
        Modbus Transport : Serial
        Serial Data Rate : 115200

    Platform Stats :
        CPU Temperature : 116.96 F
        Pi Model : Raspberry Pi 3 Model B Plus Rev 1.3
        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK
        CPU Utilization : 0.68%
        OS Name : Raspbian GNU/Linux
        OS Version : 11 (bullseye)
        System Uptime : 7:59:00
        Network Interface Used : eth0
        System Time : Friday April 7, 2023 01:50:15

basshook, 2023-04-07:
A few other noticed items:
- Controller/generator type (mentioned above)
- Frequency reported as 600MHz (running)
- RPM at 73, after engine stopped (running is accurate at 1800)
- Startup delay varies widely 1s - 1000's

jgyates, 2023-04-07:
OK, we have the comms working. Since the 4.5L models have a completely different firmware build I will have to address each of these in the code. 

From the about page, send your logs when the generator is in auto mode, and then send them again when the generator is running. Hopefully this will be enough data for me to correct these issues.

basshook, 2023-04-07:
OK, you should have or will shortly, both sets. The first set is in Auto mode, second set is with generator running.

[22 more replies]