jgyates/genmon #874: Finding run hours from Evo2 past registers?

skipfire, 2023-03-28:
I had my controller fried last year and I was hoping to recover an approximate of run hours based on logs I sent in last year.  I found the sent mail, but what registers do I need to look at and how do I decode them?  The emails themselves don't directly contain the 3 different run hours listed on the maintenance screen.

closed 2023-03-28 by jgyates

skipfire, 2023-03-28:
Thanks, found c2 so 194 hours and I can guess usage between the date of that log and the replacement.

jgyates, 2023-03-28:
and you can use the "Additional Run Hours" parameter on the advanced page to fill in the hours from your old controller so genmon will display the total hours on your generator accurately. You probably knew this but wanted to add it for the historical record.

skipfire, 2023-03-28:
Thanks, I had not seen that but thought it might be hidden on the advanced page.