jgyates/genmon #876: Genmon will not update to the latest version (1.18.17)

mxplea10, 2023-04-08:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Update to V1.18.17

### Actual Behavior

Genmon acts like it's updates but still states that V1.18.17 is available for update.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac Guardian 22kW
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.16

[image]

jgyates, 2023-04-08:
I would try to log into the pi via ssh and update from the console. This will show any errors that are occurring. 

Once you log in via ssh then type this:

   cd genmon
    ./startgenmon.sh stop
    ./genmonmain.sh -u
    ./startgenmon.sh restart

then the "y" when prompted to update. Post your output here.

mxplea10, 2023-04-08:
Thanks for the quick responsive.  It seems to have updated.  The output is below.

pi@GENmon:~ $ cd genmon
pi@GENmon:~/genmon $ ./startgenmon.sh stop
Stopping genmon python scripts
Stopping....
Stopping gensms.py
Stopping genserv.py
Stopping genmon.py

pi@GENmon:~/genmon $ ./genmonmaint.sh -u
This script will update genmon to the latest version from the github repository. This script requires internet access. If you have modified any files in the genmon directory, they will be overwritten. Configuration files in the configuration directory will not be overritten.  Continue? (y/n)  y
Updating genmon...
remote: Enumerating objects: 220, done.
remote: Counting objects: 100% (220/220), done.
remote: Compressing objects: 100% (115/115), done.
remote: Total 220 (delta 143), reused 180 (delta 105), pack-reused 0
Receiving objects: 100% (220/220), 693.11 KiB | 2.41 MiB/s, done.
Resolving deltas: 100% (143/143), completed with 27 local objects.
From https://github.com/jgyates/genmon
   89d245f..3295cf8  master     -> origin/master
Checking out files: 100% (250/250), done.
HEAD is now at 3295cf8 V1.18.17
Done.

pi@GENmon:~/genmon $ ./startgenmon.sh restart
Restarting genmon python scripts
Warning: required library voipms not installed. Attempting to install....
Copying gensms_voip.conf to /etc/genmon/
Stopping....
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
Starting /home/pi/genmon/addon/gensms.py
pi@GENmon:~/genmon $

jgyates, 2023-04-08:
Not a log to go on as all of the output looks good. The web interface calls this command:

  ./genmonmaint.sh -u -n -p 3

-u is update
-n is no prompt for confirmation
-p 3 is use python 3 (which is the default and according to your logs you are using python3)

I am going to close this issue as I don't really know what to do about this at this point, but if your run into it again let me know and we can go from there.

closed 2023-04-08 by jgyates

mxplea10, 2023-04-08:
Ok, thanks again.

skipfire, 2023-04-09:
I also ran into this, multiple attempts through the UI resulted in no update, errors, or logs.  I'm up to date now, no details to add other than it wasn't just mxplea10.

jgyates, 2023-04-09:
a few things to note, when genmon reloads a few things occur.

The program genloader.py is started, and this stops or kills any existing versions (and add ons) of the software that are running and then restarts the new versions.  Before genloader starts the new versions it does a check to make sure that all required libraries are installed. If they are not installed, it will attempt to install them. If new add ons have been added in the release, the the upgrade will install any required libraries before it attempts to restart the new. Once this is complete, it restarts genmon, genserv and any add ons enabled.

The web UI has a fixed timeout as it waits for the restart to occur. If the above does not occur in the timeout, the web interface will not be available until the process is complete.

The update is also very dependent on internet access and speed of the servers that host the raspberry pi repositories (apt), python libraries (pip) github github server access. Failure to access github during an upgrade repositories will result in a failed update (i.e. it will still use the old version of the software).

skipfire, 2023-04-09:
The big thing I noticed was it was absurdly quick after pressing the update, it immediately jumped to restarting and didn't seem to be waiting long enough to hit any timeouts.  I had also checked git status and it had not fetched or pulled.  As for internet my QOS monitor shows no instability today.

GlennGoddard, 2023-04-09:
Same issue, noticed yesterday when I went to update; same for an attempt today.  Also updated and rebooted pi with no change from 1.18.16 to 1.18.17.

Updated via script above and it states "1.1.17" on CLI but still "1.18.17" even after forced refresh.
Rebooted pi after CLI update and no web page but CLI is accessible.  CLI still indicates 1.18.17
No MQTT output.
Rebooted again and webpage updated to current 1.18.17 and MQTT output active

jgyates, 2023-04-09:
another data point:

The software checks the file ./genmon/genmonlib/program_defaults.py and looks at the variable "GENMON_VERSION"  for the current version it is running. The javascript checks this file online:

https://raw.githubusercontent.com/jgyates/genmon/master/genmonlib/program_defaults.py

for the same variable.

If the version pulled from the genmonmaint.sh script using the git command show in the script here:

 https://github.com/jgyates/genmon/blob/3295cf8144ca2f5486aa177de59e79e855420617/genmonmaint.sh#L326

is not in sync with the file pulled from the https://raw.githubusercontent.com/ link above then this could cause the behavior. I would need to think of a way to validate that is the case to prove that is what is happening but at the moment that is the best description I can come up with with little evidence to validate that theory.

aa777888-2, 2023-04-13:
I also struggled with this. Updating via the web interface failed repeatedly as described above.

Then I SSH'd in and performed apt update and apt upgrade. Web interface updated again failed. 

Then I searched the issues list for this issue and found it.

Performing the CLI procedure above worked.

So something is broken somewhere.

jgyates, 2023-04-13:
Can someone post the details of their browser and OS version that exhibited these issues so I can test with that version?

Also, a test that anyone can perform is this:

1) edit the version number so genmon thinks and update is available:
      
      cd genmon
      sudo nano ./genmonlib/program_defaults.py 
     
then change the line with "GENMON_VERSION = "V1.18.17" to "GENMON_VERSION = "V1.18.16", then save the file (Ctrl+x, then 'Y', then Enter)

2) restart genmon

      ./startgenmon.sh restart

3) then open the web interface (or refresh your browsers) and go the About page. If you edited the version number in step one, then you should see an update available. Click the "Upgrade to the latest version" button. The software should update and restart and your version number on the about page should change back from V1.18.16 to V1.18.17. 

If this fails, then it is likely an issue with a specific browser version. If it passes then I would guess it was an issue with github (described above) or something specific to the code in the last update (or the update before this one), but I am struggling to find any changes that would effect that in recent updates.

when I try the above steps it works for me so if someone wants to test, if they get a failure we can drill down from there.

skipfire, 2023-04-13:
I tried 2 browsers, both on Windows 11 22H2 22624.1546
Chrome 111.0.5563.149
Edge 112.0.1722.39

jgyates, 2023-04-13:
If you get a chance try the above test.

FYI, Chrome Version 112.0.5615.49 on a M1 Mac 13.3.1 (22E261) works for me. I also tried Windows 10 with the latest Chrome and Edge version with the test above and it works for me. I will have to wait to test Windows 11 as that system is being used at the moment and I don't have a windows 11 VM setup.

skipfire, 2023-04-13:
I tried that and it worked, but I don't think that indicates success because my local repo would already be up to date so it could do the reset from the local.  When it wasn't updating the local repo was still behind so while the reset part worked, it hadn't fetched or pulled before the reset so it didn't have the update.

jgyates, 2023-04-13:
I don't think that tells us what happened either, however it does show that there is not a browser issue.

Another thing to add to the list of data is that typically I check in changes / bug fixes as they occur but wait to update the version number after a period stability so if there was an issue with github delaying the updates then it would likely be just the version number that is not being updated. e.g. most updates in V1.18.17 were checked in weeks ago, the version number is what has chanced recently.

Another thing to note is that this update included on library install for the VoIP add on. That does require extra time and a separate repository (pip, in addition to github, and apt).

Another thing to look at if this occurs again is the on the Monitor page look at "Run time : Generator Monitor running for "... If the update to the local repository occurred but the restart to genmon did not occur then the Run time of genmon would be longer.

aa777888-2, 2023-04-14:
> Can someone post the details of their browser and OS version that exhibited these issues so I can test with that version? Also, a test that anyone can perform is this...

Windows 10 Pro 22H2 19045.2846

Brave Browser 1.50.118 (in a clean window with no extensions)

Had the same problem as others as documented above, and fixed using the CLI method as documented above.

Ran the test as requested and that worked. However, as already noted by others the download was pro forma as the files already existed on the Pi.

It may be worth noting that the user experience in the browser in all cases was identical. A progress bar for downloading followed by the same restarting indication. It sure doesn't seem like a browser issue.

jgyates, 2023-04-14:
At this point it don't think it was a browser issue either. The leading candidate is that this is an anomaly of the current design. The update works like this:

javascript calls python which calls a bash script to to the actual work of updating an restarting. A python script cannot kill and restart itself without the help of another process (in this case the bash script). The javascript has fixed timeouts so the progress bars for updating and restarting are just based on timers and not actually tied to the updating and restarting process. Once the timeout occurs the javascript will refresh the browser page.

In most cases these timeouts work fine, however if there is an update that takes longer to perform then the behavior everyone has described will occur and the update will not appear to be applied until the browser is refreshed. If this is what is happening the update and restart will eventually complete (unless there is a network or server issue), however an additional browser refresh will need to be done.

I will review the update design and try to determine if there are improvements that can be easily made.

aa777888-2, 2023-04-14:
Makes total sense. Open loop, non data or event driven design is always a bit risky.

jgyates, 2023-07-05:
FYI:   An update has been posted that may resolve this upgrade issue in the future. [See this thread for more info](https://github.com/jgyates/genmon/issues/920).