jgyates/genmon #877: Mopeka sensors

bbrown-sc, 2023-04-08:
Sorry if this is off topic. My WiFi Tankutility monitor (integrated with genmon) bit the dust right outside of warranty. No help from Tankutility or Generac. They don't make the WiFi version anymore and I don't have cellular service in my area for the LTE version. I'm looking at the Mopeka sensors. They make a Mopeka Pro Check Sensor and a Mopeka Pro Plus Sensor-Residential (Long Range Bluetooth) versions. My tank is ~60 feet from the shed that houses the genmon pi4. Mopeka's FAQ says the BT range is 300ft but I'm not sure if that's for the long range version or the regular version. Does genmon support both versions? Which one do I need. Thanks.

jgyates, 2023-04-08:
I don't know as I have never tested wit the Long Rang BT version. Also 300 ft is  clear line of sight. Walls and other obstructions will reduce that length.

The mopeka support is based on this python library:

https://github.com/spbrogan/mopeka_pro_check/issues/5

and according to this issue it does not look like the Pro Plus is supported without modifying their code.

rwskinner, 2023-04-08:
I can easily pickup the Mopeka Pro and Mopeka standard sensors from over 
150' away.  I can actually go out and test it for you if you like.
I'm not using the Pro Plus.

bbrown-sc, 2023-04-08:
Thanks @rwskinner @jgyates. I'm going to purchase the Mopeka Pro Check Sensor and hope for the best.

closed 2023-04-08 by bbrown-sc

rwskinner, 2023-04-09:
At 250 feet still updating fine.   Sent from my iPhone

rwskinner, 2023-04-09:
I was able to do 250' feet with the Mopeka Pro (line of sight), with no 
problem at all.  I had food on the BBQ so I didn't go any further but 
the signal was still strong.