jgyates/genmon #879: Mopeka stopped working?

rwskinner, 2023-04-12:
Mopeka was working fine until yesterday.  I put it in debug mode and then pulled a log.
It seems like it quit working after the Broken pipe error.
The signal from the sensor is fine and I'm collecting valid data from the tank app.

Can I tell if bluetooth is still working?

023-04-11 17:25:02,990 : starting scan
2023-04-11 17:25:17,991 : stopped scan
2023-04-11 17:25:17,991 : Tank Level in mm: 500
2023-04-11 17:25:17,991 : Tank Level in inches: 19.69
2023-04-11 17:25:17,992 : Gallons Left: 176.48
2023-04-11 17:25:17,992 : Tank Percentage: 70.33
2023-04-11 17:25:17,992 : Tank1 = 70.33
2023-04-11 17:25:17,993 : Tank Data = {"Percentage": 70.33, "Tank Name": "Mopeka Sensor Tank"}
2023-04-11 17:25:17,994 : OK
2023-04-11 17:26:17,994 : starting scan
2023-04-11 17:26:32,995 : stopped scan
2023-04-11 17:26:32,996 : Tank Level in mm: 500
2023-04-11 17:26:32,999 : Tank Level in inches: 19.69
2023-04-11 17:26:32,999 : Gallons Left: 176.48
2023-04-11 17:26:33,000 : Tank Percentage: 70.33
2023-04-11 17:26:33,000 : Tank1 = 70.33
2023-04-11 17:26:33,000 : Tank Data = {"Percentage": 70.33, "Tank Name": "Mopeka Sensor Tank"}
2023-04-11 17:26:33,001 : OK
2023-04-11 18:45:17,644 : Error in TankCheckThread: [Errno 32] Broken pipe : genmopeka.py:540
2023-04-11 18:46:17,644 : Error in TankCheckThread: [Errno 32] Broken pipe : genmopeka.py:540
2023-04-12 09:14:37,054 : Tank Address = D2:C8:88:19:89:66
2023-04-12 09:14:37,057 : min: 38.1 , max: None
2023-04-12 09:14:37,058 : Tank Type: 250_GAL
2023-04-12 09:14:37,058 : Tank Volume Calculated: 250.92
2023-04-12 09:14:37,058 : <Horizontal tank, V=57962.384459 m^3, D=30.000000 m, L=62.000000 m, spherical heads, a=15.000000 m.>
2023-04-12 09:14:38,060 : starting scan
2023-04-12 09:14:53,220 : stopped scan
2023-04-12 09:14:53,222 : No sensor comms detected.
2023-04-12 09:14:53,222 : No reading from tank sensor!
2023-04-12 09:15:53,223 : starting scan
2023-04-12 09:16:08,224 : stopped scan
2023-04-12 09:16:08,226 : No sensor comms detected.
2023-04-12 09:16:08,226 : No reading from tank sensor!
2023-04-12 09:17:08,227 : starting scan
2023-04-12 09:17:23,228 : stopped scan
2023-04-12 09:17:23,230 : No sensor comms detected.
2023-04-12 09:17:23,230 : No reading from tank sensor!

jgyates, 2023-04-12:
I would try restarting your pi. If bluetooth is the problem then this would be the best recommendation. If that does not work then I would update your pi's system software with the commands detailed here:

  https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#updating-your-pi-system-software

once this is complete I would restart your pi. Let me know how it goes.

rwskinner, 2023-04-12:
Thanks.  I had already tried the reboot.  I'll update the system and see 
how that goes.

rwskinner, 2023-04-12:
I remember seeing a document talking about the steps required to setup 
the bluetooth and mopeka sensor but I searched the wiki and never found it.
Would you kindly point me in the right direction?

jgyates, 2023-04-12:
https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmopekapy-optional

rwskinner, 2023-04-12:
I have thousands of entries in the syslog like this....  7 entries per 
second.

Apr 11 23:51:57 genmon dhcpcd[458]: wlan0: received approval for 
192.168.0.123

jgyates, 2023-04-12:
The name "genmon" in the line is due to your system name being "genmon". The genmon software for this project did not generate this message. You can see from this thread that these messages are from the dhcpcd module on your systems. This thread also has a possible solution:  https://forums.raspberrypi.com/viewtopic.php?t=265516

jgyates, 2023-04-16:
I am going to close this issue but feel free to post this this thread with follow up status or questions.

closed 2023-04-16 by jgyates