jgyates/genmon #881: MQTT Topic For CPU Temp?

basshook, 2023-04-13:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior
Maybe I missed it but is there an MQTT topic (I think that's what they are called) for CPU temp in value form? I see there is a topic for CPU temp in text. Not too familiar with MQTT but I have several values and status texts coupled to Homeseer, works great!

### Actual Behavior

Maybe I missed it{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2023-04-13:
There is not. currently the MQTT values returned on the Monitor page are all in string form. Using the "JSON for Numerics" option only apples to the data on the Status, Outage (if your controller supports reading the utility voltage) and Maintenance pages.

basshook, 2023-04-13:
Thanks for the reply.

closed 2023-04-14 by jgyates