jgyates/genmon #882: Can't connect to genmon after battery replacement

jcherney, 2023-04-15:
Hi all,

Had to put a new battery in my generator tonight (got an alarm the other day and again today). After doing so, I can't connect to the genmon IP that I'm using thru a web browser.. My router shows it as being connected. The mac address is correct. I can ping it from a command line with no issue. 

I just rebooted via the putty connection, still nothing.

Error is: connection refused

Any thoughts on what might be the issue?

Thanks,
Jack

jcherney, 2023-04-15:
I get emails stating the reboot has happened with all the correct data.

jgyates, 2023-04-15:
For clarity, the emails you are getting are from genmon and they have the IP address in the body and that IP address is the same IP address that you are putting the your browser? Does the IP address your router shows match the IP address in the email and your browser?

When genmon starts there are two emails sent. The first has the IP address of the PI (it is sent early in the genmon loading process) and the second is sent once genmon has read the registers, has the generator initial status. The most common issues is that your IP address of the pi has changed. Make sure the IP address in the first email that genmon sends contains the same IP address that you are using in your browser to access the web interface. If you saved a address in your browser favorites and your router changed the IP address assigned to your PI then you would need to update the bookmark.

More details, like the contents of the email (specifically the IP address in the message body) and the exact address you are using in your browser may help. 

Also, the browser is what is giving you the error that the connection is refused, correct?

There are other issues from the past that may be useful that are tagged "Router" that dealy with people not being able to access the UI but genmon is working. They are here: https://github.com/jgyates/genmon/issues?q=is%3Aissue+label%3ARouter

jcherney, 2023-04-16:
The first email - generator monitor starting- states the correct IP address. The second email has no IP address at all. Also, I checked the router and it shows the port forward to :8000    One of the suggestions in the link said to run sudo netstat -tulpn | grep LISTEN. The third response has no 0.0.0.0:8000

jcherney, 2023-04-16:
Running thru your link suggestions, I just ran cd genmon
  ./startgenmon.sh restart   Waited about a minute and then the webpage fired up! 
Is there something I need to fix to make this happen consistently?

jgyates, 2023-04-16:
I don't have enough information to know exactly what occurred. You can send your logs from the about page and I can look to see if there are any obvious issues in the logs but other than that I don't know what happened.

closed 2023-04-16 by jgyates

jcherney, 2023-04-16:
Just sent them- thanks for looking.  BTW- I did an update you mentioned in that link- I took a LONG time but finished. Not sure if that had any effect at all.

jgyates, 2023-04-16:
Got them. Nothing unusual in the logs. If you have any more data I would be happy to look at it but at least it is working for you now.

jcherney, 2023-04-16:
Thanks for checking that out.  Is there a reason that starting genmon
from the /genmon command line (per your suggestion)  would be
different than a "standard" restart?

-- 
All quotes are in US dollars, ex works unless stated otherwise, and
subject to prior sale.
Wire Transfer Order- US$50.00 minimum-(does not include freight charges)
Wire Transfer Order less than US$250.00 - additional US$25.00 transfer fee
Credit Card Order     US$25.00 minimum-(does not include freight charges)

Universal Bearing Co.
8521 Lyndon St.
Detroit, MI 48238
Fax 313 933 8888
Tel 313 933 1100
email  ***@***.***
"ON AIRCRAFT USE NOT PERMITTED"
Neither this information block, the typed name of the sender, or anything else
in this message is intended to constitute an electronic signature for
purposes of the
Uniform Electronic Transactions Act or the Electronic Signatures in
Global and National
Commerce Act ("E-Sign") unless a specific statement to the contrary is
included in this message.

jgyates, 2023-04-16:
No. Restarting genmon just kills any existing version running and starts a new version running in memory.