jgyates/genmon #883: Connecting to AWS MQTT Broker

redchairlabs, 2023-04-18:
Before I take on this exercise, I wanted to get an idea if connecting to an AWS IoT MQTT Core is possible when using genmqtt.py. Has anyone ever tried this before?

jgyates, 2023-04-18:
I don't know of anyone trying with the AWS IoT. From brief reading on the topic it *should* work, but I do not have any data points. If you want to try it out and run into any problems feel free to post questions to this thread and I will help how I can. Also, if you do get it working please let the community know in this thread as well.

jgyates, 2023-04-18:
FYI: Here is an example for connecting the paho.mqtt python library (the same library genmon uses) to AWA IoT MQTT:

https://gist.github.com/skirdey/9cdead881799a47742ff3cd296d06cc1

Here is the genmon code that does the same thing:

https://github.com/jgyates/genmon/blob/7e9e46deef07c769805a5ddc273d8ec14aa95ed0/addon/genmqtt.py#L469

redchairlabs, 2023-04-18:
Awesomw! I'll take a look and let you know how it goes. Thanks!

redchairlabs, 2023-04-19:
I'm goign to close this for now and update it when I am able to make this work

closed 2023-04-19 by redchairlabs

jgyates, 2023-10-18:
generatormonitor.software@gmail.com