jgyates/genmon #884: Trying to power via USB vs. Generator; not seeing generator

bdwilson, 2023-04-18:
I've run into a number of issues running off generator power (reading/writing external SD card) and believe I have attributed it to power issues (running Pi Zero 2 W) w/ external Wifi, SD Card Reader, Serial Port and Bluetooth (to read my propane percentages).  I do nightly clones with rpi-clone and noticed SD failures that I cannot replicate when I bring the unit inside and power via USB.  

Before connecting USB power, I disconnected both 5V power and ground from the generator.  When powering via USB, I can confirm the serial connector is powered (it lights up), however I don't seem to be able to talk to the generator. 

**Do I need to re-connect the ground from the Generator to the ground connection on the Pi?**  If not, any other ideas what it might be or things to try?

jgyates, 2023-04-18:
Can you give some more details on this sentence: "Before connecting USB power, I disconnected both 5V power and ground from the generator." ?

Also, when you mention that you are using USB power, is this the Generator Battery to USB adapter or are you using an outlet powered USB adapter?

If you are powering via the generator battery to USB adapter, where is the other 5V connected? and is the 5V coming from the generator connector? USB Provides power, do you also have another 5V also connected to the Pi? 

If you are not communicating with the generator, makings sure the ground is common between generator and the pi is always a good place to start. In most cases just using battery/USB power is sufficient because the transformer in the adapter has a common ground however you may need to connect the ground on the generator molex connector to the pi ground connector on the dual pin headers. if you have a multimeter you can check continuity between the pi ground and the generator ground.

bdwilson, 2023-04-19:
> Can you give some more details on this sentence: "Before connecting USB power, I disconnected both 5V power and ground from the generator." ?
> 
> 
> 

I had 5v and ground from generator (molex) going to 5v and ground pins on pi. 

> Also, when you mention that you are using USB power, is this the Generator Battery to USB adapter or are you using an outlet powered USB adapter?
> 
> 

USB adapter connected to generator battery. 

> 
> If you are powering via the generator battery to USB adapter, where is the other 5V connected? and is the 5V coming from the generator connector? USB Provides power, do you also have another 5V also connected to the Pi? 
> 
> 

The 5v USB is connected to Pi directly via microUSB. The serial connector is powered off Pi pins. 

> If you are not communicating with the generator, makings sure the ground is common between generator and the pi is always a good place to start. In most cases just using battery/USB power is [...]

I assumed I may need to have common grounds from the wiki, but noted that it said if using Generator battery to power USB, this wasn't necessary.

jgyates, 2023-04-19:
I think the issue is that you have two voltage sources (5V from the molex and 5V from transformer in the battery to USB power supply) and two grounds. You can connect the two grounds but you really don't need two supply voltages, if you do want to connect it that way you do need to connect both grounds, otherwise the voltages may not be compatible as they do not share a ground. They way you have it setup is like having a single battery at 12V with 2 12V to 5V transformers. Both grounds (from each transformer) need to be connected, but you really only need one 5V supply. Connecting both 5V outputs will likely cause stress on your components. Connecting both grounds should not have anything but a positive effect.

bdwilson, 2023-04-19:
> I think the issue is that you have two voltage sources (5V from the molex and 5V from transformer in the battery to USB power supply) and two grounds. You can connect the two grounds but you really [...]

Agreed. I would not connect both 5v as I'm not sure what would happen. I'll see if connecting the molex ground to Pi ground resolves the issue and report back.

bdwilson, 2023-04-19:
Confirming that only connecting Generator/Molex Ground to Pi Ground solved the issue for me. I can now see the generator.  Also, it appears my OS-level errors and slowness are gone now that I'm powering via 3.1a USB adapter connected to the generator battery!  Thanks again for your support and this great software.

closed 2023-04-19 by bdwilson