jgyates/genmon #885: USR-W610 - Power Issues

IntelGamer9700, 2023-04-23:
Hi,
  I just got Genmon running yesterday and it was working great for about 18 hours. However, I am having an issue with the USR-W610. 

Here is my setup.
- Genmon - Running inside a docker container on Synology NAS
- Serial over TCP/IP - Using USR-W610 with external Wi-Fi antenna outside generator

I have the USR-W610 powered via 12V SAE connector directly to the generator battery with an in-line 3A fuse. The device powered up perfectly yesterday and was running until early this morning. Then, Genmon lost connectivity to the serial device. I went outside and the device only had the red power light on (No Wi-Fi link and no serial port activity). I powered cycled the device and still no Wi-Fi or serial. I am wondering if the battery voltage has something to do with it as I saw this same behavior inside the house with a small 12V battery when I was setting it up. If I charged the 12V battery for a while, then the device would boot up properly. 

Not sure where to go from here as it seems others have the same device working without issue.

jgyates, 2023-04-23:
I would try one of two things.

1) Use power and ground from the molex connector to power the the WR-610 

or

2) connect the pin 2 on the molex to RS-232 ground pin 5 on the DB-9 connector on the WR-610

or 

3) Connect the ground from the molex connector (pin 2, see here https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable) to the ground from the battery. 

The theory here is that the controller voltage and the battery voltage are not on the same ground reference so this will make RS-232 comms problematic. This is just a guess, but I have seen issues with RS-232 on the pi when powering the pi off the battery via a USB power supply and not connecting the RS-232 ground.

Let me know how it goes.

Method one is preferred. If you method 2 first and it does not work then try method 1. I use the WR-610 with method 1 and don't have any issues.

closed 2023-04-23 by jgyates

IntelGamer9700, 2023-04-23:
Thanks for replying back so quickly. I understand the ground issue could cause serial communication issues. However, I would think the device should still boot up and the Wi-Fi should work regardless of whether the serial device is connected or not. 

I am currently using the StarTech 12V 8 Pin adapter (https://www.amazon.com/dp/B000M802RG?psc=1&ref=ppx_yo2ov_dt_b_product_details) with pins 7 and 8 going to serial port. Therefore, I still have MobileLink connected downstream from adapter. Is option 1 still possible? I did measure that the serial adapter only uses 0.8 watts which I think is less than a Pi. Should I be concerned about pulling too much power from the controller for both the serial adapter and MobileLink? I believe pins 1 and 2 go to MobileLink, so I would have to splice into them for power. Thanks.

jgyates, 2023-04-23:
In an ideal situation, I agree that Wifi should work regardless, however without the ground it it is the electrical equivalent of applying some unknown voltage to the WR-610  RS-232 interface so this could effect it in a variety of ways.

Unfortunately I don't know if this would work with mobile link connected. You could try it with the 12V. The only thing I have personally tried is the 5V and pin 2 for ground. I know others use the WR-610 but this is the first time I have seen this type of issue, so it could be resolved with the 12V, 5V or even a bad WR-610. Since you have an Evo 2.0 and appear to be using the mobile link cellular and not the Wifi adapter (which uses 12V) then trying it with pin 3 and 4 may work.

IntelGamer9700, 2023-04-23:
Ok, maybe I am looking at the pins incorrectly then. I am definitely using MobileLink with Wi-Fi. I could have sworn it was using pins 1,2,5,6 (front part of molex adapter) and pins 3,4,5,6 (back part of adapter) were available. I guess the USR-W610 could also be faulty as you suggested.

[image]

Curious if anyone has used a serial to Ethernet adapter that is PoE powered. My generator is only 6 feet from the house, so I could definitely run Ethernet inside some conduit.

jgyates, 2023-04-23:
Maybe I am mistaken in mobile link uses 12V. I have an Evo 1 Liquid Cooled unit so don't have one for reference. Also, the above diagram pin numbers show the receptacle on the controller so this would match the back of the molex cable, not the front. A picture of your cabling may help resolve any questions.

skipfire, 2023-04-23:
I have an Evo2 so it is possible it is different, but my MobileLink WiFi unit uses 5v and the 5v ground; for pins it is 1,2,5,6 filling one side of the plug and the other half of the plug has no pins.

IntelGamer9700, 2023-04-23:
Thanks skipfire. My wiring mirrors what you stated. So, the question is whether I could use pin 3 (12V accessory) and pin 4 (RS-232 port ground) to power the USR-W610? The device says it can use DC 5-36V. Technically, 12V accessory from the controller should be the same as 12V from the battery.

IntelGamer9700, 2023-04-24:
Seems like powering serial devices either via battery or controller can be problematic. Thinking for trying a Moxa serial to Ethernet device. It can be powered via PoE. The model is Moxa NPort P5150A-T). Seems like it should work as it supports TCP Server Mode. 

The manual says: In TCP Server Mode, the NPort is configured with a unique IP:Port combination on a TCP/IP network. In this case, the NPort waits passively to be contacted by the host computer. After the host computer establishes a connection with the serial device, it can then proceed with data transmission.

jgyates, 2023-04-24:
I have seen people use this project and power from PoE, the battery (via USB transformer) and the molex connector without issues. The vast majority of people using this project power from either the battery via USB transformer or from the controller. I don't know anyone using the battery directly. That is not to say that you can't I just am not aware of it. 

When people using this project do have power issues, it is typically resolved by making sure all your grounds are connected together. If they are not connected together the RS-232 receivers on the controller or on the genmon side will likely be unable to interpret the voltage level a RS-232 compatible signal. If you have two system powered by different voltage sources and you try to connect them together electrically (for example using RS-232) they must use a common ground reference or risk incompatibility or damage to the circuits that are being connected.

PoE is just as problematic as any other power source if your grounds are not properly connected, however it it makes your cabling easier then that is great. The serial TCP support in genmon is straightforward. It uses TCP as a transparent transport of the serial data sent and received from the controller. It also supports modbus TCP, but this is not used with most serial over IP devices. The genmon modbus over TCP mode is typically only used for industrial controllers so if you go that route keep that in mind. The Moxa description above does sound like it should work.

lwbeam, 2023-08-21:
I power [my Genmon setup](https://github.com/jgyates/genmon/issues/31#issuecomment-747104229) directly from the generator battery via a [12Vdc to 5Vdc buck converter module](https://www.ebay.ca/itm/DC-DC-Step-Down-Buck-Converter-Power-Supply-Module-24V-12V-9V-to-5V-5A-25W-MF/172358062967/). The output from this module goes to the two 5V GPIO pins on the Raspberry Pi. The only other generator connections used are pins 7 and 8 on the Remote Wireless Connector (J6). I don't use pin 2 on this connector as the signal ground. All ground connections are tied to the negative battery terminal.