jgyates/genmon #887: Illegal instruction error on new installation

star-glider, 2023-04-24:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Installs and runs on new 32-bit Raspberry Pi OS (Debian/Bullseye) on Zero W.

### Actual Behavior

Launch fails with "Illegal instruction" error

### Steps to Reproduce (including precondition)

1. Flash Raspberry Pi OS Lite 32 Bit on new SD and run on . . .
2. . . . a Pi Zero W (not 2W)
3. sudo apt update / full-upgrade
4. update pip with: sudo pip install -U pip
5. follow all instructions on enabling serial port (serialtest.py is successful)
6. clone repository
7. Comment out the fluid package installation line in the genmonmaint script (per issue #748; I was also hitting this problem but have no need for genmopeka anyway)
8. run genmonmaint.sh -i

### Screenshot or Pictures relating to the problem (if possible)

[image]

[image]

### Your Environment

- Generator Model: H-100 (although not relevant to this issue)
- Genmon Logs: Submit via the About page (can't access web UI)
- Genmon Version: 1.18.14

### Additional Notes ###

Initially, I skipped step 4 (updating pip). This created an interesting issue, whereby pyopenssl would install (via the script, or I could do it manually) and then completely break my pip installation. Running even just "pip3 -V" would result in an "illegal instruction" error, and the installation script would fail with consecutive "illegal instruction" errors after installing pyopenssl. Removing and reinstalling pip and even all of python3 could not repair it. I didn't experiment too deeply; just reimaged the OS and tried again. I was able to narrow down the problem to pyopenssl, as installing that would reliably break pip even without touching the genmon repository. This is obviously not a genmon-specific problem, so I didn't raise an issue.

My next step was to add in step 4 above, namely updating pip before installing any python packages. This did work to allow me to get everything installed, and the script completes successfully. pyopenssl installs, pip remains "unbroken," and I can run other python3 scripts successfully. However, I now get "illegal instruction" errors when trying to run genmon, as reported in the issue above. My guess is that some module (probably openssl) is causing trouble.

Before my addition of step 4 above, I also just tried to skip pyopenssl in the installation as a debug attempt. This actually worked out okay, but when I tried to run genmon, I would see "[Errno 111] Connection refused : myclient.py:58" in the genmonserv.log, which based on my (somewhat limited) read of the code implies that the different programs are failing to talk to each other over a local socket. It seems likely that that issue is related to the absence of the openssl library, but it's the closest I've gotten to getting anything running. My next step would be to see if I can modify the code to not use SSL at all and get away with never installing pyopenssl, since I suspect that's what's breaking things, but that's obviously a really hacky workaround, obviously not desired behavior, and I'm sure there's a deeper issue here because everyone else seems to be running it just fine.

The pi is brand-new, and I'm using a completely fresh installation with nothing changed from a vanilla image except setting a wifi SSID, adding an SSH key, and setting a username/password (all via the imager app). Running no additional software.

jgyates, 2023-04-24:
Your pi os installation is likely corrupt in some way. If you can't run "pip3" then genloader will not be able to run as it calls the pip3 binary during the start process. Also, the install script will install pip if it is not already installed. This page has the procedure:

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

although I don't see any thing out of the ordinary in your description of your setup. The errors you are seeing are in line with genloader not loading the software (which will happen if pip3 not working properly).

I would recommend one of two things:

1) If you do not see anything else wrong with the system then uninstall and reinstall pip3 until you can get "pip3 -V" to work.
2) reinstall the OS using the latest from the Raspberry pi foundation and use these steps:

     https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

Also, if you can get pip3 working using option 1 listed above, your notes above state that you are running genmon v1.18.14, which is from August 2021. To update from the command line type (from the gennmon folder):

        ./genmonmaint.sh -u

Based on your description, it could be that you initially had a dependency issue with pyopenssl, but a 2nd problem was introduced with the pip upgrade.  I have seen the pyopenssl issue before in some rare cases. It appears to be related to the pip repository pyopenssl dependancies. I am not exactly sure what is occurring but it does appear to be an issue with the pyopenssl. I have seen it happen on some 32 bit installs. Unfortunately this occurs by just installing the pyopenssl library (which is used by the web server), however it does not occur in all cases so my guess it has to do with mirror sites for pip not being in sync, but that is just a guess. If you get the pyopenssl error again then let me know and we can try to go from there. My guess is that the upgrade to pip did not resolve the pyopenssl issue, you just hit a different mirror on the subsequent install since pyoenssl is installed using pip.

Also, I assume that you don't have a pi that is capable of running the 64 bit version (the pi zero w will only to 32 bit), if you do I would recommend that as genmon uses a lot of open source packages and the package maintainers support and test the 64 bit version more than the 32 bit versions at this point.

Let me know how it goes.

star-glider, 2023-04-25:
Whew, okay I figured it out. Was quite the journey. TLDR; something is wrong with the cryptography package that pip installs. The solution that worked for me was to uninstall the cryptography module (pip3 uninstall cryptography) and then use apt to install those libraries in the correct architecture (apt install python3-cryptography). Then I bypassed your pyopenssl check in genloader to avoid having it attempt to reinstall it as part of openssl. Then everything worked.

### More Detail ###

Here's the stack trace for the illegal instruction error I was getting:

`GNU gdb (Raspbian 10.1-1.7) 10.1.90.20210103-git

Reading symbols from python...
(No debugging symbols found in python)
(gdb) run genmon.py
Starting program: /usr/bin/python genmon.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0xb61975c8 in ?? () from /lib/arm-linux-gnueabihf/libcrypto.so.1.1
(gdb) continue
Continuing.
[New Thread 0xb5e82440 (LWP 623)]
[New Thread 0xb54ff440 (LWP 624)]
[New Thread 0xb4aff440 (LWP 625)]
[New Thread 0xb42fe440 (LWP 626)]
[New Thread 0xb3afd440 (LWP 627)]

Thread 1 "python" received signal SIGILL, Illegal instruction.
0xb3210760 in pyo3::gil::ReferencePool::update_counts ()
   from /usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust.abi3.so
(gdb) continue
Continuing.
Unable to fetch general registers.: No such process.
Unable to fetch general registers.: No such process.
(gdb) [Thread 0xb3afd440 (LWP 627) exited]
[Thread 0xb42fe440 (LWP 626) exited]
[Thread 0xb4aff440 (LWP 625) exited]
[Thread 0xb54ff440 (LWP 624) exited]
[Thread 0xb5e82440 (LWP 623) exited]

Program terminated with signal SIGILL, Illegal instruction.
The program no longer exists.
Quit`

That seems to imply that the cryptography module is asking the processor to do something it can't, probably to your point about the 32 bit architecture being less-than-perfectly supported. In any event, the following procedure works to reliably work around this problem:
1) sudo pip3 install -U pip (this keeps the pyopenssl install from trashing all of pip3)
2) Follow all of the wiki instructions and run the install script, commenting out fluid due to an unrelated problem I had
3) sudo pip3 uninstall cryptography
4) sudo apt install python3-cryptography
5) comment out the openssl check in genloader.py
6) reboot and enjoy

I'm still shocked that I'm the only person hitting this, since I'm using a plain-vanilla Raspberry Pi OS installation. It is a brand-new, post-shortage Zero W, so maybe there's some esoteric processor difference in these new ones that's not been accounted for yet? In any event, I got it working! The openssl library might be somewhat damaged by all of this; I'm not sure if it's possible to use some of the SSL features like email. I'll play around and report back if I find anything interesting there.

Thanks for your help; hopefully this helps out someone else.

closed 2023-04-25 by star-glider

jgyates, 2023-04-25:
I am glad it is working for you.

it may be because you are using the pi w rather than the pi w2 or a newer pi model. The CPU is different on that model so that could be the cause of the illegal instructions slipping past the the testing the crypto maintainers perform. Also, if this is an issue with the crypto libraries then it may be resolved in the coming weeks once they figure out the issue.

Also, if you modified genloader.py then you will want to be mindful regard any software updates as this would overwrite your modification.

jarocht, 2023-06-02:
I'm also running into this issue I believe. Also on a pi zero w.

It's late -- will try these steps tomorrow on a fresh install. My current install seems borked beyond recovery at this point.