jgyates/genmon #889: H-100 Setup & Parts List

star-glider, 2023-04-30:
There don't seem to be too many examples of H-100 installations here, so I figured I'd share mine in case it helps anyone. The nice thing about the industrial controller is that it has a native DB9 port; the irritating thing is that it's mounted such that you can't close the cover. I saw one guy here physically moved the port, which seemed like a little too much work. :) I just got a simple right-angle adapter to solve the problem, and fortunately I found one that has the right "directionality" to it. The cover now serves the semi-helpful purpose of holding it in the port. Here's what it looks like all done:

[image]
[image]
[image]
[image]

I hope this helps any fellow industrial folks!

Parts list:
Right-angle serial adapter: https://www.amazon.com/dp/B001S2PYV4
Serial cable: https://www.amazon.com/dp/B00008VF6L
RS232 <> TTL board (I liked this one because it had through-hole as well as pads): https://www.amazon.com/dp/B07BJJ3TZR
Simple buck converter (I'm powering from battery): https://www.amazon.com/dp/B07Y2V1F8V
Cheap fuse holders: https://www.amazon.com/dp/B0813Q4S6P
Switch so I can restart the pi after powering it off: https://www.amazon.com/dp/B074V5WHSX
Enclosure: https://www.amazon.com/dp/B08PP1W8Q3

jgyates, 2023-04-30:
Nice! Thanks for the details

jgyates, 2023-10-24:
I added a link to this thread in the wiki: https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

closed 2023-10-24 by jgyates