jgyates/genmon #890: Setting Time crashes H-100 Controller

star-glider, 2023-04-30:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

### Expected Behavior

Set H-100 controller clock via Genmon

### Actual Behavior

Attempting to set the clock locks the controller into an error mode; genmon is unable to communicate until controller is reset (I pulled the fuse). The H-100 has an RS232 page that shows the status of the connection; the "error" indicator as well as the "receive" indicators are shown. Disconnecting the cable removes the error indicator on the H-100 screen, but only temporarily. Restarting Genmon or rebooting the Pi has no effect.

A hard reset of the controller fixes the problem. Interestingly, I had the same behavior one other time while I was on the settings page, but I'm not sure what triggered it. Attempting to set the clock is definitely does; I tested that a few times and was consistently able to reproduce. I suspect it could be that any time genmon tries to _write_ certain data to it, the H-100 freaks out and goes in to some kind of lockout mode.

Suggested fix would be to disable sending the problematic commands when control type is set to H-100.

### Steps to Reproduce (including precondition)

Connect to H-100 and attempt to set the clock.

### Screenshot or Pictures relating to the problem (if possible)

Here's a link to the H-100 OM: https://gensetservices.com/wp-content/themes/gensetservices/PartsPDFs/0F3750.pdf
The Mbus Slv: 100/RS-232 page displays: "T- R* E*" after I try to set the time, which per the manual indicates that it's not transmitting, it is receiving, and it is in error. Normally, it displays T* R* E-

### Your Environment

- Generator Model: 25kw; H-100 Industrial Controller
- Genmon Logs: Submitted on 4/29
- Genmon Version: V1.18.17

I'm happy to help troubleshoot!

jgyates, thank you so much for the fantastic software; this is a really wonderful project. I'm more than happy to help out.

jgyates, 2023-04-30:
@star-glider 

Thanks for this info. A few questions:

1) Can you submit your logs on the about page? This requires that outbound email be setup and working.
2) What is your firmware version of your H-100?
3) Are there any settings on the H-100 that deal with privilege or write protection? I don't have access to a H-100 (I wrote that with the help of remote access to a system). Could it be that another setting on the H-100 is preventing this form occurring. The reason I ask is that this was working and I don't know if this is a H-100 setting issue or the firmware has changed or this is an older firmware that does not support this Modbus command.

Once I get some more info it may be more clear what path needs to be taken to resolve this. 

Also, if anyone else using a H-100 can test and we can compare firmware versions?

star-glider, 2023-04-30:
1. Done! They were sent at 5:28 PM PT
2. Looks like it's "Release 4.0C1, HW00" (see picture)
3. I couldn't find any settings related to modbus _at all_. I poked around in the menu at length when I was trying to troubleshoot, and the manual is almost useless with virtually nothing mentioned. I don't know if there's a "secret" dealer manual floating around somewhere. I do have a good relationship with the guys who service it. Next time they're out (typically around August before the winter storms), I can ask about it.

I recognize that this is almost impossible to troubleshoot remotely, especially since I don't think the logs will be that helpful. At least I didn't see anything useful; the comms appear to just _stop_, as if I'd pulled the cable. Now, the logs inside the H-100 could be helpful, but heaven knows how to get at them or if it even keeps such things.

[image]

jgyates, 2023-04-30:
I have not received your logs. You can test outbound email on the settings page under "Optional - Outbound Email Settings"

Also, if you email is setup properly you should get two emails when genmon starts (one with the IP address of the pi and another once the initialization is complete with the generator status).

star-glider, 2023-04-30:
hmm that's odd. The email is definitely working; I get the alerts. I also see the logs in my sent mail, oddly enough. I just tried reforwarding.

jgyates, 2023-04-30:
OK. got it. it flagged the domain of you email address as junk for some reason. Not sure about the ones from yesterday as those were not in the junk folder. I have quite a few other log submissions with firmware revisions before and after yours. I don't have any data on if they are seeing a similar issue with setting the time. Let me look into this a bit more.

jgyates, 2023-05-04:
Not really a solution but here is some additional data. Of the people who have submitted registers using the H-100, the firmware breaks down like this:

   "H-100 Controller, PM-DCP Release 4.0C1, HW00": 1,
   "H-100 Controller, PM-DCP Release 4.2A1, HW00": 1,
   "H-100 Controller, PM-DCP Release 4.3A , HW00": 2,
   "H-100 Controller, PM-DCP Release 3.5C , HW00": 2,
   "H-100 Controller, PM-DCP Release 3.6D , HW00": 1,
   "H-100 Controller, PM-DCP Release 3.5Ce, HW00": 1,
   "H-100 Controller, PM-DCP Release 4.0A , HW00": 1,
   "H-100 Controller, PM-DCP Release 4.2B3, HW00": 1,
   "H-100 Controller, PM-DCP Release 4.1B1, HW00": 1,
   "H-100 Controller, PM-DCP Release 3.5": 1,

This may not reflect the current running firmware for these installations as some of these were submitted in the past and could have been updated since then. You may want to update to at least 4.2 or 4.3 as that has been around for a few years and is likely the most bug free and feature rich.

jgyates, 2023-05-05:
On the Monitor page of the web interface, there is a section on communication stats. Do you notice any errors in this section (like timeout or modbus exceptions) after you try to set the time. Note that these error counts are reset once genon is restarted. the logs you sent did not show any errors but you may have restarted genmon since the last error

Also, does genmon show your generator time correctly on the bottom of the main status page?

What I really need is another data point on this issue with this issue, or any controller logs or specific errors on the controller.  At the moment I don't have a lot to go on.

star-glider, 2023-05-06:
I looked closely at that Monitor page while I was troubleshooting the issue. Each time "Timeout Percent Errors" went to 100%, and Time Errors gradually incremented. My RS232<>TTL board also has TX/RX lights, and it showed Genmon trying to send a packet every few seconds at the same pace at which the Timeout Errors incremented, just as it would do if the cable were disconnected. There were no CRC errors, modbus exceptions, or other errors. As far as I can tell, the H-100 simply stopped communicating.

Yes, Genmon does correctly show my generator time, which is slightly different from Genmon's (and actual) time. It seems to be able to read everything just fine, and I have no other missing data except that my Run Log is empty (even thought the generator has in fact run). The Alarm Log is correctly populated.

I hear you on not having much to go on. There aren't any controller errors I can find, either: only that the Modbus display shows T- R* E*, which indicates that it's receiving, in an error state, and not transmitting. When it's working, it'll show T* R* E-, which indicates both Tx and Rx and no error state. That's all the information I could dig out of it.

For now, I think we probably close this as an H-100 bug. It would be possible to modify Genmon to disallow setting the time when it detects this firmware version, but that might not be worth the trouble. For one, we don't know which firmware versions cause the problem---just that mine in particular does. Second, this generator is ~15 years old and probably nearing replacement age. I suspect that the firmware has never been updated, so it's probably fifteen years old, too. The likelihood that someone else hits this error seems low to me.

Finally, Genmon has been working flawlessly since I opened this issue. Simply not setting the time is an easy workaround. We could add a note to the Wiki that this can happen in rare cases, and that the fix is to open up the H-100 panel, pull the middle fuse to reset it, avoid setting the time, and keep the two time sync options in Settings disabled. I'm happy to submit a write up to that effect for inclusion.

Thank you again for such great software and being almost ludicrously responsive to issues! When I finally do have to replace this, about the only thing that would steer me towards Generac over Kohler is this software.

jgyates, 2023-05-06:
Sounds good. I you want to just put what you want in this thread and I will update this page in the wiki:

 https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

closed 2023-05-06 by jgyates

star-glider, 2023-05-06:
How about:

"One user has reported that attempting to set the time on an older H-100 causes the controller to stop communicating with Genmon. The Modbus menu on the H-100 will show T- R* E*, indicating that it is receiving data but is in an error state and refusing to transmit. Genmon will show a communications failure. To reset the controller, open the front panel and pull and reinsert the middle fuse (this will NOT reset the H-100's memory or even affect its clock; all internal logs and settings will persist). The current suggested workaround is to refrain from setting the time on these older controllers and ensure that both "Sync Time" and "Sync Daylight Savings Time" options are disabled in settings. Genmon will otherwise operate normally."

jgyates, 2023-05-06:
Sounds great. I updated the wiki

https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

liltux, 2023-05-11:
@jgyates, I was reading through and would like to share some more info.  The fuse pulled is specifically F2 fuse power to the H-Panel.  You are correct it does not reset the programming.  It does however, affect the internal clock if left unpowered long enough.  There is not an internal battery backup of the clock.  Also, setting the Generator clock/time manually is as simple as going to the MENU, STATUS, and arrow over the date time fields that need to be updated.

jgyates, 2023-05-12:
Thanks for the info!