jgyates/genmon #892: HTTPS UI stops responding after 3-4 days

bbrown-sc, 2023-05-09:
### Expected Behavior

I've been having this problem for a couple of years. So I recently decided to upgrade to the lasted raspbian (bullseye) and genmon V1.18.17 to see if it would fix tit. It didn't. I'm configured:

_usehttps = True
useselfsignedcert = True
http_user = xxxxxx
http_pass =  xxxxxx
https_port = xxxx_

When I first connect I get the usual (chrome on windows):

_NET::ERR_CERT_AUTHORITY_INVALID
Issuer: Untrusted Authority
Expires on: May 6, 2024
Current date: May 9, 2023_

I click advanced and proceed (unsafe) to genmon. I login and everything works fine. After a few days the UI stops responding and using ssh I have to _./genmon/startgenmon.sh restart_ to get it working again. The genserv.py process is running and there's nothing in  _/var/log/genserv.log_ that explains why this is happening. I thought it may have something to do with the self signed certificate (supplied byFlask library) so I tried using the workaround suggested in the genmon.conf comments to generate my own keyfile and certfile. That didn't work. genserv.py fails to load and I get the error in /var/log/genserv.log  _Error in app.run: [SSL] PEM lib (_ssl.c:4022) : genserv.py:4290_  I should mention that I use dynamic dns and port forwarding in my router using the domain _https://xxxxxx/home.dyndns.org:xxxx_. I thought that might be a source to my problem but I have the same issue locally using _https://xxx.xxx.xxx.xxx:xxxx_. I don't want to run straight HTTP since I want to access genmon remotely.

### Your Environment

- Generator Model: Generac H-100 Industrial Generator
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.17

jgyates, 2023-05-09:
@bbrown-sc 

The ERR_CERT_AUTHORITY_INVALID error is expected for self signed certificates. Nothing can be done about (except change your browser settings, but that is not advisable) that as this is the browser tightening security.

Regarding generating your own certificates, the error you are seeing ([SSL] PEM lib ) is usually caused by an invalid private key. If you want to troubleshoot that we can. I good place to start is by validating your key and cert file. This article shows how to do that:    https://www.ibm.com/support/pages/how-verify-if-private-key-matches-certificate

Regarding the issue of not being able to access the web UI after a period of time. I have seen this before on occasion and due to some trial and error, the issue appears to related to certain routers and external port forwarding. You can see some of the steps that have been taken in troubleshooting this issue here:

https://github.com/jgyates/genmon/issues/271 
https://github.com/jgyates/genmon/issues/789

Typically I categorize this type of issue (unable to access web UI when using port forwarding) under the ["Router" label](https://github.com/jgyates/genmon/issues?q=label%3ARouter+). Note that not all of these are exactly like your issue, but some of them are.

The best solution is to use a VPN instead of port forwarding. See the note on [connectivity here](https://github.com/jgyates/genmon/wiki#connectivity).  The web server built into the flask libraries is not design to withstand any type of attack from the internet. A VPN is designed to to just that. Most home routers have a built in VPN and clients for a PC or mobile are widely available. If you have a spare PC or pi you can setup a VPN server on your network and connect to it remotely. I do this with OpenVPN

https://www.digitalocean.com/community/tutorials/how-to-set-up-an-openvpn-server-on-debian-11

There are android and apple clients for OpenVPN. Some routers have this built in. In my experience it is rock solid. 

Let me know if you have any other questions.

bbrown-sc, 2023-05-09:
@jgyates I do have a VPN (Tailscale) setup on this host and it works fine. I'll turn off port forwarding for genmon on my router and only use Tailscale externally to see if it stays up longer (to test the hacking theory) and report back.

closed 2023-05-09 by bbrown-sc

jgyates, 2023-05-09:
yes, please let me know hot it goes. Also, I don't have enough info to say if it is hacking or some issue with the router. It could be both (i.e. the router is proactively shutting down access to that port as it has detected possible malicious activity.

bbrown-sc, 2023-05-10:
When genserv.py becomes unreachable, it can't be accessed locally with direct ip:port either. My router is pretty old and doesn't have any intelligence or firewall, so I don't think router forwarding is the problem. I have other ports open that have never had a problem.

bbrown-sc, 2023-05-21:
@jgyates I wanted to let you know that after disabling port forwarding on my router I haven't had genserv.py go off line for 2 weeks. I suspect it was hacking or DOS attacks that caused this problem.

jgyates, 2023-05-21:
Good to know. Thanks for the update.