jgyates/genmon #894: SNMP SetupSNMP: socket() failed: [Errno 97] Address family not supported

msteil4vse, 2023-05-10:
Hello All,

We purchased one pre-assembled unit from pintsize.me.  Successfully installed it, and it is working as expected and very happy with the unit.  Now the supplied unit was based on the RB Pi 3A+ platform.   We have 3 more identical Generator installations to deploy on, but with no WiFi available, so we then purchased all the needed components to build 3 more, including associated boot flashcards, and then purchased 3 RB Pi 3B+ units to accommodate the need for a Ethernet port.   So in short the only real difference in the installations is the Pi Platform itself.  All assembly went well, and very happy overall with Pintsize.me and highly recommend them as they have been very helpful.   All functions of GenMon that we have configured work as advertised on the 3B+ platform with the exception of SNMP.

Now the issue at hand is when attempting to enable the SNMP module on the 3B+ units the module fails to load with the error:

Error in SetupSNMP: socket() failed: [Errno 97] Address family not supported by protocolcaused by <class 'OSError'>: [Errno 97] Address family not supported by protocol : gensnmp.py:1524

see log files:

[genmon_logs.tar.gz](https://github.com/jgyates/genmon/files/11442614/genmon_logs.tar.gz)

I have done some consultation with professor Google and while there is information available, most is past my understanding.

Now I know I have omitted a great deal of information that may be relevant, in an attempt to keep the initial post simple and focused.  So ask and I will attempt to get the needed supporting information.

Does anyone know what this error is trying to tell me and can you point me in the correct direction? 

Best Regards

jgyates, 2023-05-10:
It looks like this is failing when registering a SNMP IPv6 transport. Have you disabled IPv6 on the pi? From a console you can type:

      ifconfig

you should see someting like this:

    eth0      Link encap:Ethernet  HWaddr 00:1C:C0:AE:B5:E6  
          inet addr:192.168.0.1  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::21c:c0ff:feae:b5e6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:41620 errors:0 dropped:0 overruns:0 frame:0
          TX packets:40231 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:21601203 (20.6 MiB)  TX bytes:6145876 (5.8 MiB)
          Interrupt:21 Base address:0xe000  

eth0 may be wlan0 or something else but if you do not have "inet6 addr" in your output then your IPv6 is disabled. it is enabled by default with the Pi. The SNMP add on assumes it is enabled (I could possibly make a change to allow it to be disabled). I will try to make that change and check it in. I will let you know once it is there. Or you can enable IPv6 and test it yourself.

msteil4vse, 2023-05-10:
Now that was quick, I just stepped away on an another issue and hear is possibly the solution.

Yes I do have IPV6 disabled as it initially gave me fits trying to do updates.  Seems to work on WiFi, but not Ethernet for some reason.

As for your solution and the changed code I'm already on the V1.18.17 and I see the notes in the change log, as you state,  but no increment on the version number.  The Web interface does not seem to do upgrade, or at least no change in the log output

Is there a better way to force an upgrade, or confirm that it actually change something?

skipfire, 2023-05-10:
You can call the command line update `./genmonmaint.sh -u -n -p 3`. 

From .16 to .17 I had to use git to update manually ( #876 ).  git fetch to have it sync, status will tell you if you are current or not, git pull will yank the files down.
```
git fetch
git status
git pull
```

jgyates, 2023-05-10:
if you click the "Update to the latest" button on the about page, it will update the repository and restart the software irregardless of the version number. This update should result in gensnmp loading (if it is enabled) without the previous error. 

Sometimes the update may take some time to make it to all of github mirrors.  You can validate you have the update by looking at the date of the file ./genmon/addon/gensnmp.py. From the genmon folder:

      ls -la ./addon.gensnmp.py

It should be todays date if the update occurred. From the command line you can update (from the genmon folder) with this command:

    ./genmonmaint.sh -u

then type 'Y' to confirm the update.

Let me know if you have any issues or questions.

jgyates, 2023-05-10:
I am glad it is working for you now. Let me know if you have any other issues.

closed 2023-05-10 by jgyates