jgyates/genmon #895: Bluetooth/Mopeka on 3A+ Pi Raspbian GNU/Linux 11 (bullseye)

msteil4vse, 2023-05-16:
Hello All,

This is most likely not the correct place to post, but the underlying goal is to get some Mopeka sensors to work with GenMon.

First off for the record everything is update/upgraded to the latest and greatest and using a standard installation from pintsize.me

The Mac address is configured, the associated switches and tank size is selected.

the log returns

2023-05-16 13:49:17,547 : Error in TankCheckThread: [Errno 19] No such device : genmopeka.py:550

doing some investigation I do not think the issue s GenMon related, I believe it is an underlying Bluetooth issue 

Initially I do not believe the Bluetooth service was running, I started it and here is the status output

root# systemctl status bluetooth
‚óè bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-05-16 14:22:07 CDT; 1s ago
       Docs: man:bluetoothd(8)
   Main PID: 21990 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 407)
        CPU: 75ms
     CGroup: /system.slice/bluetooth.service
             ‚îî‚îÄ21990 /usr/libexec/bluetooth/bluetoothd

May 16 14:22:07 HeadendGenMon.vsenterprisesltd.com systemd[1]: Starting Bluetooth service...
May 16 14:22:07 HeadendGenMon.vsenterprisesltd.com bluetoothd[21990]: Bluetooth daemon 5.55
May 16 14:22:07 HeadendGenMon.vsenterprisesltd.com systemd[1]: Started Bluetooth service.
May 16 14:22:07 HeadendGenMon.vsenterprisesltd.com bluetoothd[21990]: Bluetooth management interface 1.22 initia>

root# bluetoothctl show
No default controller available

I have done considerable reading and searching on the issue and seems I am not the only one with Bluetooth woes, but I cannot seem to hit on the right article that fixes my difficulties.  Obviously the controller needs to be available / operational  but find the magic bullet....

Is there someone that knows what/how to make this work that can post the appropriate instructions here for myself and those to follow it would be a great help if the issues solution could be put in this issue.

Advice would be greatly appreciated.

jgyates, 2023-05-16:
I did not see this in your initial post so we should probably check the serial port usage. By default the bluetooth port uses a serial port (and so does genmon) so the setup is a little different when using the serial port.

The details are [here](https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmopekapy-optional).

To use the mopeka sensor the Bluetooth interface must be enabled on the raspberry pi. Normally, the Bluetooth device on the pi would default to using the onboard serial port, however genmon typically uses the onboard serial port so the serial setup program serialconfig.py disables the Bluetooth device. Since Bluetooth is required to communicate with the mopeka sensor the Bluetooth device needs to be enabled, but mapped to the secondary serial port. To do this run the following command in the /genmon/OtherApps folder from the console on your pi:

  sudo python3 serialconfig.py -e -b

then reboot.

then type

    sudo bluetoothctl

then type

    exit

to quit.

At this point your bluetooth controller should be using an alternate serial port. Let me know if that does not work.

msteil4vse, 2023-05-16:
Yes, I did this step when I did the initial config per the documentation, can Yes I see that I was not specific my apologies.

after doing the outlines procedure again, Now I am seeing some improvement as now I get back that the "Agent Registered"

genmonpi@HeadendGenMon:~$ sudo systemctl status bluetooth.service
‚óè bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-05-16 16:40:18 CDT; 2s ago
       Docs: man:bluetoothd(8)
   Main PID: 15706 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 407)
        CPU: 124ms
     CGroup: /system.slice/bluetooth.service
             ‚îî‚îÄ15706 /usr/libexec/bluetooth/bluetoothd

May 16 16:40:17 HeadendGenMon.vsenterprisesltd.com systemd[1]: Starting Bluetooth service...
May 16 16:40:18 HeadendGenMon.vsenterprisesltd.com bluetoothd[15706]: Bluetooth daemon 5.55
May 16 16:40:18 HeadendGenMon.vsenterprisesltd.com systemd[1]: Started Bluetooth service.
May 16 16:40:18 HeadendGenMon.vsenterprisesltd.com bluetoothd[15706]: Bluetooth management interface 1.22 initialized
genmonpi@HeadendGenMon:~$ sudo bluetoothctl
Agent registered
[bluetooth]# 
[bluetooth]# exit
genmonpi@HeadendGenMon:~$ 

But I still get in the genmopeka.log 

2023-05-16 16:42:05,017 : Error in TankCheckThread: [Errno 19] No such device : genmopeka.py:550

jgyates, 2023-05-17:
Can you try to restore your original serial settings

      sudo python3 seiralconfig.py -r

then set them again?

    sudo python3 serialconfig -e -b 

Then reboot the pi. Once you reboot try this command again:

    sudo bluetoothctl

Once this command completes, you should get a MAC address for your bluetooth controller. To me it looks like the BT service is started but your hardware is not setup as the driver is expecting.

If that does not work then what is the output of this command:

      cd OtherApps
     sudo python3 serialconfig.py -c -b

This will check that the recommended settings for BT with the mopeka add on are correct.

The output should look like this:

    Checking : Is enable UART in boot config: OK
    Checking : BT alternate serial port: OK
    Checking : Turbo Clocking: OK
    Checking : Disable BT Disable Overlay: OK
    Checking : Serial console command line removed: OK
    Checking : Serial console service disabled: OK
    Checking : BT service enabled: OK

What might have occurred (a guess) is that if you have the pintsize.me product and it was pre-configured then you may need to restore the original boot.txt file that serialconfig modifies then set it up explicitly for bluetooth rather than just plain serial. By default BT and the serial port are not compatible together unless you take some extra steps. Let me know how it goes.

jgyates, 2023-05-18:
Were you able to get this working? I am going to close this issue for now but let me know if you have any update on this issue.

closed 2023-05-18 by jgyates

msteil4vse, 2023-05-19:
Well, my previous post I must not have hit the comment button or something, as it is not here.

No, not working, and when I try to reset back to defaults I get..

genmonpi@HeadendGenMon:~/genmon/OtherApps$ sudo python3 ./seiralconfig.py -r
python3: can't open file '/home/genmonpi/genmon/OtherApps/./seiralconfig.py': [Errno 2] No such file or directory
genmonpi@HeadendGenMon:~/genmon/OtherApps$ ls -lsa
total 92
 4 drwxr-xr-x  2 genmonpi genmonpi  4096 May 16 12:58 .
 4 drwxr-xr-x 12 genmonpi genmonpi  4096 May 16 13:01 ..
 4 -rw-r--r--  1 genmonpi genmonpi  2406 May 16 12:58 check_monitor_system.py
 8 -rw-r--r--  1 genmonpi genmonpi  7214 May 16 12:58 ClientInterface.java
12 -rwxr-xr-x  1 genmonpi genmonpi 10760 May 16 12:58 DIY2TankSensorCalibrate.py
 4 -rwxr-xr-x  1 genmonpi genmonpi  3301 May 16 12:58 kwlog2csv.py
16 -rw-r--r--  1 genmonpi genmonpi 12364 May 16 12:58 modbusdump.py
 8 -rw-r--r--  1 genmonpi genmonpi  5143 May 16 12:58 mopeka_utility.py
20 -rw-r--r--  1 genmonpi genmonpi 16757 May 16 12:58 serialconfig.py
 8 -rw-r--r--  1 genmonpi genmonpi  4362 May 16 12:58 serialtest.py
 4 -rw-r--r--  1 genmonpi genmonpi  2576 May 16 12:58 sockettest.py
genmonpi@HeadendGenMon:~/genmon/OtherApps$

jgyates, 2023-05-19:
there was a typo in the earlier message, it should be serialconfig.py  not seiralconfig.py

to restore:

     sudo python3 serialconfig.py -r

To setup serial for bluetooth type:

     sudo python3 serialconfig.py -e -b

To check if BT is setup type:

    sudo python3 serialconfig.py -c -b

msteil4vse, 2023-05-19:
Well, this is encouraging, looks like Bluetooth is finally happy.

sudo python3 serialconfig.py -c -b

Checking : Is enable UART in boot config: OK
Checking : BT alternate serial port: OK
Checking : Turbo Clocking: OK
Checking : Disable BT Disable Overlay: OK
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service enabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

genmonpi@HeadendGenMon:~/genmon/OtherApps$ sudo bluetoothctl
Agent registered
[bluetooth]# show
Controller B8:27:EB:EE:51:B6 (public)
        Name: HeadendGenMon.vsenterprisesltd.com
        Alias: HeadendGenMon.vsenterprisesltd.com
        Class: 0x002c0000
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0537
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
[bluetooth]# 

So, now where am I at with the sensor config?    also thinking about the next person that is trying to make this work and having similar problems, I am hoping to make the final post some sort of coherent steps for the next person.

jgyates, 2023-05-19:
regarding sensor config, follow the steps listed in the wiki related to the mopeka_utility.py program:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmopekapy-optional

I am adding a note regarding the restore step to the wiki for people who configured the serial port before bluetooth was configured.