jgyates/genmon #897: Genmon Mopeka - No Mopeka Fuel Tank Status Guage

bcrazycramer, 2023-05-18:
### Expected Behavior

Mopeka Fuel Guage on Status Page

### Actual Behavior

We had this remote working before replacing the propane tank for a larger size. Now when I go to status page I don't have the Mopeka fuel guage. 

### Steps to Reproduce (including precondition)

I have the Mopeka Add-On enabled with 120 Gallon Vertical tank and the mac address.  

### Screenshot or Pictures relating to the problem (if possible)
I've tried this with Safari and Google Chrome. I saw there was an update available so I applied the update. After rebooting I had the Mopeka Fuel Guage Sensor but it went away after rebooting the second time.

I see in the logs:

2023-05-18 14:06:00,603 : Error in ConvertTankReading: Input height is above top of tank : genmopeka.py:519
2023-05-18 14:08:29,554 : Error in ConvertTankReading: Input height is above top of tank : genmopeka.py:519
2023-05-18 14:20:04,443 : Error in ConvertTankReading: Input height is above top of tank : genmopeka.py:519
2023-05-18 14:21:51,873 : Error in ConvertTankReading: Input height is above top of tank : genmopeka.py:519
2023-05-18 14:29:43,332 : Error in ConvertTankReading: math domain error : genmopeka.py:519
2023-05-18 14:31:35,922 : Error in ConvertTankReading: Input height is above top of tank : genmopeka.py:519
2023-05-18 14:44:58,072 : Error in ConvertTankReading: Input height is above top of tank : genmopeka.py:522

### Your Environment

- Generator Model: G0067292
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.17

jgyates, 2023-05-18:
Have you submitted your logs?

when you replaced the propane tank for a larger size did you update the tank size on the mopeka add on settings?  Please make sure you are using the very latest version of the software by clicking the button to update the software on the about page.

your copy and past of the logs do not tell me if the add on is loading or if these errors are from a previous error as I can not compare the timestamps to the genmon.log start times. Do the timestamps on the log entries above occur before when  genmon is restarted or did they occur in the past?

jgyates, 2023-05-19:
I received you logs. A few comments:

1) it does look like the genmopeka add on is running because the entries in your genmonpeka.log file are later than the entries in genmon.log.
2) If the gauge is not showing up, first try refreshing your browser. 
3) The error " Input height is above top of tank" is caused by the sensor giving a measurement that exceeds the hight of the tank. This can occur if you select the wrong tank or if your sensor is not placed correctly. 

If you still do not see a gauge then send a picture of your add on settings.

bcrazycramer, 2023-05-19:
I tried a few different browsers and had my coworker try it. 

Will the gauge show up if it can't communicate with Mokpeka Sensor?

Ok that error was probably when I had it on the side of the tank trying to get the gauge to work. It's a little difficult to get it on the bottom of a 120 gallon vertical tank. 

[image]

jgyates, 2023-05-19:
if it can't find a sensor the add on will exit and the gauge will not show up. If the sensor returns invalid data the gauge will not show up. If one valid reading occurs then the add on will communicate that info to genmon and the sensor should be visible.

The data that the web user interface gets the info from the gauges from genmon. To view this data you can use this URL:

     http://192.168.1.10:8000/cmd/start_info_json

You would need to change the IP address to match your pi. This URL will return some data. You can post that data here.,

another trouble shooting step is to edit the file /etc/genmon/genmopeka.conf and add a line like this:

debug = True

Save the file and then restart genmon. This will allow the genmopeka add on to log additional data to the log file. You will not want to remove the debug = True once you have it working as over time this will reduce the life of your SD card.

jgyates, 2023-05-20:
I am going to close this issue for now but feel free to post here with any follow up status or questions.

closed 2023-05-20 by jgyates