jgyates/genmon #899: Pushover Stopped Working After Genmon Software Update

GitMoGen, 2023-05-29:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior
Pushover sends notifications

### Actual Behavior
Pushover no longer sending notifications

### Steps to Reproduce (including precondition)

So, I upgraded to the latest version of Genmon.  To do this, I needed to remove the pi from the generator and bring it into the shop.  I used a wall USB power supply and of-course got numerous voltage errors that pushover sent to my phone.  As I continued the upgrade, pushover seemed to stop sending notifications from the pi at around 255 from what the pushover account says.  After the upgrade, pushover doesn't work when I create an event on the generator that normally would create a pushover notification. 

I can go into my pushover account and send a test notification to my phone, so the app works but the pushover message generator on the pi doesn't seem to work.  

Besides upgrading to the latest version of Genmon, the external Wifi USB antenna on the Pi didn't work so went back to just the built in Wifi.  Other than that, nothing changed.  

How do I get the pushover to start sending events again?  Thanks in advance! 

### Screenshot or Pictures relating to the problem (if possible)

None

### Your Environment

- Generator Model: Generac 10kw
- Genmon Logs: Submit via the About page
- Genmon Version: latest version

jgyates, 2023-05-29:
I mistakenly checked in a typo in my last update that would cause this issue. I just fixed this (thanks to you reporting this issue) and checked it into the github repository. Try updating your software again. If it does not correct the issue, wait 30 min and try updating the software again as sometime it takes a few minutes for the updates to be available on all github mirror servers. Let me know if you have any issues or problems.

closed 2023-05-29 by jgyates

GitMoGen, 2023-05-29:
Thanks so much, will give it a try in about 30 minutes and let you know. Very appreciate the quick response.

GitMoGen, 2023-05-29:
So, unfortunately, still no push messages from the generator after I did the "Update Software" which took about 30 seconds to download per the screen and reboot.

I am open to additional suggestions. Thanks again!

Bryan

jgyates, 2023-05-29:
I assume when you say "Software Update" you are updating from the About page? If you are updating any other way then that will not restart the software. If you are updating from the About page then it is possible that github has not updated yet so you would just have to wait. One way to tell if this is the case is to look at the date on this file:

log into the pi via ssh and change directories into the genmon directory:

     cd genmon

then type this command:

    ls -la ./genmonlib/mynotify.py 

The date of this file should be todays day. If it is not then you do not have the latest.

If you do have a file with todays date and you updated form the About page (this will restart the software) and it still does not work then you would need to submit your logs on the about page (outbound email has to be working for this to work). If you do this I can examine your /var/log/genpushover.log file for errors.

If you do not have the latest version you will likely not see any errors in the log as the bug prevented the add on form loading.

jgyates, 2023-05-29:
also, this command will show all python scripts running:

     ps -fA | grep python

you should see an output like this:

     root      7823     1  5 22:23 pts/0    00:00:27 /usr/bin/python3 /home/pi/genmon/genmon.py -c /etc/genmon/
     root      7840     1  0 22:23 pts/0    00:00:01 /usr/bin/python3 /home/pi/genmon/genserv.py -c /etc/genmon/
     root      7842     1  0 22:23 pts/0    00:00:00 /usr/bin/python3 /home/pi/genmon/addon/genpushover.py -c /etc/genmon/
     pi       12476  4491  0 22:32 pts/0    00:00:00 grep --color=auto python

GitMoGen, 2023-05-29:
ok, the file is from two days ago so will wait a bit or morning :) to update and let you know.

Thanks again for the great software and support. Will let you know how it goes,

Bryan

GitMoGen, 2023-05-29:
All good and up and running after the update. Thanks again for the support and great software.

jgyates, 2023-05-29:
Thanks for the update. I am glad it is working.

hedgemybets, 2023-08-05:
This is just additional information that might help someone else. I had the same problem after upgrade to 1.18.18, though in my case the About page upgrade failed to load pyowm among other Python modules. I had not upgraded in a couple of years. Per the troubleshooting guide, I used genmonmaint.sh -i and -r and then startgenmon.sh restart to get things going again. 

Today I noticed that Pushover notifications were not working. There was an error in the logs: "signal_handler: Close() missing 1 required positional argument: 'self'". After seeing issue #920, I tried the genmonmaint.sh -u and then startgenmon.sh restart, but same error occurred in genpushover.log. A reboot of the Pi seems to have fixed it. I am using a Pi Zero W, so this error may be due to limited resources.

BTW, this has been quite a reliable system and great work from @jgyates. I recently made a donation!

jgyates, 2023-08-05:
@hedgemybets Thanks for the kind words and donation!

hedgemybets, 2024-01-08:
More info on the problem I had above.  After updating to 1.19.0 and 1.19.01, Pushover notifications stopped again and the error returned in the genpushover log: "signal_handler: Close() missing 1 required positional argument: 'self'" on every start or restart. In troubleshooting this, I believe that the Pushover add-on is starting up too quickly on my Pi Zero W, so I just added a line in the genloader.conf file at the end of the genserv module configuration:

postloaddelay = 10

The 10 second delay before loading the genpushover add-on seems to fix the problem. I only have the one add-on. Hope this helps.

jgyates, 2024-01-08:
I just checked in a minor mod to genpushover.py that adds a little more info in the log that may point to the source of the issue.

skipfire, 2024-01-08:
That fixed it for me.  The notifications were getting to the app but without a notification alert, I updated and it sent the alerts.

jgyates, 2024-01-09:
hmm. it is working for me when I test, but I don't use this plugin all the time. 

From what I am reading, it does not sound like you guys have the same symptoms? @hedgemybets, yours stops working, @skipfire, yours does not send an app pop up on your app?

What type of pi are you guys using and are you using the 64 bit OS? Are you running any other add ons?
 Trying to duplicate this a closely as possible.

skipfire, 2024-01-09:
Before you updated today it wasn't giving me the popup, however after todays update the popups returned so I'm good.

jgyates, 2024-01-09:
the only think I changed in the pushover add on today was some debug prints so this may be an intermittent issue.

hedgemybets, 2024-01-09:
For me, it's Pi Zero W Rev 1.1 running 32-bit Raspbian Stretch 9 with no
other add-ons.

In my case, I think that the processes are not fully up when genpushover.py
is launched. I have noticed that I can't get to the webpage for a number of
seconds after a start or restart and during that time the TCP listener is
not yet open (using netstat to confirm). What led me to my solution was
that after a restart, I was able to kill the genpushover.py process
manually and then launch it manually and pushover notifications worked fine
with no signal errors.

So then I added the 10 second delay and it seemed to allow genserv.py to
fully load and genpushover.py launched normally.

jgyates, 2024-01-09:
if you don't have it enabled already, on the settings page there is a setting named "Optimize for slower CPUs". This makes genmon be a little kinder to the CPU when doing modbus processing. With this enabled genmon will have a reduce CPU usage.