jgyates/genmon #900: Pi Zero W - takes about 7 minutes after boot for dial images to appear in web UI

wtnh, 2023-05-30:
Not really a bug, but it caused me to think I had a broken installation initially.

The dial images on the status page do not render in the web UI for approximately 7 minutes after a boot on a Raspberry Pi Zero W. The other pages appear to be OK shortly after a boot. After 7 minutes (or so) everything appears normal.

Wondering if the Pi Zero is performing some heavy-duty task (graphics or computation) that it is bogging down on.

jgyates, 2023-05-30:
the first thing to look at is your modbus communications (serial or TCP).

The seconds is: Do you have the option "Optimize for slower CPUs" enabled on the settings page? 

Marginal wifi may be the issue. When genmon starts, the software requires that all of the registers have been read at least once before it is considered to be finished with the initialization process as this is needed to identify the controller type. If you are seeing communication errors once you can see the gauges in the web interface then I would start by trying to reduce the startup time. The "Optimize for slower CPUs" setting will reduce the CPU usage for each modbus transaction. 

If you communications is not functional then there would be about 200 modbus communication attempts (around 50 normal registers plus 150 log registers) that would timeout before the initialization sequence is complete, each is about 3 seconds which puts you right at 6-7 minutes. 

You can view you communication stats on the Monitor page in the web interface.

jgyates, 2023-06-01:
I am going to close this issue. If you have any updates or questions feel free to post them to this thread.

wtnh, 2023-06-01:
Thanks - I was testing prior to actually connecting to the generator, so was puzzled by the lack of dials initially.

I have a PowerPact unit and need to modify the cables to bring +12 battery power to the pHat.

Assuming I am successful, I can provide some details on how to get Genmon working on a PowerPact (at least for my version which is about 6 years old).

Thanks

Whit

jgyates, 2023-06-01:
That would be great. Any info on how to get it working on specific generators is helpful. I can add that info to the wiki.

jgyates, 2023-06-10:
Thanks! this looks great. I will add this to the wiki

jgyates, 2023-06-10:
https://github.com/jgyates/genmon/wiki/Appendix-R---Replacing-Generac-MobileLink-with-Genmon-on-a-PowerPact-7.5-KW

wtnh, 2023-06-10:
Looks like a few of the diagrams did not make it over...

jgyates, 2023-06-10:
thanks for the sanity check. I had the wrong file name extension for the png files. it should be correct now. Let me know if you see anything else.

wtnh, 2023-06-10:
Looks good!

Thanks

Whit Turner

wtnh, 2023-06-10:
Opps - spoke too soon. Some of the hyperlinks did not make it over.

Whit

jgyates, 2023-06-10:
fixed. Hopefully I did not miss anything else :)

I added a link at the top of the main wiki page under the power pact bullet: https://github.com/jgyates/genmon/wiki

wtnh, 2023-06-10:
I think you nailed it. I know General has sold a ton of these PowerPact models, so I hope this is useful info for a few folks.

It goes without saying, thanks for all your great work on Genmon!

Whit

skipfire, 2023-06-10:
@wtnh your writeup says that the owner needs to verify the presense of the 8-pin Mini Fit Jr connection on the underside of the controller, but you also have content about modifying the existing 6-pin cable.  If there is an 8-pin connection on the controller, why does any work need to be done with a 6-pin connector?

wtnh, 2023-06-10:
The only reason to verify that the controller has an 8-pin MiniFit-type connector at the accessory port location is to ensure it is an Evolution-type controller. One user who thought he had a PowerPact had a different connector which was not likely to be compatible. If you think we should re-word this a bit, I would be happy to do so.

The article is about re-using the old MobileLink cable and only requires changing the 6-pin connector.

Perhaps it might be clearer if we just eliminate mention of the 8-pin connector on the accessory port and just assume it is Evolution-compatible.

Let me know your thoughts.

Whit

skipfire, 2023-06-11:
I think I'm just not following why the cabling is different for the PowerPact than other generators.  If it has an Evolution type controller with an 8-pin Mini Fit Jr, it seems like it should follow the same cabling needs unless it is trying to keep a 6-pin accessory connected as well.

wtnh, 2023-06-11:
In my case, it is not different. The 8-pin connector going to the controller remains intact. The only change is to replace the 6-pin connector going to the old MobileLink with an 8-pin which will plug into the pHat. The MobileLink modem gets tossed because it no longer works.

I am not aware of all the differences in Generac units. All I know about is my 7-year-old PowerPact 7.5KW and the fact that the new MobileLink units do not work with it. (Generac confirmed this with me - so there is definitely something different about the older PowerPact units).

Can you help me re-word this so it is clearer?

Thanks

Whit

skipfire, 2023-06-11:
With your description here it sounds like there should not be any cable modifications required since there's an 8-pin on the controller and an 8-pin on the HAT both Mini Fit Jr, with the 6-pin only being for an unusable MobileLink connector.  If the MobileLink connector was just ignored it seems like it may work without needing any custom cable work.

wtnh, 2023-06-11:
Ok - we could eliminate the reference to checking 8-pin on the controller if that would help. The only reason I had that in the writeup is there was one person who said he had a PowerPact and posted pictures of the controller which was clearly NOT an Evolution type.

Whit

skipfire, 2023-06-11:
I think checking for the 8-pin connector is valid to make sure it is likely to be compatible.  What I don't understand is why not just plug the 8-pin Molex cable in on both ends and be done without any custom cable work?

wtnh, 2023-06-11:
Because the controller in the PowerPact does not supply 12 Volts. A separate battery run is required, so the cable you supply would need to be modified as well. This would require cutting wires and adding ring terminals for the battery. I can copy the original MobileLink instructions and send them if that would clarify.
Whit

skipfire, 2023-06-11:
Ok, that explains it, thanks.  One of the other HATs may have been a better fit as they have a different cable to connect to the battery for power so no modifications are needed.  Thanks for helping me understand why the cable had to be customized.

wtnh, 2023-06-11:
This may also help to clarify things (for other use cases as well). What they appear to be saying is that any generator with a Nexus, Evolution (I assume version 1.0), Sync, or Sync 2 will NOT supply 12 volts through the 8-pin accessory connector, therefore requiring separate leads to the battery. Of course, it makes sense that these leads have an inline fuse for safety, which they provide in the harness.

They go on to say that Evolution 2.0 and Sync 3.0 controllers DO supply 12 volts through the 8-pin connector, therefore making the separate battery connection unnecessary.

In my case, my controller appears to be similar to Evolution 1.0, although Genmon reports it as "Power Pact Evolution", so probably a specific version for my vintage PowerPact - about 7 years old.

This is the installation page from the original MobileLink installation manual. Mine says it was published in 2016:

[image]

skipfire, 2023-06-11:
Thanks, the current generation of PintSize boards handle that with a specific cable for powering the unit from the starter battery so no custom cabling would be necessary, that was only needed since you purchased the legacy board (that power handling is the primary difference between the current and legacy boards).  I appreciate you helping me understand why you had to make the custom cable.