jgyates/genmon #901: Can't connect to Genmon with FF, Edge or Chrome

deanminn, 2023-06-01:
I started this in the wrong repository so I am re-posting the issue with responses to jgyates questions.

 I have a new build with Pi Zero W. I can not connect with any browser, I can not with ssh and can ping the device. I can also ping the device with CMD from Win 11. I have checked the logs in my router and do not see anything issue there.
These are the responses I get when I try to log on.
FF
We can’t connect to the server at genmon.local.
If you entered the right address, you can:
Try again later
Check your network connection
Check that Firefox has permission to access the web (you might be connected but behind a firewall)
Edge
Hmmm… can't reach this page
genmon.local refused to connect.
Try:
• Search the web for [genmon local](https://www.bing.com/search?form=ANLKDR&q=genmon%20local)
• Checking the connection
• Checking the proxy and the firewall
ERR_CONNECTION_REFUSED
Chrome
This site can’t be reached
genmon.local refused to connect.
Try:
• Checking the connection
• Checking the proxy and the firewall
ERR_CONNECTION_REFUSED
Any help would be appreciated. 

 **From jgyates:**
_I will need some more info to provide any suggestions.

What is the exact address you are using for the address in your browser? The default address is_
http://xxx.xxx.xxx.xxx:8000/

From the browsers I have used:
genmon.local:8000
http://10.68.193.16/
https://10.68.193.16/
None work
IP of the PI is 10.68.193.16
Where the x's are replaced with the IP address of the pi. In the message above it looke like you are using the domain name genmon.local. If you can ping genmon.local then try this address

http://genmon.local:8000/
I have tried this and still no connection

Genmon uses port 8000 for the web interface by default.

If you can not ssh into the pi then you need to make sure the SSH service is enabled on your pi. You can do this on the command line or with the graphical interface on the pi:

I can ssh into the PI and sign in with my password.
I can ping the IP from the CMD window in Win 11 and ping it from the terminal page.

 I have a Pi zero W with the most current Genmon software
Linux genmon 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l

jgyates, 2023-06-01:
The next step is to validate that genmon is started. To do this then log into the pi via ssh and then:

1) change directories into the genmon folder:

       cd genmon

2) then restart genmon

      ./startgenmon.sh restart

What is the output of the last command?

Can you access the web interface after you issue the commands above?

deanminn, 2023-06-01:
Not sure if I am using the cd genmon correctly but this is what I did and
the response.

***@***.***:~ $ cd genmon
***@***.***:~/genmon $ ./startgenmon.sh restart
Restarting genmon python scripts
Error in ExecuteCommandList  : b'E: Encountered a section with no Package:
header\nE: Problem with MergeList
/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages\nE:
The package lists or status file could not be parsed or opened.\n'
Error: Unable to run apt-get update. Retrying...
Error in ExecuteCommandList  : b'E: Encountered a section with no Package:
header\nE: Problem with MergeList
/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages\nE:
The package lists or status file could not be parsed or opened.\n'
Error: Unable to run apt-get update.
Installing cmake...
Error in ExecuteCommandList  : b'E: Encountered a section with no Package:
header\nE: Problem with MergeList
/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages\nE:
The package lists or status file could not be parsed or opened.\n'
Error: Unable to install cmake.
Stopping....
Stopping genserv.py
kill: (1718): No such process
Stopping genmon.py
kill: (1713): No such process
Starting....
Starting /home/deanm/genmon/genmon.py
Starting /home/deanm/genmon/genserv.py

jgyates, 2023-06-01:
based on the above errors my best guess is that you either:

1) have not when through the installation procedure for genmon

or 

2) have a corrupted file system or corrupted operating system.

The more likely one is number one. In your initial response you mention you have a new build, but did you go through the installation procedure? This page has the information:

    https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

To give a little more detail you can show the output of the /var/log/genloader.log file

     cat /var/log/genloader.log

If you do not run the install scripts then genmon will not start automatically. Based on the output above, this looks like what is happening. Also the install scripts will install required components that reside outside of the github repository.

deanminn, 2023-06-01:
>    1. have a corrupted file system or corrupted operating system.

Here is the output of the /var/log/genloader.log file

PS C:\Users\deanm> cat /var/log/genloader.log
cat : Cannot find path 'C:\var\log\genloader.log' because it does not exist.
At line:1 char:1
+ cat /var/log/genloader.log
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound:
(C:\var\log\genloader.log:String) [Get-Content], ItemNotFoundException
    + FullyQualifiedErrorId :
PathNotFound,Microsoft.PowerShell.Commands.GetContentCommand

I did follow the installation procedure. I will run it again to make sure.

jgyates, 2023-06-01:
you would need to run the command on the pi, not on a windows system

ssh into the the pi, then run

     cat /var/log/genloader.log

If you have not run the install scripts listed in the link above, I would do that first and then let me know your results.

deanminn, 2023-06-01:
Sorry, thought I was signed in through ssh. Here is the output of cat
/var/log/genloader.log

***@***.***:~ $ cat /var/log/genloader.log
2023-05-30 19:13:23,678 : Installing cmake...
2023-05-30 19:13:30,601 : Error in ExecuteCommandList  : b"E: Unmet
dependencies. Try 'apt --fix-broken install' with no packages (or specify a
solution).\n"
2023-05-30 19:13:30,607 : Error: Unable to install cmake.
2023-05-30 19:13:35,191 : Warning: required library pyotp not installed.
Attempting to install....
2023-05-30 19:14:56,682 : Warning: required library mopeka_pro_check not
installed. Attempting to install....
2023-05-30 19:15:41,383 : Warning: required library voipms not installed.
Attempting to install....
2023-05-30 19:16:24,929 : Warning: unable to find config file:
/etc/genmon/genloader.conf Copying file to /etc/genmon/ directory.
2023-05-30 19:16:24,987 : Warning: Missing entry: gencallmebot , adding
entry
2023-05-30 19:16:25,303 : Warning: Missing entry: gencallmebot
2023-05-30 19:16:25,352 : Running one time maintenance check
2023-05-30 19:16:58,659 : Copying genmon.conf to /etc/genmon/
2023-05-30 19:16:58,671 : Copying mymail.conf to /etc/genmon/
2023-05-30 19:16:58,681 : Copying gengpio.conf to /etc/genmon/
2023-05-30 19:16:58,690 : Copying gengpioin.conf to /etc/genmon/
2023-05-30 19:16:58,700 : Copying gensms.conf to /etc/genmon/
2023-05-30 19:16:58,709 : Copying mymodem.conf to /etc/genmon/
2023-05-30 19:16:58,719 : Copying genpushover.conf to /etc/genmon/
2023-05-30 19:16:58,729 : Copying genmqtt.conf to /etc/genmon/
2023-05-30 19:16:58,740 : Copying genslack.conf to /etc/genmon/
2023-05-30 19:16:58,750 : Copying genexercise.conf to /etc/genmon/
2023-05-30 19:16:58,760 : Copying genalexa.conf to /etc/genmon/
2023-05-30 19:16:58,770 : Copying gentankutil.conf to /etc/genmon/
2023-05-30 19:16:58,783 : Copying gentankdiy.conf to /etc/genmon/
2023-05-30 19:16:58,792 : Copying genemail2sms.conf to /etc/genmon/
2023-05-30 19:16:58,801 : Copying gensnmp.conf to /etc/genmon/
2023-05-30 19:16:58,811 : Copying gentemp.conf to /etc/genmon/
2023-05-30 19:16:58,820 : Copying gengpioledblink.conf to /etc/genmon/
2023-05-30 19:16:58,830 : Copying gencthat.conf to /etc/genmon/
2023-05-30 19:16:58,840 : Copying genmopeka.conf to /etc/genmon/
2023-05-30 19:16:58,850 : Copying gencustomgpio.conf to /etc/genmon/
2023-05-30 19:16:58,859 : Copying gensms_voip.conf to /etc/genmon/
2023-05-30 19:16:58,870 : Copying gencallmebot.conf to /etc/genmon/
2023-05-31 21:36:20,788 : Error in ExecuteCommandList  : b'E: Encountered a
section with no Package: header\nE: Problem with MergeList
/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages\nE:
The package lists or status file could not be parsed or opened.\n'
2023-05-31 21:36:20,801 : Error: Unable to run apt-get update. Retrying...
2023-05-31 21:36:28,949 : Error in ExecuteCommandList  : b'E: Encountered a
section with no Package: header\nE: Problem with MergeList
/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages\nE:
The package lists or status file could not be parsed or opened.\n'
2023-05-31 21:36:28,955 : Error: Unable to run apt-get update.
2023-05-31 21:36:28,962 : Installing cmake...
2023-05-31 21:36:33,921 : Error in ExecuteCommandList  : b'E: Encountered a
section with no Package: header\nE: Problem with MergeList
/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages\nE:
The package lists or status file could not be parsed or opened.\n'
2023-05-31 21:36:33,927 : Error: Unable to install cmake.
2023-05-31 21:38:33,916 : Error in ExecuteCommandList  : b'E: Encountered a
section with no Package: header\nE: Problem with MergeList
/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages\nE:
The package lists or status file could not be parsed or opened.\n'
2023-05-31 21:38:33,924 : Error: Unable to run apt-get update. Retrying...
2023-05-31 21:38:45,950 : Error in ExecuteCommandList  : b'E: Encountered a
section with no Package: header\nE: Problem with MergeList
/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages\nE:
The package lists or status file could not be parsed or opened.\n'
2023-05-31 21:38:45,958 : Error: Unable to run apt-get update.
2023-05-31 21:38:45,967 : Installing cmake...
2023-05-31 21:38:50,953 : Error in ExecuteCommandList  : b'E: Encountered a
section with no Package: header\nE: Problem with MergeList
/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages\nE:
The package lists or status file could not be parsed or opened.\n'
2023-05-31 21:38:50,961 : Error: Unable to install cmake.

jgyates, 2023-06-01:
In addition to not having run the install script, you have some unusual errors in you log, namely " Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify asolution)"

If you can not run the first two instructions in the install procedure

     sudo apt-get update
     sudo apt-get install git

Then try running:

      sudo apt --fix-broken install

and see if that resolves your issues.

deanminn, 2023-06-01:
I am going to start over with a new install of Raspberry Pi OS on a new
card and then install Genmon from the instructions.

Thanks for your help and input.

jgyates, 2023-06-01:
I am glad it is working for you. Thanks for the status update.

closed 2023-06-01 by jgyates