jgyates/genmon #902: Can not connect to Genmon with browsers

deanminn, 2023-06-02:
I had Genmon working then tried to clone the sd and lost genmon so I just reinstalled it on the sd. V1.18.17
I now cannot connect to genmon through any browsers again.  FF, Edge or Chrome
I am using http://10.68.193.16:8000/ or genmon.local:8000
I can ssh into the Pi zero W
I can ping 10.68.193.16 in the terminal window and also can ping it using the cmd window in Win 11.

I have restarted genmon with the following
 /startgenmon.sh restart
And get the following
deanm@genmon:~ $ cd genmon
deanm@genmon:~/genmon $ ./startgenmon.sh restart
Restarting genmon python scripts
Running one time maintenance check
Stopping....
Starting....
Starting /home/deanm/genmon/genmon.py
Starting /home/deanm/genmon/genserv.py

Ran this command, cat /var/log/genloader.log this is the output

deanm@genmon:~ $ cat /var/log/genloader.log
2023-06-02 09:51:03,433 : Warning: Missing entry: gencallmebot , adding entry
2023-06-02 09:51:03,744 : Warning: Missing entry: gencallmebot
2023-06-02 09:51:03,793 : Running one time maintenance check

Ran:
deanm@genmon:~ $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done

Ran:
deanm@genmon:~ $ sudo apt-get install git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.30.2-1+deb11u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Ran:
deanm@genmon:~ $ sudo apt --fix-broken install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Ran cat /var/log/genloader.log again and received the following

deanm@genmon:~ $ cat /var/log/genloader.log
2023-06-02 09:51:03,433 : Warning: Missing entry: gencallmebot , adding entry
2023-06-02 09:51:03,744 : Warning: Missing entry: gencallmebot
2023-06-02 09:51:03,793 : Running one time maintenance check

Appears that I am missing gencallmebot 
Is this causing me not to be able to connect

jgyates, 2023-06-02:
I am linking your past issue for the benefit of others: https://github.com/jgyates/genmon/issues/901

No, the log entry about gencallmebot is normal. This is newly added add on program and it is just letting you know that new entries are added to genloader.conf.

Everything above looks normal. Have you made any changes to the file /etc/genmon/genmon.conf or is this just a plain install from the genmonmaint.sh script? 

when the install script ran, did you select 'yes' to have genmon start when the pi boots?

You mentioned that you ran this:

     deanm@genmon:~ $ cd genmon
     deanm@genmon:~/genmon $ ./startgenmon.sh restart
     Restarting genmon python scripts
     Running one time maintenance check
     Stopping....
     Starting....
     Starting /home/deanm/genmon/genmon.py
     Starting /home/deanm/genmon/genserv.py

If you did not reboot after installing (which is required if you made any changes to the serial port configuration) then the above may be normal. If you did reboot then this appears that genmon was not running already, otherwise you would get some output after "Stopping"

Also, if your pi is not connected to your controller you can expect a delay in the browser fully connecting (it will just show a spinning arrow while waiting on genmon to fully initialize).

deanminn, 2023-06-02:
I did not make any changes to the script, plain install.
I did select to have genmon start when Pi boots.
I just did a reboot. I can ping genmon, when I try to connect with FF get this-Firefox can’t establish a connection to the server at genmon.local:8000. With Edge I get ERR_CONNECTION_REFUSED.

I am not connected to the controller but I am not even getting to the spinning arrow.

I just ran these command:

deanm@genmon:~ $ cd genmon
deanm@genmon:~/genmon $ ./startgenmon.sh restart
Restarting genmon python scripts
Stopping....
Stopping genserv.py
kill: (1167): No such process
Stopping genmon.py
kill: (1162): No such process
Starting....
Starting /home/deanm/genmon/genmon.py
Starting /home/deanm/genmon/genserv.py
deanm@genmon:~/genmon $ ssh genmon.local
deanm@genmon.local's password:
Linux genmon 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jun  2 11:38:27 2023 from 10.68.193.16
deanm@genmon:~ $

jgyates, 2023-06-02:
Your latest output shows this:

deanm@genmon:/genmon $ ./startgenmon.sh restart
Restarting genmon python scripts
Stopping....
Stopping genserv.py
kill: (1167): No such process
Stopping genmon.py
kill: (1162): No such process

This tells me that genloader attempts to start genmon, but for some reason it did not start or exited on errors.

Let's look at the file /var/log/genmon.log

And also try this command from the ssh console:

    cd genmon
    sudo python3 genmon.py

This will attempt to start genmon, if there are errors then it will show them to the console output. If genmon does start you can type control + C to stop the above command.

jgyates, 2023-06-02:
This will show the contents of /var/log/genmon.log

     cat /var/log/genmon.log

deanminn, 2023-06-02:
This is what I get with /var/log/genmon.log

***@***.***:~ $ /var/log/genmon.log
-bash: /var/log/genmon.log: Permission denied

This is what I get with

cd genmon
sudo python3 genmon.py

***@***.***:~/genmon $ sudo python3 genmon.py
Error on open serial port /dev/serial0: [Errno 2] could not open port
/dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
Error on serial port init!
Error opening modbus device.

deanminn, 2023-06-02:
***@***.***:~/genmon $ cat /var/log/genmon.log
2023-06-02 09:52:13,568 : New version detected: Old = V1.18.17, New =
V1.18.17
2023-06-02 09:52:13,651 : Selected Controller: generac_evo_nexus
2023-06-02 09:52:13,733 : Error opening modbus device: Error opening modbus
device. : generac_evolution.py:326
2023-06-02 09:52:13,735 : Possible cause is invalid serial port specified.
2023-06-02 11:23:02,491 : Selected Controller: generac_evo_nexus
2023-06-02 11:23:02,643 : Error opening modbus device: Error opening modbus
device. : generac_evolution.py:326
2023-06-02 11:23:02,645 : Possible cause is invalid serial port specified.
2023-06-02 11:26:06,152 : Selected Controller: generac_evo_nexus
2023-06-02 11:26:06,314 : Error opening modbus device: Error opening modbus
device. : generac_evolution.py:326
2023-06-02 11:26:06,315 : Possible cause is invalid serial port specified.
2023-06-02 11:34:00,435 : Selected Controller: generac_evo_nexus
2023-06-02 11:34:00,503 : Error opening modbus device: Error opening modbus
device. : generac_evolution.py:326
2023-06-02 11:34:00,505 : Possible cause is invalid serial port specified.
2023-06-02 11:36:19,702 : Selected Controller: generac_evo_nexus
2023-06-02 11:36:19,740 : Error opening modbus device: Error opening modbus
device. : generac_evolution.py:326
2023-06-02 11:36:19,742 : Possible cause is invalid serial port specified.
2023-06-02 11:38:59,946 : Selected Controller: generac_evo_nexus
2023-06-02 11:38:59,984 : Error opening modbus device: Error opening modbus
device. : generac_evolution.py:326
2023-06-02 11:38:59,985 : Possible cause is invalid serial port specified.
2023-06-02 12:06:04,102 : Selected Controller: generac_evo_nexus
2023-06-02 12:06:04,163 : Error opening modbus device: Error opening modbus
device. : generac_evolution.py:326
2023-06-02 12:06:04,165 : Possible cause is invalid serial port specified.

jgyates, 2023-06-02:
Based on the output, genmon is not loading because there is no serial port configured. Are you running a raspberry pi? Did you select the option to setup the serial port in the install script? Did you reboot after the script completed?

You can view your serial ports on a pi by typing this:

      ls -la /dev/ser*

deanminn, 2023-06-02:
I am using a Pi zero W

Do not recall if I enabled the serial port.

I believe that I reboot.

***@***.***:~/genmon $ sudo ls -la /dev/ser*
lrwxrwxrwx 1 root root 7 Jun  2 11:24 /dev/serial1 -> ttyAMA0

deanminn, 2023-06-02:
Can I enable serial port using sudo raspi-config?

jgyates, 2023-06-02:
yes, but you are better off using the program serialconfig.py (that the setup program calls) as you will also need to disable some other stuff to make it work properly as the default pi config enables the bluetooth port to use the serial port. To use serialconfig.py from a ssh console type:

       cd genmon
       cd OtherApps
       sudo python3 serialconfig.py -e

More info on serialconfig.py is found here:  https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsserialconfigpy-optional

Once you enable the serial port and reboot you can validate it with the command above "ls -la /dev/ser*". Genmon uses /dev/serial0 by default. If it is not found then it will exit. That is what is happening to your installation now.

deanminn, 2023-06-02:
This is the output I received:
***@***.***:~ $ sudo ls -la /dev/ser*
lrwxrwxrwx 1 root root 7 Jun  2 12:43 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Jun  2 12:43 /dev/serial1 -> ttyS0

I am able to connect with the browsers.

Thanks again

jgyates, 2023-06-02:
Excellent. I am glad it is working for you.

closed 2023-06-02 by jgyates