jgyates/genmon #903: Remote Start Not Working

jarocht, 2023-06-02:
### Expected Behavior

Press "Start Generator" in UI, generator starts

### Actual Behavior

Generator does not start, no warning or error is shown

### Steps to Reproduce (including precondition)

Generator is in manual mode -- does it need to be in automatic for this feature to work?

### Screenshot or Pictures relating to the problem (if possible)

N/A

### Your Environment

- Generator Model: Evolution 2.0, 18KW
- Genmon Logs: N/A
- Genmon Version: Newest -- installed today

jgyates, 2023-06-02:
Manual mode will start in the generator if selected. It will remain in manual mode as long as it is running.

The Off mode will stop a manual start and the engine will remain off. Off will also remove the generator from Auto mode. If memory serves, remote commands will not work in Off mode. That is function of the firmware.

closed 2023-06-03 by jgyates

jarocht, 2023-06-07:
Documenting for others who may come across this:

For a Evolution 2.0 controller:
Remote start & remote start, including with transfer switch, only works in automatic mode, with mains power connected.

Remote start & stop will not work if the following conditions are true:
- mains power is disconnected, in any mode
- the generator is in 'off'
- the generator is running in 'manual'

@jgyates -- it may be helpful to add this to official docs.

jgyates, 2023-06-07:
Thanks for the detailed update. I will add this to the wiki.

There was a request to make an add on that would allow for a software controlled throttling of the generator during a power outage. I don't think the internal firmware commands for start and start / transfer are good for much of anything except testing the transfer switch and engine.

jgyates, 2023-06-07:
https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmonpy-required