jgyates/genmon #904: Not communicating with generator

deanminn, 2023-06-02:
Pi Zero W, new install of genmon.

Pi is connected to the controller. I can connect to the Pi with ssh at http\\10.68.193.16 or genmon.locaI:8000. I

 know the cabling is good because it has connected and communicated in the last day or so with the generator.

Showing errors in this log
cat /var/log/myserial.log
2023-06-02 12:06:04,139 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:125

Showing errors in this log
cat /var/log/genmon.log
2023-06-02 14:22:56,252 : WARNING: Initialization not complete after 30 seconds, possible communication failure. Check cabling.
2023-06-02 14:49:32,057 : Generator Monitor Shutdown
2023-06-02 14:50:52,218 : Selected Controller: generac_evo_nexus
2023-06-02 14:51:16,361 : GenMon Loaded for site: River Home using python 3.9
2023-06-02 14:51:48,617 : WARNING: Initialization not complete after 30 seconds, possible communication failure. Check cabling
 
 

- Generator Model: Generac 11 KW
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.17

jgyates, 2023-06-02:
First, please do not create a new issue every time. You have three issues in the lats 24 hours. I would suggest that you just keep posting to the existing issue even if the issue is marked closed. This allows others who may have similar issues to search the issues and follow the path that some other person took to resolve any similar issues. 

Second, the time stamps on the errors in your myserial.log and genmonlog are 2 hours apart. This indicates that the earlier one is form your previous issue of not running through the setup and properly configuring the serial port.

The errors in genmon.log indicate that you are not communicating with the controller. You can see details of the communication errors on the Monitor page of the web interface. I would guess that you have timeout errors. Timeout errors indicate that genmon is transmitting to the controller but receives nothing back. In most cases this is due to improper wiring of the cable or not providing the proper grounding between the controller and the pi. The ground between the controller and pi must be connected. This is covered in this page: https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

The template for each of the three issues you created have links for troubleshooting serial communication issues. Have you followed the steps in the recommended serial communications troubleshooting page? Here is the link for your reference:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Also, providing details on the history of your setup, for example did you build the cable? did you buy a cable? have you tested this with the loopback test as described in the link above?

If you have performed the troubleshooting steps and you have created your own cable then you can post a picture of your cable and this may help determine where you are going wrong.

deanminn, 2023-06-03:
I am not going to be able to get to this for a number of days. You can
close this issue.

Thanks

deanminn, 2023-06-07:
I was able to reprogram Raspberry Pi and Genmon and now things are working.

closed 2023-06-07 by deanminn