jgyates/genmon #905: Inspect Battery Alarm

jf5264835, 2023-06-03:
Generator is throwing warning for check battery, which seems to be getting interpreted as an alarm rather than a warning. The issue I'm having is that there seems to be no way to remotely clear this alarm condition, requiring an on-site visit to correct.

### Expected Behavior

Navigating to Maintenance>Alarm Condition and selecting "Reset Alarm" clears the inspect battery alert and removes the system alarm.

### Actual Behavior

Selecting "Reset Alarm" does not clear inspect battery condition.

### Steps to Reproduce (including precondition)

Precondition: Generator with alarm for "Inspect Battery"

1. Select "Reset Alarm"
2. ???
3. Profit?

### Your Environment

- Generator Model: G0070433
- Genmon Logs: Email not setup on genmon, can email to you if needed.
- Genmon Version: Version V1.18.16

jgyates, 2023-06-03:
Genmon does not care if it is an alarm or warning, it treats both the same.The firmware dictates the behavior of an alarm vs warning. 

I believe the "Inspect Battery" alarm is a timer based maintenance alarm, much like "Maintenance A Due" or "Maintenance B Due". If you have a liquid cooled controller you can try switching to off and back to Auto (liquid cooled Evolution models have the remote commands to do this). That will clear some alarms, the "Reset Alarm" command will clear others, but at the end of the day the firmware controls all of this. If you are looking for a remote way to do this, the "Reset Alarm" and "Off" command are the only two options that I know of for Evolution controllers and the "Off" only works for Liquid cooled Evolution.

I don't think that this alarm condition will dis-allow running of the power is out since it is a maintenance condition.

griffths, 2023-06-04:
My experience with the battery alert/alarm is that it must be more than a timer/reminderI have a liquid cooled Nexus Quietsource genset circa 2012I have reset the alarm quite often over the past few months while shopping for a replacement battery and it comes back at random times so I am guessing that the controller must be noticing something irregular about the battery accepting a charge and throwing an alert when it does.  Jay

jgyates, 2023-06-07:
"Inspect Battery" is timer based, i.e. it goes off after a certain period of time or engine hours.  "Low Battery" is not timer based and it is based on a low battery voltage reading by the firmware, i.e. low battery alarm is triggered after a battery voltage below a predetermined threshold for a predetermined time period.

closed 2023-06-07 by jgyates