jgyates/genmon #906: Failure to update GenMon

rwskinner, 2023-06-07:
GenMon doesn't update.  
About page shows - You are not running the latest version.
    Current Version: V1.18.16
    New Version: V1.18.17

Click Update, it says downloading for about 10 seconds, then reboots and the same version is listed.
Looking at the logs really didn't show me anything but I submitted them to you anyways.

Any ideas where to look next?

skipfire, 2023-06-07:
I had to log in with ssh and manually do a git pull, then restart genmon.

closed 2023-06-07 by jgyates

rwskinner, 2023-06-07:
#876 worked fine with no issues and no errors.

cd genmon
./startgenmon.sh stop
./genmonmaint.sh -u
./startgenmon.sh restart

jgyates, 2023-07-05:
FYI:   An update has been posted that may resolve this upgrade issue in the future. [See this thread for more info](https://github.com/jgyates/genmon/issues/920).