jgyates/genmon #908: Hardware Comment

craigwitter, 2023-06-13:
I just wanted to take a moment and say thanks for writing this awesome software. I also wanted to share a method that I used to hook it up. I, like most have had trouble finding a raspberry pi, I also didn't love the idea of this being outside on wifi.  Last weekend, I bought this RS232 device off Amazon (that does POE) https://www.amazon.com/dp/B0BWK9PJ5F?psc=1&ref=ppx_yo2ov_dt_b_product_details and wired it up to the port on the generator using your cable instructions. I then build a quick docker container for Genmon. It has been running perfectly.  I have three wires running to the RS232 device (ground, tx, and rx), and then a single POE network cable from inside the house.  Feel free to delete this if it's not appropriate, but thought it might help someone with a less expensive, possibly more reliable method then using a PI.  Thanks again!

jgyates, 2023-06-13:
@craigwitter, This is great info! Thanks. I will add this to the wiki.

Did you use the Serial TCP setting or the Modbus TCP setting?

jgyates, 2023-06-13:
I updated this page: https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP

craigwitter, 2023-06-13:
Just the serial setting.  Other settings are all defaults.

Thanks,
Craig

jgyates, 2023-06-15:
I ordered one of these and it works great. I prefer the setup on the waveshare product to the USR device. The only caveat in the setup I ran across is this:

1) the product wiki is the manual: https://www.waveshare.com/wiki/RS232_TO_POE_ETH_(B)
2) The default IP for me was 192.168.1.202. The wiki lightly implied the default was DHCP but mine was setup for a static IP. Once you have the IP you can log in to the device and set the serial parameters to 9600,N,8, and 1, then give genmon your port and IP address and it works great.
3) Looks like it is 100MB Ethernet, which is perfectly fine. My Ethernet cable to my generator is long-ish (~200ft) and it works great.

craigwitter, 2023-06-15:
Totally agree, finding the manual was the hardest part. The utility was
basic but worked to set the ip.

Thanks again for the awesome project.

closed 2023-06-19 by craigwitter

peterkeen, 2023-08-02:
Just wanted to chime in thank @craigwitter for the suggestion on the waveshare unit (and @jgyates for writing genmon in the first place!) I got the PoE model up and running today with genmon running in a docker container on a thin client in my basement. Took me a bit to figure out that I needed to swap tx and rx, but after that realization it's been working wonderfully.

Also, for the record, my device also came stock with static IP set to 192.168.1.202.

mstackler, 2023-10-24:
just a quick question on the waveshare unit and sockettest.  I have a waveshare unit and i am running socket test and it keeps coming back with error communicating timed out.  does it need to be connected to the generator to do this test?  i have the whole thing sitting on my desk and was trying to do all the tests before moving it outside.

mark

skipfire, 2023-10-24:
@mstackler are you referring to this HAT? https://www.waveshare.com/wiki/Serial_Expansion_HAT

If so, it looks like it doesn't use the standard serial ports so you would have to follow their setup scripts to get the Pi to recognize the I2C serial port, and then set the serial port manually in genmon or in the config file directly if the genmon UI isn't loading. The serial test script does accept an input parameter of the serial port in question so you can verify before editing in genmon's config.

mstackler, 2023-10-24:
sorry it is the waveshare rs-232 to eth adapter mentioned here https://www.waveshare.com/wiki/RS232_TO_POE_ETH_(B).  i have it programmed with an ip address and i can get to the web page to configure it from my pi on the same network.  when i run the socket test app it does turn on a bluye light on the wave share but does come back with that error so was thinking i needed it all connected to the generator to function properly

skipfire, 2023-10-24:
Do you have a loopback wire in place on the waveshare so that it has data to send back?  The socket test app is a loopback test and cannot be run when connected to the generator, I'm not sure how it handles serial over TCP though.

mstackler, 2023-10-24:
Adding the loopback wire between tx and rx solved the issue.   Thank you

jgyates, 2026-02-11:
Hi @jkosharek 

For Nexus, Evo1 and Evo2 you want Modbus RTS over TCP/IP.

Yes, transparent.

https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP

jgyates, 2026-02-12:
Great, I am glad it is working