jgyates/genmon #910: USB to DB9 for H-100

CSRetire, 2023-06-19:
### Your Environment

Residental Generac 100KW with H-100 controller

Is it possible to use a DB9 to USB for serial connection to the Pi?

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2023-06-19:
yes, you will just need to use the USB device name on the "Port for Serial Communication" parameter. On a raspberry pi, this is something like /dev/ttyUSB0

Once you plug in your USB serial device you should be able to type this command:

     ls -la /dev/ttyUSB*

and it will list all the USB serial ports. To make sure the operation system is finding your USB device that is plugged in, this command will show USB devices, but not the port name of the serial port:

     lsusb

jgyates, 2023-06-19:
I am going to close this for now but feel free to post related questions to this thread.

closed 2023-06-19 by jgyates

jgyates, 2023-06-20:
@CSRetire wrote:

I think I followed the direction.

chris@raspberrypi:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
chris@raspberrypi:~ $ ls -l /dev/serial*
lrwxrwxrwx 1 root root 7 Jun 20 18:22 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Jun 20 18:22 /dev/serial1 -> ttyS0

/dev/serial:
total 0
drwxr-xr-x 2 root root 60 Jun 20 18:22 by-path

In the genmon app, I used the address of: dev/serial1

That disconnected the xxx.xxx.xx.xx:8000. It will not re-connect. I do still have ssh access.

jgyates, 2023-06-20:
you typed:

     ls -la /dev/serial*

you should type:

      ls -la /dev/ttyUSB*

if you have a device named /dev/ttyUSB0 or /dev/ttyUSB1 then you should use the one of these names in genmon instead of /dev/serial1

CSRetire, 2023-06-21:
Thank you for the quick reply.  I sent some $ for your work, hope it helps.  It is hard to believe your dedication, thank you.

The one change made to “serial1” would not allow re-connection thru :8000, no matter how many reboots.  I wiped and reinstalled.  I was able to connect thru 8000 but still no joy or dials showing.  I have the Generac manufactured cable to allow serial connection to a tech computer.  I put that in between the H-100 and the db9 to usb connector ( it has lights showing data flow for all 9 pins and power from USB).  Prior to insertion, no flow but power.  After insertion, I have steady lights on DTR and RTS, plus power from USB however, there is no data flowing.

Any ideas?

Thanks

Chris

The information contained in this e-mail and any accompanying documents is confidential, may be privileged, and is intended solely for the person and/or entity to whom it is addressed (i.e. those identified in the "To" and "cc" box). They are the property of Kova Corporation. Unauthorized review, use, disclosure, or copying of this communication, or any part thereof, is strictly prohibited and may be unlawful. If you have received this e-mail in error, please return the e-mail and attachments to the sender and delete the e-mail and attachments and any copy from your system. Kova Corporation thanks you for your cooperation.

jgyates, 2023-06-21:
Hi Chris,

Thanks for the donation! 

If you have not done so already, to use genmon with the H-100 you need to make two changes to the software settings. This page describes both the changes:

https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

You can make both of these from the Advanced Settings page in the web interface. To access the Advanced Settings page, double click the gear icon in the upper right of the web interface. You need to change the "Modbus slave address" setting and the "Controller Type" setting and then save your settings at the bottom of the page. Saving the settings will restart genmon. Let me know how it goes.

CSRetire, 2023-06-21:
Thank you for the success.  It is now working as expected (lots untested yet).  At the risk of being business like, can you tell me your background?  I run a small company that does some great things.  I am guessing you have a strong (background/education) in Linux?

Thanks

Chris

https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

CSRetire, 2023-06-21:
Spoke too soon.  Now it is not responding after a reboot.

https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

jgyates, 2023-06-21:
If it is not working after a reboot then it is probably due to the the installation script not being executed. The install script ask if you want to start the software on reboot. If you select yes then it will add an entry in the system start up scripts (crontab) that will automatically start the sofware once the system starts. If you want to run the installation script to correct this issue be advised that it will ask you a few questions. One of the questions is "Start genmon on boot? (y/n)?", select Y for this. One of the other questions is "Copy configuration files to /etc/genmon?", select N for this as you have already setup genmon and if you select Y then it will overwrite you config files and you will have to update your settings again.

To run the install script log into the pi via ssh and then type:

     cd genmon
     bash ./genmonmaint.sh -i

The script will run and install the needed libraries. once the script is finished runing you can reboot your system an see if genmon is running after your pi boots. 

Also, there is 30 second delay between the system booting and genmon starting.

Let me know if you have any other questions or if that does not solve your problem.

CSRetire, 2023-06-21:
Followed the steps.  Still no joy.

jgyates, 2023-06-21:
If you ssh into the pi, what is the output of this command:

      sudo crontab -l

also, what is the output of this command:

     cat /var/log/genmon.log

You can also start genmon manually to see if it will display any errors:

 get into the genmon directory:

         cd genmon

then type:
     
        ./startgenmon.sh restart

Also copy and past the output of the above startgenmon command.

CSRetire, 2023-06-21:
***@***.***:~ $ sudo crontab -l
@reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start
***@***.***:~ $ cat /var/log/genmon.log
2023-06-21 02:58:44,714 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:58:47,747 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:58:50,776 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:58:53,804 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:58:56,828 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:58:59,854 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:02,877 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:05,905 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:08,933 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:11,960 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:14,990 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:18,021 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:21,049 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:24,074 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:27,100 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:30,130 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:33,154 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:36,178 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:39,204 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:42,232 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:45,263 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:48,286 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:51,311 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:54,339 : Unable to ID controller, possiby not receiving data.
2023-06-21 02:59:57,368 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:00,410 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:03,443 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:06,469 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:09,492 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:12,517 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:15,549 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:18,578 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:21,604 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:24,636 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:27,666 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:30,691 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:33,715 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:36,746 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:39,772 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:42,801 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:45,834 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:48,864 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:51,896 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:54,923 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:00:57,948 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:00,975 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:04,000 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:07,027 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:10,056 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:13,088 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:16,119 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:19,146 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:22,173 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:25,201 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:28,223 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:31,249 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:34,281 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:37,313 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:40,337 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:43,362 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:46,392 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:49,422 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:52,449 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:55,478 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:01:58,507 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:02:01,537 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:02:04,565 : Unable to ID controller, possiby not receiving data.
[... 92 lines cut ...]
[... cut ...]
2023-06-21 03:07:22,532 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:07:25,558 : Unable to ID controller, possiby not receiving data.
2023-06-21 03:07:28,588 : Unable to ID controller, possiby not receiving data.
***@***.***:~ $ cd genmon
***@***.***:~/genmon $ ./startgenmon.sh restart
Restarting genmon python scripts
Stopping....
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
***@***.***:~/genmon $

CSRetire, 2023-06-21:
Sorry just sent verbatim.

jgyates, 2023-06-21:
It looks like genmon is starting as it should on boot, however the error in the log "Unable to ID controller, possiby not receiving data." indicates that your serial communication between the pi and the controller is not working. 

I would recommend checking to make sure your cable is securely connected to both the pi and the controller.

Here are some steps for troubleshooting serial communication issues.

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Also, since you are using USB, validate that the USB serial device is still connected and active. 

        ls -la /dev/ttyUSB*

and

       lsusb 

Other things to check are:

1) make sure you have a ground connection between the controller and your serial interface
2) Make sure your Pi has sufficient power. This command:

       vcgencmd get_throttled

should output "throttled=0x0" if the pi's power is OK, otherwise it will report a non zero value. Serial USB devices typically do not take a lot of power, but a less than adequate power source will cause issues with external devices like USB.

CSRetire, 2023-06-21:
As requested:

***@***.***:~/genmon $ ./startgenmon.sh restart
Restarting genmon python scripts
Stopping....
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
***@***.***:~/genmon $

CSRetire, 2023-06-21:
throttled=0x0

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

3.  vcgencmd get_throttled

CSRetire, 2023-06-21:
***@***.***:~/genmon $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

CSRetire, 2023-06-21:
***@***.***:~/genmon $ ls -la /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 Jun 21 03:37 /dev/ttyUSB0

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

CSRetire, 2023-06-21:
I’m guessing that Generac has code that changes based on their config variables

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

jgyates, 2023-06-21:
Based on the data I see my best guess is that the problem is with the serial cable / connection. Going through the steps on [this page](https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting) is your best first step in resolving this issue.

A second guess would be that the modbus slave address is not correct or has changed but you mentioned that it was working before so to me this appears to be less likely. The H-100 has the ability to change the modbus address used but the default is 64 hex (100 decimal). Both the controller and genmon must be set to the same modbus address.

The web interface will not show anything until genmon has finished initializing. To finish initializing genmon must attempt to read all registers at least once. Since serial comms are not working it should take several minutes due to serial timeouts occurring. Once the web interface is up you can go to the monitor page and view the serial comm stats. If you post the serial comm stats it may give more insight to the problem.

CSRetire, 2023-06-25:
Hi again,

Finally have narrowed the issue.  The serial port on the H-100 is not responding.  Using serialtest with genmon not running and pins 2+3 jumpered together, the com port passes.  One plugged into the H-100, it fails, under dev/ttyusb0.

A solid copper #8 was brought from the negative terminal and used as a ground to the raspberry pi (pin 20), the H-100 panel (just in case), and the common surfaces of the cables.

The method shown under issues using serial0 was also tried without success.

Thinking it may be low voltage (even though it returned a 0x0), the voltage for that device was changed from 3.3 to 5v, without success.  For each device the port was changed from serial0 to ttyusb0

Is there another trick to coax the H-100 to actually communicate?

Thanks,

Chris

    Communication Stats :
        Packet Count : M: 666, S: 0
        CRC Errors : 0
        CRC Percent Errors : 0.00%
        Timeout Errors : 665
        Timeout Percent Errors : 99.85%
        Modbus Exceptions : 0
        Validation Errors : 0
        Sync Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 401473
        Packets Per Second : 0.33
        Modbus Transport : Serial
        Serial Data Rate : 9600

jgyates, 2023-06-25:
One thing to try is to reverse the tx and rx pins. Some equipment label the pins as to their function and others label the pins as to what they should be connected to. Loopbacks will work even if Tx and Rx are reversed so the only way to tell if this is the case is to just try it out. Most RS-232 tranceivers have protections to handle this case. 

You may also want to check the communication settings on the H100. I can't recall if you can set the rate to some other value, but genmon will communicate a 9600, no parity, 8 data bits and 1 stop bit (9600,N,8,1).

liltux, 2023-06-25:
@CSRetire i have recently ran into issues with the H-panel coms.  What I have, for now, found is that when the communication is stopped all of a sudden as in a power loss or unplugging the cable cause the H-100 to lock up the port.  I have recently had to power cycle the H-100 to get comms back up when not stopping the program before shutting down the pi.  Might try to power cycle the h-100 controller by removing the F2 fuse behind the H-PANEL.

CSRetire, 2023-06-25:
It is set at 9600,8,1,N.  On the serial I tried with reversed pins in the pi.  On the usb I tried a null modem cable (Generac’s) as well.  Still no activity.

CSRetire, 2023-06-26:
Good afternoon,

Thank you for all of your help.  The final piece was the resetting of the F2 fuse.  Everything started working correctly.

I am absolutely certain I would have never found that solution, thank you again for all of the assistance and wonderful coding to make this available.

My next project is using AI to parse out particular data from a speech to text transcription.  I need six fields populated.  The text is no more that 250 words.  Sometimes the data needed is very structured (always appearing in the same spot, just after a key word.  Other times it appears randomly.  It is to be used for public safety so the accuracy must be very high.

Any thoughts or friends that are expertise in this field?

Thank you again.

Chris

mstackler, 2023-11-06:
> @CSRetire i have recently ran into issues with the H-panel coms. What I have, for now, found is that when the communication is stopped all of a sudden as in a power loss or unplugging the cable [...]

I ran into this issue with my industrial 300kw generac running the H-100 controller.  I would connect genmon as normal and it would work.  I found if i made any changes to the genmon software it would lock up the H-100 controller and would need to pull the fuse and reset.  I now disconnect the serial port if i need to make a genmon change

mark

CSRetire, 2023-11-06:
It happens so frequently, I am going to install a N/C momentary switch on the outside of the panel, to make resets without having to pull the fuse each time.

jgyates, 2023-11-06:
There are three people that I know of using the H-100 with an issue with the controller locking on restart. After some thinking I got to looking at the code and I made a modification to the way that the modbus module was shut down. This change should prevent a shutdown in the middle of a modbus packet. If you update from the about page you can try out this update and see if it avoids the problem if the H100 controller locking. Fingers crossed..

Let me know how it goes.

mstackler, 2023-11-07:
> There are three people that I know of using the H-100 with an issue with the controller locking on restart. After some thinking I got to looking at the code and I made a modification to the way that [...]

I think you fixed it.  I have made several changes to genmon and have not needed to reset the controller

mark

jgyates, 2023-11-07:
Great! Thanks for testing this.

mstackler, 2023-11-10:
I tried to add email recipients to the notifications tab and it locked out
again.  I will try and reset it again today.

mark

jgyates, 2023-11-10:
are you using serial or serial over IP? Trying to find some kind of commonality so any other details about your setup you can share may help. RS-232 or RS-485, etc

mstackler, 2023-11-10:
Using serial gpio rs232 heading to the site now. Let me know what you want
me to look at.  I can also try serial over ip I have a wave share box

Mark

jgyates, 2023-11-10:
I would be interesting to see how repeatable this was with serial. Also if it occurs with serial over IP.

At the moment the only thing that I can think to add would be an additional delay for the H100 on closing the serial port.

jgyates, 2023-11-10:
I checked in an update that changes the order of how things are shutdown and added a half second delay for H100 before the serial device is closed. If you can see how repeatable it is with direct serial before applying this update, then try with the update.

mstackler, 2023-11-10:
i have made several changes and it did not lock up again.  going to try the
update and will just keep monitoring it

mstackler, 2023-11-10:
and the update bricked the controller again.

jgyates, 2023-11-10:
did updating to the last change brick it or did it brick after the update was finished and a change was made? If it was the former then the update may still work as it would not of been applied yet (the restart occurs on the old code).

mstackler, 2023-11-10:
It bricked after the update and restart.   It is now running 1.19 and is stable. Installing a second site next week and perform more testsMarkSent from my Mark's  iPhone