jgyates/genmon #911: yes, you will just need to use the USB device name on the "Port for Serial Communication" parameter. On a raspberry pi, this is something like /dev/ttyUSB0

CSRetire, 2023-06-20:
yes, you will just need to use the USB device name on the "Port for Serial Communication" parameter. On a raspberry pi, this is something like /dev/ttyUSB0

Once you plug in your USB serial device you should be able to type this command:

     ls -la /dev/ttyUSB*

and it will list all the USB serial ports. To make sure the operation system is finding your USB device that is plugged in, this command will show USB devices, but not the port name of the serial port:

     lsusb

_Originally posted by @jgyates in https://github.com/jgyates/genmon/issues/910#issuecomment-1597768846_

closed 2023-06-20 by jgyates

CSRetire, 2023-06-20:
I think I followed the direction. 

chris@raspberrypi:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
chris@raspberrypi:~ $ ls -l /dev/serial*
lrwxrwxrwx 1 root root  7 Jun 20 18:22 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root  5 Jun 20 18:22 /dev/serial1 -> ttyS0

/dev/serial:
total 0
drwxr-xr-x 2 root root 60 Jun 20 18:22 by-path

In the genmon app, I used the address of:  dev/serial1

That disconnected the xxx.xxx.xx.xx:8000.  It will not re-connect.  I do still have ssh access.

jgyates, 2023-06-20:
my response is in the original thread: https://github.com/jgyates/genmon/issues/910

mrjware, 2023-06-21:
I also had similar issues with a USB to serial console cable I had laying around. In the web GUI I had entered a USB port that didn't exist and that caused the web interface to not start so the solution was to run the install script a second time and that fixed it keeping in mind that it takes 30 to 40 seconds for the wed interface to start after the pi reboots. The trick was adding /dev/ttyUSB0 as the port in the web gui. Then I ran "sudo python3 /home/pi/genmon/OtherApps/serialconfig.py -e " to ensure UART was set as it wanted. From that point, it was just a matter of running the serial loopback test and making sure that my transmit and receive cables were correct once I plugged it in directly. I monitored the web interface face that shows the CRC errors, and as I had many stacking up with no responses from the slave, I just flip-flopped transmit and receive cables.  Poof , worked.