jgyates/genmon #912: Twilio A2P 10DLC Registration

mxplea10, 2023-06-21:
This is not an issue with genmon but asking for advice and recommendations with using Twilio.  I keep getting a message that "A2P 10DLC registration required for US messaging.  A registration process will be required for each US local number sending SMS/MMS messages to the US. Initiate A2P 10DLC registration or check registration status".  I don't see how this applies to the way Twilio is being used with genmon. The messages are sent from a device in the U.S. to a cell phone in the U.S.  But, since I kept getting the emails, I registered for a Sole Proprietor Brand and Sole Proprietor Brand Campaign.  The campaign failed with the explanation "Not Sole Proprietor. Does not meet small business Sole Prop (EIN) criteria set by TCR and mobile carriers. (801)".

Is this required?  Is Twilio the best option?  Twilio has been working great but I can't see paying an additional $10 a month for the service.  I might be missing something but the campaign seemed to be straight forward.  Has anyone else gone through this process?

- Generator Model: Generac 22kW
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.17

jgyates, 2023-06-21:
Just a guess, but this may have to do with the FCC crackdown on spam text messages and robocalls. I am guessing the Twillio has to change the verification process for a phone number. 

email to SMS (assuming your carrier has that service) is a free way to send messages and there is an add on for that. The VoIP SMS add on for voip.ms is much cheaper than twillio also.

From the twillio site:
A2P 10DLC Announcements
Effective July 5, 2023, all SMS and MMS messages sent to U.S. phone numbers using 10DLC phone numbers must be sent via a registered application-to-person (A2P) Campaign. Messages sent using unregistered phone numbers will be subject to a gradual increase of message blocking by Twilio, beginning on July 5, 2023, ultimately leading to a full Twilio Network block of all unregistered U.S.-bound messages sent after August 31, 2023. Twilio messaging rates will still apply, even for blocked messages. To check the A2P 10DLC registration status of your Twilio phone numbers, use our [How do I check that I have completed US A2P 10DLC registration?](https://support.twilio.com/hc/en-us/articles/4418081745179-How-do-I-check-that-I-have-completed-US-A2P-10DLC-registration-)
 guide.

There are currently delays in the vetting process for 10DLC Campaigns resulting from the industry shift towards 3rd party reviews of all 10DLC Campaigns. Please continue submitting Campaign requests, but note that processing may take several weeks due to the high volume of requests.

All new Twilio A2P Campaigns registrations are now being charged a one-time $15 Campaign verification fee at the time of registration. [Learn more](https://support.twilio.com/hc/en-us/articles/11587910480155-A2P-10DLC-Campaign-Vetting-Changes-January-2023)

All existing Sole Proprietor (Starter) Campaigns are now incurring $2 monthly fees. We recommend that you delete any unused Sole Proprietor (Starter) Campaigns. [Learn more](https://support.twilio.com/hc/en-us/articles/1260803965530-What-pricing-and-fees-are-associated-with-the-A2P-10DLC-service-)

Customers, sending fewer than 3,000 message segments per day to U.S. destinations, are now required to complete A2P registration for each of their phone numbers. Messages sent with unregistered phone numbers are now incurring additional per-message carrier fees, and may be subject to filtering or blocking.

mxplea10, 2023-06-21:
Yes, that's it.  Basically, they give little information on setting this up and no help.  Do you recommend the VoIP SMS, is it simple to setup and does it work as good as Twilio?  Do you use this service?  Thanks...

jgyates, 2023-06-21:
Personally I use email for notifications as they provide more information.

I have a voip.ms account that I ported my home phone to. The SMS works good.

Another option is pushover, which is not a SMS but if you install the pushover app on you mobile device it will push messages to your device. that is also free.

Most mobile carriers in the US support email to SMS.

ATT: https://www.att.com/support/article/wireless/KM1061254/
Verizon: enter the 10-digit mobile number of the recipient followed by @vtext.com (e.g., 9495551212@vtext.com
T-Mobile: phonenumber@tmomail.net

mxplea10, 2023-06-21:
I use google voice for my home phone.  Do you know if this works with genmon?

mxplea10, 2023-06-21:
I like the instant notification of text.  I don't get notified when I have an email.

mxplea10, 2023-06-21:
Sorry for thinking out loud and sending so many messages but if SMS can be received via AT&T, I'll try to configure genmon to work with it.  Seems like I had a problem with this before, though.

mxplea10, 2023-06-21:
Is there a tutorial for configuring mobile carriers on genmon?

mxplea10, 2023-06-21:
Ok, I already have outbound email setup so I'll try it and let you know the outcome.  Thanks...

jgyates, 2023-06-21:
I am going to close this thread for now but feel free to post any follow up to this thread.

closed 2023-06-21 by jgyates

mxplea10, 2023-06-21:
Ok, I got it working once I generated a google app password for genmon.  I see that one of the reason that I went to Twilio was probably because the SMS addon isn't very detailed.

[image]
[image]
[image]

mxplea10, 2023-06-22:
I have the VOiP.ms account setup, API configured and received the DID activation email.  The Add-on is configured in genmon but I'm not receiving notifications for some reason.  I don't know if the log file will help but I'll send it.

jgyates, 2023-06-22:
I received your logs, the log file (/var/log/gensms_voip.log) as one error that is not that informative:

2023-06-22 16:31:42,000 : Error in MyVoipMs:SendSMS: The SMS message was not sent : myvoipms.py:54
2023-06-22 16:31:42,001 : Error sending message in QueueWorker, callback failed, retrying

This is from the voip.ms library at https://github.com/4doom4/python-voipms which is used by the add on.

First thing to check is that your DID and destination number are in the right format. When I tested mine were 9 digit numbers, e.g. 2025551234

jgyates, 2023-06-22:
another thing to validate is that you have a positive balance on your account. Then you can test sending a SMS from the customer portal. From the menu DID Numbers-> (Beta) SMS MMS Message Center. From there you can send a test message to make sure you account is working.

This page has a description of the process for sending from their web interface to test.

mxplea10, 2023-06-23:
Once I enable SMS (DID Settings -> Message Service (SMS/MMS) [Enable]) I was able to send a test message from the VoIP.ms customer portal.  This also allow genmon to send notifications to my cell.  Thanks for your help and guidance.

jgyates, 2023-06-23:
great! I am glad it is working for you.

mxplea10, 2023-06-23:
When I tested to make sure that I was receiving a status from the generator via genmon, I stop and started the generator with genmon.  Are there better ways to test that notifications are being received from the generator via the genmon VoIP.ms Add-on?

jgyates, 2023-06-23:
restarting the software will also test the notifications. you can restart from the advanced settings page. Double click the gear icon in the upper right to access the advanced settings page. At the bottom of the page are buttons that will allow you to restart the software. Note that these buttons do not have a progress bar so you will need to wait while the software restarts, then refresh your browser once the software has restarted.

mxplea10, 2023-06-24:
Ok, thanks.  This is a better way to test.