jgyates/genmon #913: Cannot connect to genmon

Barrygordon, 2023-06-21:
Good morning.  I have been using Genmon for since september 2022 with NO issues whatsoever.

Changed my email system but the address (barry@the-gordons.net) stayed the same  I was no longer getting emails from genmon, but the browser couild access the web page with no issue .Yesterday I made one change to the Genmon setup. I had it use the port 8000  There is one spot where that is done, but (I am 83 years old, and cn't remember anything) The port genmon was on was 5900 using wi-fi when I made the change.  I can no longer open the genmon app in the browser.  If I use port 5900 I get a message:

This page isn’t working
192.168.1.85 sent an invalid response.
ERR_INVALID_HTTP_RESPONSE

If I try port 8000 or 80 I get the following message:
This site can’t be reached192.168.1.85 refused to connect.
Try:

Checking the connection
[Checking the proxy and the firewall](chrome-error://chromewebdata/#buttons)
ERR_CONNECTION_REFUSED

I am using a wi-fi connection but I believe all the connectivity is correct. It is just when I made the one setup change that the issue started.  I could no longer get to genmon through my browser.

I am stuck.  I am not sure what I can do.  I am fairly adept with computers but have forgotten much since I started In the IT non-industry in 1963. I am looking for the read me on how to get started again if I have to reload the genmon software.  I have rebooted the genmon RPi several times, no change.  I tried logging in with TTY but it won''t accept what I thought was my uid/pswd.  Was there a default UID/Pswd so I can look at what the RP/Genmon are looking for using puTTY.  Where can I find the inirial download/setup instructions if there is no ther solution.  Have taken the RPi out of the case, left the PHat on, and powering it through a 12 V PSU.

 I just found the genmon wikki and assueme I will be able to start from scratch.  If I do start from scratch do I need to remove the Phat?
Please advise

Barry Gordon

jgyates, 2023-06-21:
older version of the Raspberry pi os used pi/rasbperry as the default username and password. If you could log in via ssh then it would be possible to recover / change settings / determine the current settings. One other things to try is to use the port 5900 with https instead of http. If you enabled secure web settings then it will use a username and password with https.

     https://192.168.1.99:5900

Of course change your IP address in the above to match your pi.

jgyates, 2023-06-22:
I don't think you will need to remove the pHat if you restart. you just need to re-flash your SD card.

jgyates, 2023-06-22:
I am going to close this issue but feel free to post to this thread with any follow up questions or status.

closed 2023-06-22 by jgyates