jgyates/genmon #914: Question: SD card size

Barrygordon, 2023-06-22:
What size micro SD card is needed to run Genmon.  What size is recommended

skipfire, 2023-06-22:
An 8GB card will be adequate, I would suggest 16GB.  However, more important than the space is the quality of the card.  I'm using a Gigastone High Endurance, even then if I get a full 2 years out of it I will be happy.

jgyates, 2023-06-22:
I agree with @skipfire. 

You can get by with less space if you chose the Lite version of the OS. If you don't need the Raspbian GUI and are comfortable with the console only then you can use the Lite version, which does not include the graphical components.

I use this one: https://www.amazon.com/gp/product/B07FCMBLV6/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

This one is 64G which is probably overkill but it is pretty cheap. I think a class 10 (speed rating) SD card is the recommended for a raspberry pi.

also if you setup a RAM drive for the operating system log files you can extend the life of your SD card: https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#extending-the-life-of-your-sd-card

jgyates, 2023-06-24:
I am going to close this issue but feel free to post follow up questions here.

closed 2023-06-24 by jgyates