jgyates/genmon #915: Serial Issues

Barrygordon, 2023-06-23:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

liltux, 2023-06-23:
What are your serial issues?

Barrygordon, 2023-06-23:
When I connect to the generator using pHat the generator monitor page shows timeout in communication stats   This system was working perfectly till a week ago when I I turned on web security at 8000’i have rebuilt the system on the same pi 4 and with the same phat.  I am pretty sure the serial port settings are wrong. I am getting ready to run the serialtest.pu. Is there any way I of checking the port parameters, what Genmon is using and what they should be?any help greatly appreciated.TIA,BarryBarry V Gordon

On Fri, 23 Jun 2023 at 11:49 AM Jared wrote:

What are your serial issues?

[
{
"target": "https://github.com/jgyates/genmon/issues/915#issuecomment-1604471917",
"url": "https://github.com/jgyates/genmon/issues/915#issuecomment-1604471917",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

jgyates, 2023-06-23:
If you have not done so, make sure you have run the installation script. It will ask if you want to setup the serial port. Select Y for yes. This will enable the pi's serial port. The instructions to do this are here: https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

Barrygordon, 2023-06-23:
I am running the latest version.  I have done all of that and what I get when I run the serial test is 0 bytes received.  Let me recap from the begining (about 3 days ago;

The system was working perfectly using version 17.??.  I use a RPI 4B and a pHat. The system ran with that configuration for about a year. I had to change email suppliers so I went to setup and made the changes needed.  I also changed web server securty settings from off to on.  The system no longer worked.  The browser would no longer log into the rpi at 192.168.1.85:8000. 

I decided to rebuild the entire system. I started with a fresh copy of Raspbian. I then did all the things as described in the wiki document "3.3 setup genmon software, however I did not run the serial test. I placed the pHat on the RPi and connected the Rpi and pHat to the generator supplying power through the generator (12v) to the barrel connector on the pHat, and connected the Molex cable to the pHat.  I could access the system through the browser, but there was no communication between the RPi/pHat and the generator.  I looked at the monitor page and saw that the connection status stated timeouts and 0 bytes were received.

I removed the system (Rpi/pHat) from the generator, removed the pHat and powered up the RPi . I logged in through the browser (192.168.1.85:8000 and saw the same issue as before when connected to the generator.  Stopped genmon and ran the serialtest.py  The serial test stated there was no received data, as I expected since I did not jump the GPIO pins.  I put a jumper on pins 8 & 10 (GPIO14 and GPIO15). I believe these are the tx and rx pins. Result was the same as before. no data received.

I am wondering how I should be runniing serialtest.py.  I have only run it with the pHat off so I had access to the Tx/Rx pins to jumper them. 

I am quite frustrated. I am 83 years old and have been retired for the last 20+ years. My entire working life was spent in IT.  I was part of the development teaam for GPS. I was a programmer and when I retired I was the director of opeartion for Northrop Grumman IT.  I used to have a fantastic memory, not so any more.  I now write everything down or capture it.  I played around with Linux and know my way around the command line with fat typing fingers.

I could really use some help.

I have attached a text file that shows what I have done (I re-installed Genmon)

[installing genmon.txt](https://github.com/jgyates/genmon/files/11852557/installing.genmon.txt)

Barrygordon, 2023-06-23:
No matter what I do I cannot seem to get serialtest.py to work. I have the Tx/RX pins (8 and 10) jumpered, genmon is stopped. the pHat is removed so it is just the RPi. Here is what I get from the serialconfig.py
 
root@barry:/home/pi/genmon/OtherApps# python serialconfig.py -c
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service disabled: OK
****************************************************************************

Here is what I get from the serialtest.py
root@barry:/home/pi/genmon/OtherApps# python serialtest.py

Note: Genmon must NOT be running for this test to work properly. If Genmon is running this test will not function properly,

Loopback testing for serial port /dev/serial0...

Using python serial library V3.5b0
Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes
root@barry:/home/pi/genmon/OtherApps#
**************************************************************************
I am coming to the conclusion that perhaps the RPi is bad.
Can I run Genmon on an RPi model 2 B V1.1 or an RPi model B+ v1.2? I have several of them lying around from 2014.   Which one would be preferred?.

Barrygordon, 2023-06-24:
I have ordered a RPi 3B+. (new not used)  It will arrive on Sunday.   I will try it from scratch (new SD card, loading and initializing gen,on from scratch).  Hopefully that will resolve the issue as being the RPi 4 I own.

jgyates, 2023-06-24:
Hi Barry,

While it is possible the pi is bad the odds are low that only the TTL serial port is bad as this comes out of the system on a chip and there is not any additional circuitry on the pi for the serial port between the SoC and the header pins.

 Here are a few suggestions:

1) From a fresh install, after you run the install script (which runs serialconfig.py) you need to reboot your system, if you have not already.
2) In most cases the changes are made to the serial port and the serial port shows up as /dev/serial0. On occasion I have see it get set to /dev/serial1. Let's run this command to make sure we have the right port:

        ls -ls /dev/serial*

This should show something like this:

      lrwxrwxrwx 1 root root 7 May 25 20:53 /dev/serial0 -> ttyAMA0
      lrwxrwxrwx 1 root root 5 May 25 20:53 /dev/serial1 -> ttyS0

We are looking for ttyAMA0 to be associated with /dev/serial0. If it is associated with /dev/serial1 then use /dev/serial1 insetad of /dev/serial0 in serialtest.py and the file /etc/genmon/genmon.conf

3) The serialtest.py can be used to loopback at the GPIO pins (8 and 10), or after the RS-232 board, or at the end of the cable. Each one of these test will test an additional part of the chain of the entire RS-232 path. 
4) If you basic loopback at pin 8 and 10 are not working and we have validated the port mapping in number 2 above, the next step is to send a picture so we can validate your looping of pin 8 and 10 are correct. Sometimes it helps to validate the orientation of the pin header.

Barrygordon, 2023-06-24:
Hi
It is 6/24/12:59 EST US. Here are some pictures showing the jumbers on pins 8 and 10 along with pictures of the RPi settings which might be germane.  I have included in prior parts of this thread some terminal screen copies that are probably also germane.  Everything appears s it should (IMHO).  I am going to start over fresh (build new 16Gb class 10 microSD cards).  I will make one for the RPi 4 and one for the RPI 3b+ which should be here tomorrow. I need to slow down and organize the work.  It really sucks getting old, memory, eyes, ears, finger control; They all go downhill. I guess the alternative to ageing is worse. 

[image]

[image]

[image]

[image]

jgyates, 2023-06-24:
That looks good, not can you show the output of this command:

    ls -ls /dev/serial*

Barrygordon, 2023-06-24:
As you requested:
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jun 23 19:58:21 2023
barry@barry:~ $ ls -ls /dev/serial*
0 lrwxrwxrwx 1 root root 7 Jun 23 19:58 /dev/serial0 -> ttyAMA0
0 lrwxrwxrwx 1 root root 5 Jun 23 19:58 /dev/serial1 -> ttyS0
barry@barry:~ $

jgyates, 2023-06-24:
That looks correct. Now if you connect the wires on your jumper (8 and 10) then run:

      python3 serialtest.py

show the output here

Barrygordon, 2023-06-24:
This is a little more than you requested but I am sure it is okay
******************************************************************************************************

login as: barry
barry@192.168.1.84's password:
Linux barry 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Jun 24 13:23:12 2023 from 192.168.1.202
barry@barry:~ $ sudo su
[sudo] password for barry:
root@barry:/home/barry# cd ../
root@barry:/home# cd ./pi/genmon
root@barry:/home/pi/genmon# ls
addon               Diagrams             genmon.py   requirements.txt
changelog.md        feature_requests.md  genserv.py  startgenmon.sh
ClientInterface.py  genloader.py         LICENSE     static
conf                genmonlib            OtherApps   templates
data                genmonmaint.sh       README.md
root@barry:/home/pi/genmon# bash startgenmon.sh stop
Stopping genmon python scripts
Stopping....
Stopping genserv.py
Stopping genmon.py
root@barry:/home/pi/genmon# cd OtherApps
root@barry:/home/pi/genmon/OtherApps# ls
check_monitor_system.py     kwlog2csv.py       serialconfig.py
ClientInterface.java        modbusdump.py      serialtest.py
DIY2TankSensorCalibrate.py  mopeka_utility.py  sockettest.py
root@barry:/home/pi/genmon/OtherApps# python3 serialtest.py

Note: Genmon must NOT be running for this test to work properly. If Genmon is running this test will not function properly,

Loopback testing for serial port /dev/serial0...

Using python serial library V3.5b0
Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes
root@barry:/home/pi/genmon/OtherApps#

root@barry:/home/pi/genmon/OtherApps# python3 serialconfig.py -c
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

root@barry:/home/pi/genmon/OtherApps#
**********************************************************************************************************

jgyates, 2023-06-24:
Now try this:

       cd genmon
       sudo ./startgenmon.sh stop
       ps -fA | grep python

Show the output of the above commands. This will stop genmon and then show all python processes to make sure we don't have a zombie version of genmon running.

Also, do you have a multimeter? if you do it may be worthwhile to check your loopback to make sure it is making a good connection.

Barrygordon, 2023-06-24:
Here it is:

barry@barry:~ $ cd ../pi/genmon
barry@barry:/home/pi/genmon $ sudo ./startgenmon.sh stop
[sudo] password for barry:
Stopping genmon python scripts
Stopping....
barry@barry:/home/pi/genmon $ ps -fA | grep python
barry        880       1  0 11:31 ?        00:00:00 /usr/bin/python3 /usr/share/system-config-printer/applet.py
barry       3222    3179  0 14:09 pts/0    00:00:00 grep --color=auto python
barry@barry:/home/pi/genmon $

Being an electrical engineer, I shecked the jumper out a while ago.  The jumper is good.  Pins 8 & 10 Are shorted together.

Maybe there is a bright side to this:  My Linux/debian skills are getting better!

jgyates, 2023-06-24:
OK, we are narrowing things down.. :)

Lest try this next. It looks like you have the pyserial library version 3.5.0b installed. That is a beta version. I am not sure how that go installed, I am guessing it is because a pip mirror server was down and one of the backup mirrors is out of date. Let's try to resolve that and see if it makes any difference.

       sudo pip3 uninstall pyserial
       sudo pip3 install pyserial

Then run this with the loopback installed on 8 and 10:

      sudo ./startgenmon.sh stop
      cd OtherApps
      sudo python3 ./serialtest.py

Barrygordon, 2023-06-24:
interesting:

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Jun 24 14:06:08 2023 from 192.168.1.202
barry@barry:~ $ cd ../pi/genmon
barry@barry:/home/pi/genmon $ sudo pip3 uninstall pyserial
[sudo] password for barry:
Found existing installation: pyserial 3.5b0
Not uninstalling pyserial at /usr/lib/python3/dist-packages, outside environment                                      /usr
Can't uninstall 'pyserial'. No files were found to uninstall.
barry@barry:/home/pi/genmon $ cd ../
barry@barry:/home/pi $ sudo su
root@barry:/home/pi# cd genmon
root@barry:/home/pi/genmon# pip3 uninstallpyserial
ERROR: unknown command "uninstallpyserial" - maybe you meant "uninstall"
root@barry:/home/pi/genmon# pip3 uninstall pyserial
Found existing installation: pyserial 3.5b0
Not uninstalling pyserial at /usr/lib/python3/dist-packages, outside environment /usr
Can't uninstall 'pyserial'. No files were found to uninstall.
root@barry:/home/pi/genmon#
************************************************************************
Note the sudo su and the directory I am in as root.

jgyates, 2023-06-24:
hmm, try this command:

     sudo apt uninstall python3-serial

It is possible to install this serial package with apt instead of pip. Some other software may have done that. if that works then run the commands:

     sudo pip3 uninstall pyserial

this should show that is it not installed. then:

     sudo pip3 install pyserial

Barrygordon, 2023-06-24:
No luck.  I am running as root (sudo su). and am in the genmon directory

root@barry:/home/pi/genmon# apt uninstall python3-serial
^[[DE: Invalid operation uninstall
root@barry:/home/pi/genmon# apt uninstall python3-serial
E: Invalid operation uninstall
root@barry:/home/pi/genmon# apt-uninstall python3-serial
bash: apt-uninstall: command not found
root@barry:/home/pi/genmon# sudo apt uninstall python3-serial
E: Invalid operation uninstall
root@barry:/home/pi/genmon#

jgyates, 2023-06-24:
sorry, getting my commands mixed up. Also let's add this command line option to full get rid of everything:

    sudo apt purge --auto-remove python3-serial

Barrygordon, 2023-06-24:
Here is the result (puTTY latest version is very nice it automaticall copies what you select to the clipboard (win 10)
****************************************************************************

root@barry:/home/pi/genmon# apt purge --auto-remove python3-serial
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  docutils-common* libfuse2* mypy* pylint* python3-astroid* python3-asttokens* python3-docutils* python3-isort*
  python3-jedi* python3-lazy-object-proxy* python3-logilab-common* python3-mccabe* python3-mote* python3-mypy*
  python3-mypy-extensions* python3-parso* python3-pygments* python3-roman* python3-send2trash* python3-serial*
  python3-typed-ast* python3-typing-extensions* python3-wrapt* raspinfo* sgml-base* thonny* xml-core* xsel*
0 upgraded, 0 newly installed, 28 to remove and 0 not upgraded.
After this operation, 57.3 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 96205 files and directories currently installed.)
Removing thonny (4.0.1-1~bpo11+1~rpt1) ...
Removing python3-docutils (0.16+dfsg-4) ...
Removing docutils-common (0.16+dfsg-4) ...
Removing libfuse2:arm64 (2.9.9-5) ...
Removing mypy (0.812-1) ...
Removing pylint (2.7.2-3) ...
Removing python3-astroid (2.5.1-1) ...
Removing python3-asttokens (2.0.4-1) ...
Removing python3-isort (5.6.4-1) ...
Removing python3-jedi (0.18.0-1) ...
Removing python3-lazy-object-proxy (1.5.2-1) ...
Removing python3-logilab-common (1.8.1-1) ...
Removing python3-mccabe (0.6.1-3) ...
Removing python3-mote (0.0.4) ...
Removing python3-mypy (0.812-1) ...
Removing python3-mypy-extensions (0.4.3-2) ...
Removing python3-parso (0.8.1-1) ...
Removing python3-pygments (2.7.1+dfsg-2.1) ...
Removing python3-roman (2.0.0-5) ...
Removing python3-send2trash (1.6.0~b1+git20210122.2eb3242-1) ...
Removing python3-serial (3.5~b0-1) ...
Removing python3-typed-ast (1.4.2-1) ...
Removing python3-typing-extensions (3.7.4.3-1) ...
Removing python3-wrapt (1.12.1-4+b1) ...
Removing raspinfo (20230123-1) ...
Removing xml-core (0.18+nmu1) ...
Removing sgml-base (1.30) ...
Removing xsel (1.2.0+git9bfc13d.20180109-3) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for gnome-menus (3.36.0-1) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u5) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for shared-mime-info (2.0-1) ...
Processing triggers for mailcap (3.69) ...
Processing triggers for desktop-file-utils (0.26-1) ...
(Reading database ... 90334 files and directories currently installed.)
Purging configuration files for xml-core (0.18+nmu1) ...
dpkg: warning: while removing xml-core, directory '/etc/xml' not empty so not removed
Purging configuration files for docutils-common (0.16+dfsg-4) ...
dpkg: warning: while removing docutils-common, directory '/etc/sgml' not empty so not removed
Purging configuration files for sgml-base (1.30) ...
Purging configuration files for pylint (2.7.2-3) ...
root@barry:/home/pi/genmon#
****************************************************************************
Next Step?

jgyates, 2023-06-24:
Now let's do this:

     sudo pip3 uninstall pyserial

That should validate it is gone from python's view. then install with pip3

    sudo pip3 install pyserial

then run serialtest.py with the loopback. serialtest.py will show the python version in the output so post that here.

Barrygordon, 2023-06-24:
okay.  No luck still has loopback problem. Python serial version is now V3.5 
**************************************************************************
root@barry:/home/pi/genmon/OtherApps# python3 serialtest.py

Note: Genmon must NOT be running for this test to work properly. If Genmon is running this test will not function properly,

Loopback testing for serial port /dev/serial0...

Using python serial library V3.5
Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes
root@barry:/home/pi/genmon/OtherApps#

jgyates, 2023-06-24:
Have there been any other software installed besides genmon on this system?

I am at a loss as to why this is not working for you. Other than reinstalling and *only* following the instructions on this page:

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

Do not install any additional software and do not make any changes to the serial port in the graphical user interface or with the raspi-config program. The only thing you need to be installed / enabled by the raspberry pi image is SSH.

The fact that a beta version of the python serial package was installed with apt instead of pip is unusual. Genmon does not use apt to install the serial package so I am not sure how that happened.

Barrygordon, 2023-06-24:
I shutdown then unplugged power on the Pi and double checked pins 8 and 10 on the back side of the board with a multimeter; and they are connected.  

I then re-applied power, brought the system backup.

I stopped genmon and ran the serial test.  Same result as before.  

My new microSD class 10 cards just arrived.  I am going to build a system from scratch for the Pi 4.  If the serial test works then we are home.  
If not, I will build a system for the Pi 3B+ which should arrive tomorrow.
Till then I am going to read and go out to dinner.  Thanks for the help.  I will contact you tomorroe

Barrygordon, 2023-06-26:
I have 2 new MicroSD cards (16 GB Class 10).  I have a new RPI 3B+. I have my old RPi 4 (was working fine for a year.

I am going to start fresh (I am retired so I have lots of time) tomorroe.

Where do you get the raspian OS images for the SD card from? 
I am using the Raspberry Pi Imager v1.7.5. 

I am planning on using the Raspberry Pi OS (64 bit)  a port of debian Bullseye with the RPi Desktop. released 5/3/2023; since it is stated to be compatible with both the Pi 3 and the Pi 4.

My other choices tis the Raspberry Pi OS  (legacy)  which is a port of Debian Buster.  I am just not sure if it will work on the Pi 4.

Your advice appreciated.

I did some testing with the pi 3 and serial test worked after I did the 
git clone step but before I ran the "genmonmaint .sh -i"

When I ren the genmonmaint.sh -i script on the Pi 3 it got up to the point "preparing wheel metadata . . ." which errored out and then started a long loop begining with preparing wheel Metadata The lines in the loop where mostly red and indicated errors.  I have attached a copy of the puTTY session which I aborted after the section kept repeating itself.

[genmon install.txt](https://github.com/jgyates/genmon/files/11862940/genmon.install.txt)

closed 2023-06-26 by jgyates

Barrygordon, 2023-06-27:
All is working now.  It was a combination of things.  The RPi 4 that was working the past year or so was the prime culprit. The serial chip is definately bad.  We had some sever electrical storms about the time the trouble started.  Of the two new micro SD cards one was bad.  I bought cheap ($8.00 for two cards).  The Pi 3B+ is working fine.  The upside of all this:  I started re-learning my Linux command line skills and dealing with Raspberry Pi.

Thank you for all the support.  The card is now back in its enclosure behind the generator .  The browser is talking to it fine,  Emails between the generator my email system are working fine.  I am going to get a good nights sleep not thinking about this "past" problem.

Thank you again