jgyates/genmon #916: Text messages to iPhone

Barrygordon, 2023-06-28:
I am looking for a way to have genmon send notifications as text messages to my iPhone.  Is there an add-on that does that.  If so what do I need and how do I proceed?  If possible restrict which notifications are sent e.g. generater start and stop times and outage notifications.

TIA,
Barry Gordon

madisdev, 2023-06-28:
There is pushover notification available.

jgyates, 2023-06-28:
The Pushover service is free, there is the voice over IP add on, the twillio add (gensms) has some issues as twillio is changeing their cost structure. This thread has some details on twillio and which genmon add on programs are good alternatives.

https://github.com/jgyates/genmon/issues/912

closed 2023-06-30 by jgyates