jgyates/genmon #917: MQTT Integration JSON for Numerics doesnt include temperature sensors

chrkov, 2023-06-29:
### Expected Behavior

Enabling 'JSON for Numerics works as expected, but doesn't convert the temperature sensors.

[image]

### Actual Behavior

Works as if 'JSON for Numerics is off'

### Steps to Reproduce (including precondition)

Turn on 'JSON for Numerics' and look at temperature sensors

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: 16kw
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2023-06-29:
Good catch. That was an oversight on my part. I checked in an update that corrects this issue. The version number has not changed. Let me know if you have any issues.

closed 2023-06-29 by jgyates

chrkov, 2023-06-29:
Thanks.   I am currently on 1.18.17 software.   How to I get the fix installed if the version doesn't change?

jgyates, 2023-06-29:
Just go to the about page and click the update button. the software will update and restart, but the version number will not change.

chrkov, 2023-06-29:
Did that, but seeing the same outcome as before.
[image]

jgyates, 2023-06-29:
Sometimes github takes a while to update. To validate you have the latest you can check the file date of ./genmon/genmonlib/controller.py. Log into the pi via ssh and type:

    cd genmon
    ls -la ./genmonlib/controller.py 

You should some an output with todays date. If the date if from before today then you should wait and try it again later. The output I have on my system is this:

    -rw-r--r-- 1 pi pi 122934 Jun 29 15:49 ./genmonlib/controller.py

The time and date is from when I performed the update on my pi. If the update is not posted in the github mirror you are using then it would be from an earlier date.

chrkov, 2023-06-29:
My file shows a date of May 3rd right now.   I will try again later and check the date again.  

Thanks!

chrkov, 2023-07-02:
FYI.   Still not updating.   I just tried again today to do the upgrade and my file still shows May 3rd.

jgyates, 2023-07-03:
Can you describe how you are updating the software? The changes that I checked in are showing up from the web interface:

https://github.com/jgyates/genmon/blob/c2b611f1bfd71c84cb2e3b0aba76b6ee9b0bbb31/genmonlib/controller.py#L2088

If you are updating the software and changes are not occurring this is either an issue with github (it works when I update) or a local issue with you pi.

chrkov, 2023-07-03:
I am going to the PI via chrome at http://192.168.151.163:8000
Clicking on 'About'
Clicking on 'Upgrade to latest version'

It does have access to the internet.  I can ping 4.2.2.2.  DNS also works as I can ping google.com.  Resolves to 142.251.40.206

jgyates, 2023-07-03:
try logging in to the pi via ssh and then type:

      cd genmon
      ./genmonmaint.sh -u

Then when prompted type "y" for Yes to update to the latest. Post your output here.

If you type this command:

       grep "popitem" ./genmonlib/controller.py 

It will output nothing if the update is not there. If the update is there it will output this:

      TempKey, TempData = TempDict.popitem()

chrkov, 2023-07-03:
That worked.   I tried one more time from the gui and still that date on that file remained as May 3rd.

Did it from the CLI and it updated and is showing the json on the temp sensor now.

Here is the output if you still wanted it.

'''
genmonpi@genmon:~/genmon $ ls -la ./genmonlib/controller.py
-rw-r--r-- 1 genmonpi genmonpi 121690 May  3 19:22 ./genmonlib/controller.py
genmonpi@genmon:~/genmon $  ./genmonmaint.sh -u
This script will update genmon to the latest version from the github repository. This script requires internet access. If you have modified any files in the genmon directory, they will be overwritten. Configuration files in the configuration directory will not be overritten.  Continue? (y/n)  y
Updating genmon...
remote: Enumerating objects: 109, done.
remote: Counting objects: 100% (109/109), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 109 (delta 63), reused 94 (delta 48), pack-reused 0
Receiving objects: 100% (109/109), 1009.07 KiB | 4.50 MiB/s, done.
Resolving deltas: 100% (63/63), completed with 23 local objects.
From https://github.com/jgyates/genmon
   4261395..c2b611f  master     -> origin/master
Updating files: 100% (264/264), done.
HEAD is now at c2b611f V1.18.17
Done.
genmonpi@genmon:~/genmon $ ls -la ./genmonlib/controller.py
-rw-r--r-- 1 genmonpi genmonpi 122967 Jul  2 22:10 ./genmonlib/controller.py
genmonpi@genmon:~/genmon $
'''
[image]

chrkov, 2023-07-03:
> ok. If you you outbound email working you can submit your logs. There may be info as to why the update from the about page was not working for you in your log files.
>
>
> ***Notice: This email may contain privileged or confidential information and is for the sole use of the intended recipient(s). If you are not the intended recipient, any disclosure, copying, distribution, or use of the contents of this information is prohibited and may be unlawful. If you have received this electronic transmission in error, please reply immediately to the sender that you have received the message in error, and delete it. Thank you.

Ok.  I just sent the logs over to you.

jgyates, 2023-07-05:
I looked at your logs and did some digging. I *think* I resolved the issue but you may have to do one more manual upgrade to get the fix. [See this thread for more info](https://github.com/jgyates/genmon/issues/920).