jgyates/genmon #918: Notifications via Pushover

bourg2, 2023-06-30:
Notifications from Pushover are no longer being sent to my devices. I've verified application ID & Pushover user keys are correct from my Pushover dashboard. I've tried creating an entirely new application in Pushover but the result is the same. The last message received through Pushover was received on 5/23/23. Current Genmon version is V1.18.17.

skipfire, 2023-06-30:
Was 5/23 the end of your 30 day trial?  
Have you logged into the pushover portal and tested the send from there?
Have you checked the logs on your Pi?  `cat /var/log/genpushover.log`

bourg2, 2023-06-30:
I use Pushover for other apps and have purchased a license.
Pushover portal test message is successful, other apps are sending notifications through it. also
genpushover.log is empty.

jgyates, 2023-06-30:
Try updating genmon from the About page. There was a bug that was briefly introduced that stopped all notifications form happening but it was corrected within 24 hours. If you updated your software in that 24 hour period you may need to update from the About page and restart. The version number will not update. Let me know if that resolves your issue.

bourg2, 2023-06-30:
That seems to have resolved the issue. Thanks for your help.

jgyates, 2023-06-30:
No problem. I am glad is is working for you.

closed 2023-06-30 by jgyates

griffths, 2023-09-21:
I have been experiencing this trouble as well.  When i review the logfile as @skipfire suggests I see the following error message but have no idea what it means or how to resolve.

2023-09-20 12:19:19,022 : Error sending message in QueueWorker, callback failed, retrying
2023-09-20 12:21:19,371 : Send Notice Error: genpushover.py:241: Bad sound: must be in ('Klaxon', 'alien', 'bike', 'bugle', 'cashregister', 'classical', 'climb', 'cosmic', 'echo', 'falling', 'gamelan', 'incoming', 'intermission', 'magic', 'mechanical', 'none', 'persistent', 'pianobar', 'pushover', 'siren', 'spacealarm', 'tugboat', 'updown', 'vibrate'), was 'klaxon'

My pushover account is still active and I can successfully send test messages to my IOS device directly from the Pushover webpage.

Not sure what the second line is saying but my custom sound is still working on my iPhone 

I have updated to the latest version from the about page a few days ago and still have the trouble.

Thanks

Jay

skipfire, 2023-09-21:
In GenMon Add-ins under Pushover there is a value for "Push Sound" and you currently have it set to "klaxon".  It needs to be changed to Klaxon (note the upper case) or to any of the following: 

'alien', 'bike', 'bugle', 'cashregister', 'classical', 'climb', 'cosmic', 'echo', 'falling', 'gamelan', 'incoming', 'intermission', 'magic', 'mechanical', 'none', 'persistent', 'pianobar', 'pushover', 'siren', 'spacealarm', 'tugboat', 'updown', 'vibrate'

griffths, 2023-09-21:
Thanks for the quick reply...

The custom sound was/is "Klaxon" on the add on config page.

Will try a stock sound to see if that makes any difference.

griffths, 2023-09-21:
OK,  it works again with a stock sound.

Any suggestions why custom sound is not working?

skipfire, 2023-09-21:
The error message was that you were using an invalid sound name.  Without looking at the code the best guess I have is that GenMon might be making the sound lower case but Pushover may now require that capital K on Klaxon.

jgyates, 2023-09-21:
I looked at the code for genpushover.py and it looks like the sound is being converted to lower case which would prevent Klaxon from working. When this was created the Klaxon sounds was not available, all of them were lower case (and case sensitive) so I added the conversion to lowercase to make things more user friendly.

I will post to this thread when I have a fix.

griffths, 2023-09-21:
Changing the name to lower case on the Pushover webpage solved the trouble.

Thanks to both for your assistance.

Jay

jgyates, 2023-09-21:
Klaxon is not listed on their web site. I am assuming this is a custom sound. I updated the tool tip in the web app to include the lowercase requirement. Thanks for the feedback

Thansk @skipfire for the quick response

bourg2, 2023-11-18:
Pushover notifications have stopped working again. I've tried the previous suggestion of updating and restarting but that does not resolve the issue.

This is the log from Pi after reboot using cat /var/log/genpushover.log

2023-11-18 11:28:30,209 : Send Notice Error: genpushover.py:230: 'NoneType' object has no attribute 'raw'
2023-11-18 11:28:30,211 : Error sending message in QueueWorker, callback failed, retrying
2023-11-18 11:30:42,497 : Send Notice Error: genpushover.py:230: 'NoneType' object has no attribute 'raw'
2023-11-18 11:30:42,498 : Error sending message in QueueWorker, callback failed, retrying
2023-11-18 11:32:54,785 : Send Notice Error: genpushover.py:230: 'NoneType' object has no attribute 'raw'
2023-11-18 11:32:54,786 : Error sending message in QueueWorker, callback failed, retrying
2023-11-18 11:35:07,073 : Send Notice Error: genpushover.py:230: 'NoneType' object has no attribute 'raw'
2023-11-18 11:35:07,074 : Error sending message in QueueWorker, callback failed, retrying
2023-11-18 11:37:19,361 : Send Notice Error: genpushover.py:230: 'NoneType' object has no attribute 'raw'
2023-11-18 11:37:19,362 : Error sending message in QueueWorker, callback failed, retrying
2023-11-18 11:38:32,705 : Send Notice Error: genpushover.py:230: 'NoneType' object has no attribute 'raw'
2023-11-18 11:38:32,707 : Error sending message in QueueWorker, callback failed, retrying

jgyates, 2023-11-18:
I will need more info to look into this.

Did they stop randomly or did they stop after a change was made to the the software or system?

Have you updated your software to the latest version? V1.19.00 is the latest. If that does not fix the problem I will need to see your logs which you can submit from the about page. the logs give me more info on version of the OS, version of the software, python version etc.

also, from a ssh console what is the output of this command:

     pip3 freeze

this will show the version of the python libraries.

bourg2, 2023-11-18:
They stopped randomly. I did make a change in network equipment but notifications through pushover were working from genmon initially.

I hadn't noticed but updating from the software about page doesn't update to the latest. It goes through the process and restarts but keeps me on ver. 1.18.18. 

I will submit logs through email shortly

output from pip3 freeze

bleson==0.1.8
certifi==2022.9.24
cffi==1.15.1
charset-normalizer==2.1.1
chump==1.6.0
click==8.1.3
configparser==5.3.0
crcmod==1.7
cryptography==38.0.1
dbus-python==1.2.16
distro-info==1.0
Flask==2.2.2
fluids==1.0.22
geojson==2.5.0
idna==3.4
importlib-metadata==5.0.0
itsdangerous==2.1.2
Jinja2==3.1.2
ldap3==2.9.1
MarkupSafe==2.1.1
mopeka-pro-check==0.1.0
numpy==1.23.3
paho-mqtt==1.6.1
ply==3.11
psutil==5.9.2
pyasn1==0.4.8
pycparser==2.21
pycryptodomex==3.15.0
PyJWT==2.5.0
pyOpenSSL==22.1.0
pyotp==2.7.0
pyowm==2.10.0
pyserial==3.5
pysmi==0.3.4
pysnmp==4.4.12
python-apt==2.2.1
pytz==2022.4
requests==2.28.1
scipy==1.9.1
smbus==1.1.post2
spidev==3.5
twilio==7.14.2
unattended-upgrades==0.1
urllib3==1.26.12
voipms==0.2.5
Werkzeug==2.2.2
zipp==3.8.1

jgyates, 2023-11-18:
I have not received your logs yet, but in the interim if you try to manually update it should tell you the error that is occurring during the update. To manually update:

       cd genmon
       ./genmonmaint.sh -u

bourg2, 2023-11-19:
Ok so I'm an idiot. I never noticed that my pi DNS server was still set static to my old IP address pool therefore not allowing an internet connection. I fixed this and now I'm getting pushover notifications from genmon. That would explain why I couldn't update the genmon software. Sorry for that.

jgyates, 2023-11-19:
No worries, I am glad it is working for you now.