jgyates/genmon #919: Genmon stops working after 2-3 weeks.

RoyRomeo, 2023-07-03:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Genmon working continuously

### Actual Behavior

Genmon works for 2-3 weeks then stops.  If I go out to the generator and cut power to the Genmon and reconnect, it will work for another 2-3 weeks.  Any suggestions?

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 6244
- Genmon Logs: Submitted
- Genmon Version: V1.18.17

jgyates, 2023-07-03:
Can you give more details on what is not working? Are you getting serial errors? Can you access the web interface? I can offer guesses, but if you want to pursue this further we can try to get more info the next time it happens.

If it happens agains make a note of or try the following:

1) Can you access the web interface
2) Are there serial errors occurring? They will be shown on the Monitor page
3) if there are serial issues we can look at the file /var/log/myserial.log and look for any errors
4) also if there are serial issues we can look at your cable to make sure it is not marginal
5) We can also look at monitor page and see if there are power issues (under Platform Stats)
6) if you can not access the web UI, can you log in to the pi via ssh?
7) if you can not access the web UI, if you are using external access via your router, try disabling that and see if you still see the issue.

RoyRomeo, 2023-07-03:
I typically get a "site cannot be reached" error.

jgyates, 2023-07-04:
ok. Numbers 5,6 and 7 above will be the steps to take if / when this occurs again. If you can log in via ssh then we would use the ClientInterface program to see if genmon is still responding and if the problem is only with the web interface. To see if genmon is still sending and receiving we can do this from the ssh console:

      cd genmon
      python3 ClientInterface.py

This will give a prompt that you can type commands. Type this:

    generator: monitor

This will show the contest of the Monitor page (without the web UI). Make a note of the serial stats and then issue the command again (generator: monitor) and see if the serial packet count is changing.

To exit the ClientInterface program type "exit".

If that looks good then the problem is likely some interaction with your router and the web interface. If you have setup your router with external access to the web interface, then try disabling that. I have see a few instances where some routers will block web servers on the local network as the router assumes it is a malicious attack on your network. I have also seen instances where people have tried to access externally exposed instances of genmon. If you are using external access, this is not recommend and leaves your generator vulnerable to hackers. If you want external access to your generator then use a VPN. See this thread for more info: https://github.com/jgyates/genmon/issues/702

I have see about 10 different instances where the web interface stopped working after a period of time and almost all of them were related to people setting up external access to genmon. the internal web server that genmon uses was not designed to handle high levels of traffic. If you put a web server on the internet, people will try to access it, even if you do not let it be known it is there.

RoyRomeo, 2023-07-04:
Okay, thanks very much.

jgyates, 2023-07-04:
No problem. I am going to close this thread but feel free to post here with follow up.

closed 2023-07-04 by jgyates