jgyates/genmon #920: Genmon upgrade issue / new version

jgyates, 2023-07-05:
I am posting a new version of genmon (V1.18.18) that has a few new features and fixes. One issue that I have addressed is related to a problem with genmon not upgrading on some systems. I believe I have tracked this down to the file permissions on the local repository are not in an expected state. I have a fix for this in the new version, however you have to be able to upgrade to get the version. ***If you attempt to upgrade your software and it does not then do the following***:

1) Log into your pi via ssh, then type:

       cd genmon
       sudo ./genmonmaint.sh -u

Then enter 'y' for Yes when prompted. When this is complete restart genmon with this command:

      ./startgenmon.sh restart

Refresh your web browser and you should see the new version on the About page. Future upgrades should work as expected.

If clicking the upgrade button on the About pages works for you then this issue was not present on your pi so just continue as in the past.

mxplea10, 2023-07-05:
Thanks... I upgraded with the steps above.

w-u-2-o, 2023-07-06:
Update through the web interface failed. Used the above procedure and am now at .18. Thanks!

wtnh, 2023-07-25:
The above procedure worked flawlessly for me too. Thanks!

closed 2023-10-05 by jgyates