jgyates/genmon #922: dd/mm/yyyy Date format

jmone1, 2023-07-12:
Just a minor one, about having the option for dd/mm/yyyy date format (instead of mm/dd/yyyy).  
Thanks
Nathan

jgyates, 2023-07-12:
Hi Nathan,
Are you using and Evolution or Nexus controller? changing this for H100 or Powerzone is a little more problematic. if you are using Evolution or Nexus I will put this on the feature list.

jgyates, 2023-07-12:
https://github.com/jgyates/genmon/blob/master/feature_requests.md

jmone1, 2023-07-12:
Hi, 

Was just thinking about the dates in the GenMon GUI / Logs etc.
- Controller Detected : Evolution 2.0, Air Cooled
- Running on a : Pi Model : Raspberry Pi 4 Model B Rev 1.2
Thanks
Nathan

jgyates, 2023-07-12:
I have add a setting on the Advanced Settings Page called "Alternate Date Format" enable this to get the dd/mm/yyyy format. Double click the gear icon in the upper right for the Advanced settings page. Let me know if you have any issues. Just update from the About page. The version did not change, it is still V1.18.18.

closed 2023-07-12 by jgyates

jmone1, 2023-07-12:
Thanks!  Working great.

lwbeam, 2023-09-07:
Just tried the _Alternate Date Format_ and have notice a minor glitch. When you have the format set to dd/mm/yyyy, the simplified calendar at the top of the **Logs** page displays the events as mm/dd/yyyy. **Example:** An event that occurred on September 4, 2023 gets converted to 04/09/2023, but appears in the calendar as if it occurred on April 9, 2023.

BTW, big thanks for implementing this feature! Always found mm/dd/yyyy counter-intuitive.

jgyates, 2023-09-08:
Thanks for catching this. I will put this on the list of things to fix.

jgyates, 2023-10-27:
The fix for this is checked into github. The version number has not yet been updated so the latest version is still V1.18.18. Let me know if you have any issues.