jgyates/genmon #923: Daily emails and text notifications as if PI zero is restarting

SD585, 2023-07-13:
Hello I have a pi zero running the newest version of genmon on my 12kw air cooled generator. I just installed this a few days ago and have received the same text messages a couple times a day almost as if the PI is resetting.

Utility service is up

System health OK

Software is up to date

Pi health ok

Generator ready

Do you have any thoughts or ways to rule out what is going on? This should not do this if it were the wifi dropping out correct?

Thanks!

skipfire, 2023-07-13:
It should not do that and a dropping WiFi should not cause a reboot.  The two common causes I have found for a Pi rebooting is a bad (or low quality) SD card or a power issue.  What is your hardware setup, did you buy a HAT from me (PintSize.Me) or make your own interface?

What kind of controller do you have?  (Evo, Evo 2, Nexus?)

How are you supplying power to the setup?

jgyates, 2023-07-13:
I agree with Jon on the power and SD card and the messages being a symptom of the pi restarting. If you look on the Monitor page under "Platform Stats", what are the values of these entries:

        Pi CPU Frequency Throttling
        Pi ARM Frequency Cap
        Pi Undervoltage

If any of these are anything other than 'OK' then that is another vote for a bad power source.

jgyates, 2023-07-15:
I am going to close this issue but feel free to post any updates or questions to this thread.

closed 2023-07-15 by jgyates