jgyates/genmon #924: cant get to gen mon page

lampe208, 2023-07-16:
For some reason , i cannot access the gen mon page to see my generator.  I know its working fine because I found your upgrade. Even before your upgrade with ssh I noticed my router assigned a new IP.  I go to the new IP and i get "site cannot be reached refused to connect.  Any help would be appreciated greatly .

skipfire, 2023-07-16:
Did you forget the :8000/ on the address?  That's happened to me more than once.  You can also try manually restarting GenMon when logged in via ssh with `./startgenmon.sh restart`.

lampe208, 2023-07-16:
I switch to another computer and now its working....my mac and laptop will not connect to it?  What would cause that?

skipfire, 2023-07-16:
Typo, maybe the computer is trying to automatically push it to https instead of http, or network configs.  Is the working one wired and the non-working wireless?  I have seen some routers with a setting that blocks wireless devices from talking to each other.

lampe208, 2023-07-16:
Yes, on is wired and others are wireless.   Thanks for your help.  I will check into the settings.

closed 2023-07-16 by lampe208