jgyates/genmon #926: Monitoring

stealthbootc, 2023-07-18:
Hello,

I have Genmon setup and running and love it but want to track historical data. I have added it to Librenms but am unsure how to get all the additional data. Has someone already done this or is there a guide im missing somewhere. I was able to do a SNMP walk but that just gives me a TON of information to decypher.

----------------------- New Test -----------------------
Paessler SNMP Tester - 20.2.4 Computername: SNT-Z690-P01 Interface: (192.168.1.254, 100.110.6.23)
7/18/2023 1:16:13 PM (6 ms) : Device: genmon.schwartznet.com
7/18/2023 1:16:13 PM (9 ms) : SNMP v1
7/18/2023 1:16:13 PM (11 ms) : Walk 1.3.6.1.4.1.58399
7/18/2023 1:16:13 PM (43 ms) : 1.3.6.1.4.1.58399.0.0.1 = "OK" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (58 ms) : 1.3.6.1.4.1.58399.0.0.2 = "Generator Monitor running for 2 days, 22:20:34." [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (75 ms) : 1.3.6.1.4.1.58399.0.0.3 = "0.00 MB of 15.00 MB" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (92 ms) : 1.3.6.1.4.1.58399.0.0.4 = "V1.18.18" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (126 ms) : 1.3.6.1.4.1.58399.0.1.1 = "M: 7961499, S: 7961497" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (143 ms) : 1.3.6.1.4.1.58399.0.1.2 = "0 " [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (155 ms) : 1.3.6.1.4.1.58399.0.1.3 = "0.00%" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (167 ms) : 1.3.6.1.4.1.58399.0.1.4 = "1" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (179 ms) : 1.3.6.1.4.1.58399.0.1.5 = "0.00%" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (193 ms) : 1.3.6.1.4.1.58399.0.1.6 = "514" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (206 ms) : 1.3.6.1.4.1.58399.0.1.7 = "0" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (218 ms) : 1.3.6.1.4.1.58399.0.1.8 = "0" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (230 ms) : 1.3.6.1.4.1.58399.0.1.9 = "0" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (242 ms) : 1.3.6.1.4.1.58399.0.1.10 = "0" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (257 ms) : 1.3.6.1.4.1.58399.0.1.11 = "62.88" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (269 ms) : 1.3.6.1.4.1.58399.0.1.12 = "0.0312 sec" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (283 ms) : 1.3.6.1.4.1.58399.0.2.1 = "132.44 F" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (295 ms) : 1.3.6.1.4.1.58399.0.2.2 = "Raspberry Pi 3 Model A Plus Rev 1.0" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (327 ms) : 1.3.6.1.4.1.58399.0.2.3 = "OK" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (340 ms) : 1.3.6.1.4.1.58399.0.2.4 = "OK" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (353 ms) : 1.3.6.1.4.1.58399.0.2.5 = "OK" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (369 ms) : 1.3.6.1.4.1.58399.0.2.6 = "7.11%" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (381 ms) : 1.3.6.1.4.1.58399.0.2.7 = "Raspbian GNU/Linux" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (398 ms) : 1.3.6.1.4.1.58399.0.2.8 = "11 (bullseye)" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (410 ms) : 1.3.6.1.4.1.58399.0.2.9 = "3 days, 16:20:34" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (422 ms) : 1.3.6.1.4.1.58399.0.2.10 = "wlan0" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (435 ms) : 1.3.6.1.4.1.58399.1.0.0.1 = "Auto" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (451 ms) : 1.3.6.1.4.1.58399.1.0.0.2 = "Off - Ready" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (465 ms) : 1.3.6.1.4.1.58399.1.0.0.3 = "" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (480 ms) : 1.3.6.1.4.1.58399.1.0.0.4 = "" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (495 ms) : 1.3.6.1.4.1.58399.1.0.0.5 = "13.40 V" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (528 ms) : 1.3.6.1.4.1.58399.1.0.0.6 = "" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (541 ms) : 1.3.6.1.4.1.58399.1.0.0.7 = "0 " [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (555 ms) : 1.3.6.1.4.1.58399.1.0.0.8 = "0.00 Hz" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (568 ms) : 1.3.6.1.4.1.58399.1.0.0.9 = "0 V" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (583 ms) : 1.3.6.1.4.1.58399.1.0.0.10 = "0.00 A" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (598 ms) : 1.3.6.1.4.1.58399.1.0.0.11 = "0.00 kW" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (610 ms) : 1.3.6.1.4.1.58399.1.0.0.12 = "0 " [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (623 ms) : 1.3.6.1.4.1.58399.1.0.1.1 = "243 V" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (636 ms) : 1.3.6.1.4.1.58399.1.0.1.2 = "251 V" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (649 ms) : 1.3.6.1.4.1.58399.1.0.1.3 = "185 V" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (661 ms) : 1.3.6.1.4.1.58399.1.0.1.4 = "156 V" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (677 ms) : 1.3.6.1.4.1.58399.1.0.1.5 = "" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (690 ms) : 1.3.6.1.4.1.58399.1.0.1.6 = "" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (703 ms) : 1.3.6.1.4.1.58399.1.0.2.1 = "07/06/23 17:47:06 No Rotation Warning : Alarm Code: 0000" [ASN_OCTET_STR]
7/18/2023 1:16:13 PM (732 ms) : 1.3.6.1.4.1.58399.1.0.2.2 = "" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (778 ms) : 1.3.6.1.4.1.58399.1.0.2.3 = "07/17/23 11:47:05 Stopped - Auto " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (794 ms) : 1.3.6.1.4.1.58399.1.0.3.1 = "Tuesday July 18, 2023 13:16:12" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (808 ms) : 1.3.6.1.4.1.58399.1.0.3.2 = "Tuesday July 18, 2023 13:16" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (823 ms) : 1.3.6.1.4.1.58399.1.1.0.1 = "G0072101" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (837 ms) : 1.3.6.1.4.1.58399.1.1.0.2 = "3011984539" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (852 ms) : 1.3.6.1.4.1.58399.1.1.0.3 = "Evolution 2.0, Air Cooled" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (866 ms) : 1.3.6.1.4.1.58399.1.1.0.4 = "3600" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (879 ms) : 1.3.6.1.4.1.58399.1.1.0.5 = "20 kW" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (894 ms) : 1.3.6.1.4.1.58399.1.1.0.6 = "60 Hz" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (908 ms) : 1.3.6.1.4.1.58399.1.1.0.7 = "Propane" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (942 ms) : 1.3.6.1.4.1.58399.1.1.0.8 = "80.72 %" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (957 ms) : 1.3.6.1.4.1.58399.1.1.0.9 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (970 ms) : 1.3.6.1.4.1.58399.1.1.0.10 = "999 cc" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (986 ms) : 1.3.6.1.4.1.58399.1.1.0.11 = "107 F" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (998 ms) : 1.3.6.1.4.1.58399.1.1.0.12 = "2.25 kWh" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1012 ms) : 1.3.6.1.4.1.58399.1.1.0.13 = "0.64 gal" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1032 ms) : 1.3.6.1.4.1.58399.1.1.0.14 = "0.64 gal" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1046 ms) : 1.3.6.1.4.1.58399.1.1.0.15 = "0.29 h" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1062 ms) : 1.3.6.1.4.1.58399.1.1.0.16 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1075 ms) : 1.3.6.1.4.1.58399.1.1.0.17 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1092 ms) : 1.3.6.1.4.1.58399.1.1.0.18 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1108 ms) : 1.3.6.1.4.1.58399.1.1.0.19 = "OK" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1120 ms) : 1.3.6.1.4.1.58399.1.1.0.20 = "0.29 h" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1148 ms) : 1.3.6.1.4.1.58399.1.1.1.1 = "1483 " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1164 ms) : 1.3.6.1.4.1.58399.1.1.1.2 = "1480 " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1177 ms) : 1.3.6.1.4.1.58399.1.1.1.3 = "1052 " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1190 ms) : 1.3.6.1.4.1.58399.1.1.1.4 = "240 V" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1209 ms) : 1.3.6.1.4.1.58399.1.1.1.5 = "24 kW" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1226 ms) : 1.3.6.1.4.1.58399.1.1.1.6 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1242 ms) : 1.3.6.1.4.1.58399.1.1.1.7 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1259 ms) : 1.3.6.1.4.1.58399.1.1.1.8 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1275 ms) : 1.3.6.1.4.1.58399.1.1.1.9 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1288 ms) : 1.3.6.1.4.1.58399.1.1.1.10 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1302 ms) : 1.3.6.1.4.1.58399.1.1.1.11 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1315 ms) : 1.3.6.1.4.1.58399.1.1.1.12 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1345 ms) : 1.3.6.1.4.1.58399.1.1.1.13 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1360 ms) : 1.3.6.1.4.1.58399.1.1.2.1 = "Weekly Tuesday 17:00 Quiet Mode On" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1376 ms) : 1.3.6.1.4.1.58399.1.1.2.2 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1389 ms) : 1.3.6.1.4.1.58399.1.1.3.1 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1404 ms) : 1.3.6.1.4.1.58399.1.1.3.2 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1419 ms) : 1.3.6.1.4.1.58399.1.1.3.3 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1431 ms) : 1.3.6.1.4.1.58399.1.1.3.4 = " " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1446 ms) : 1.3.6.1.4.1.58399.1.1.3.5 = "195 hrs or 03/23/2025 " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1458 ms) : 1.3.6.1.4.1.58399.1.1.3.6 = "395 hrs or 03/23/2027 " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1469 ms) : 1.3.6.1.4.1.58399.1.1.3.7 = "03/23/2024 " [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1480 ms) : 1.3.6.1.4.1.58399.1.1.3.8 = "4.00 h" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1492 ms) : 1.3.6.1.4.1.58399.1.1.3.9 = "V1.03" [ASN_OCTET_STR]
7/18/2023 1:16:14 PM (1502 ms) : 1.3.6.1.4.1.58399.1.1.3.10 = "V1.15" [ASN_OCTET_STR]

jgyates, 2023-07-18:
I do not have experience with Librerms however the SNMP definitions are there:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gensnmppy-optional

There is also an SNMP MIB file in ~./genmon/data/mib

Another data point is that the SNMP data aligns with the order of the data as it is presented in the web interface.

Let me know if you have any questions.

liltux, 2023-07-18:
What I do is just use the genmon interface commands to report data in JSON form, parse it out, and store the data in a sqlite db.

jgyates, 2023-07-18:
This wiki article describes how to export JSON data: https://github.com/jgyates/genmon/wiki/Appendix-C--Interfacing-Generator-Monitor-to-External-Applications

jgyates, 2023-07-20:
I am going to close this issue but feel free to post any follow up questions or status to this thread.

closed 2023-07-20 by jgyates

stealthbootc, 2023-07-23:
Is there any way to change the SNMP to where it gives numbers as integers instead of strings? We will have to use regex or something to remove the V or kW from what most other projects just give us as a number and identify what it is. Text should still be in string format though. This would make importing the data much easier.

jgyates, 2023-07-23:
Yes. The latest version of the SNMP add on has an option named "Force Integers" that will return the SNMP data type INTEGER for integer values. Floating point values are returned as strings (with the units removed) if this option is enabled.