jgyates/genmon #927: No generator data during outage

el0552t, 2023-07-22:
Hello,

I have been using Genmon now for several months now. I have experienced the documented issue of log file errors which I have learned to live with.  I thought that was my only issue until last week. In one week I experienced two different power outages. The first since installing Genmon. 

The serial connection to my Pi is made using the OpenGenSet from https://pintsize.me/

During the first I was away on vacation, during the second I was at my residence. Since I live in a rural area, every time the power goes out so does my internet due to power loss to the cable amplifiers. I mention this because it relates to my question. During the second outage I accessed Genmon on my LAN to see the generator stats. To my surprise when I opened the web interface there was no data available. 

My question, is there some dependency of an internet connection for Genmon or OpenGenSet to function properly? When I view the interface during the normal weekly exercise I see the expected data populated. When I viewed during the last outage the only data displayed was the Pi CPU temp. 

Systems details below:

Monitor :
    Generator Monitor Stats :
        Monitor Health : OK
        Controller : Evolution 2.0, Air Cooled
        Run time : Generator Monitor running for 0:46:02.
        Power log file size : 0.06 MB of 15.00 MB
        Generator Monitor Version : V1.18.18
        Update Available : No

    Communication Stats :
        Packet Count : M: 42699, S: 42698
        CRC Errors : 0
        CRC Percent Errors : 0.00%
        Timeout Errors : 0
        Timeout Percent Errors : 0.00%
        Modbus Exceptions : 8
        Validation Errors : 0
        Sync Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 30.91
        Average Transaction Time : 0.0641 sec
        Modbus Transport : TCP

    Platform Stats :
        CPU Temperature : 132.80 F
        Pi Model : Raspberry Pi 4 Model B Rev 1.4
        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK
        CPU Utilization : 0.56%
        OS Name : Raspbian GNU/Linux
        OS Version : 11 (bullseye)
        System Uptime : 19 days, 3:00:44
        Network Interface Used : eth0
        System Time : Friday July 21, 2023 22:58:36

### Your Environment

- Generator Model: Generac G0070432 22kw
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.18

jgyates, 2023-07-22:
There is no dependance on internet access when trying to access the web interface. Genmon does send notices (SMS, email ,etc) out via the internet and if it can not send them it will queue the message until the internet is available. 

There may be an issue on the way you have your system setup or the way you are accessing genmon. When you tried to access the web interface during an outage on your local LAN, did you use an IP address or a domain name (i.e. http://192.168.1.10:8000 or http://genmon.local or some other name)?

If you are not using an IP address then your router may not resolve the name genmon if the router can not access your ISPs DNS server. That is a possibility that is dependent on your router.

Also, what specifically are the "experienced the documented issue of log file errors" that you mentioned?

jgyates, 2023-07-22:
one other question: When you wrote "when I opened the web interface there was no data available." Can you give more details regarding what not data available means? Does this mean that the web page is not available or the web page is available but the generator data is not  populated?

skipfire, 2023-07-22:
Since it apparently works when there is not a power outage, I've got some tests and questions to help rule things out.
1. Does it keep working when exercising?  If no, can you ping the OpenGenSet?
2. (Skip if #2 is no) Does it keep working if you do a start & transfer test from the maintenance page?  If no, can you ping the OpenGenSet?
3. If you cause an outage by flipping the breaker in the top of your ATS, does it work? If no, can you ping the OpenGenSet?
4. Did you try pinging OpenGenSet during the outage?
5. Do you have your network equipment on UPSes?
6. Do you have a single device for modem, router, and WiFi?  If not, how is your networking equipment split up?
7. How quickly (if you know) did it start working after the power outage ended?

el0552t, 2023-07-25:
Sorry for the delay in responding. I had time this evening to simulate an
outage. Genmon worked as expected without any problems. After switching to
the generator and checking Genmon I then unplugged the cable modem to
disconnect the internet and checked Genmon again. It still performed as
expected. I apologize for taking your time on this.

To answer your questions even though problems seem resolved my
configuration is as follows: Cable modem > Zyxel USG security appliance >
POE switch. All devices are powered from my UPS.
.
I access Genmon via the ip address @ port 8000

As for the abnormal notifications, I will forward to you a random day of
notifications in another email. It is a representation of the notifications
being sent for what appears to be no reason. At some point last month, so
many were being sent to my cell phone, my provider AT&T marked the sender
as spam. I had to call AT&T to have the block removed! LOL.

This is one I noticed from yesterday. I have not seen this before.

Time: 11:50:01 PM
Date: Sun 23-Jul-2023

NOTE: This message is a notice that the state of the generator has changed.
The system is not in alarm.

Status :

    Engine :
        Switch State : Auto
        Engine State : UNKNOWN: 3f3d0000
        Battery Voltage : 13.60 V
        RPM : 0
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Output Current : 0.00 A
        Output Power (Single Phase) : 0.00 kW
        Active Rotor Poles (Calculated) : 0

    Line :
        Utility Voltage : 239 V
        Utility Max Voltage : 244 V
        Utility Min Voltage : 231 V
        Utility Threshold Voltage : 156 V

    Last Log Entries :

        Logs :
            Alarm Log : 11/21/22 17:55:34 Charger Missing AC : Alarm Code:
0000
            Service Log :
            Run Log : 07/19/23 14:05:34 Stopped - Auto

    Time :
        Monitor Time : Sunday July 23, 2023 23:50:00
        Generator Time : Sunday July 23, 2023 23:49

No Alarms: 0001:3f3d0000

Outage :
    Status : Last outage occurred at 2023-07-24 20:06:30 and lasted 0:08:37.
    System In Outage : No
    Utility Voltage : 239 V
    Utility Voltage Minimum : 224 V
    Utility Voltage Maximum : 244 V
    Utility Threshold Voltage : 156 V
    Utility Pickup Voltage : 190 V
    Startup Delay : 5 s

    Outage Log :
        07-24-2023 20:06:30, Duration: 0:08:37, Estimated Fuel: 0.21 gal
        07-17-2023 20:17:22, Duration: 0:00:22, Estimated Fuel: 0.00 gal
        07-07-2023 21:41:52, Duration: 0:00:00, Estimated Fuel: 13.18 gal
        07-07-2023 21:41:51, Duration: 0:00:00, Estimated Fuel: 13.18 gal
        07-07-2023 21:41:49, Duration: 0:00:01, Estimated Fuel: 0.00 gal
        07-07-2023 21:41:48, Duration: 0:00:00, Estimated Fuel: 13.18 gal
        07-07-2023 21:41:46, Duration: 0:00:00, Estimated Fuel: 13.18 gal
        07-07-2023 21:41:45, Duration: 0:00:00, Estimated Fuel: 13.18 gal
        07-07-2023 21:41:43, Duration: 0:00:00, Estimated Fuel: 13.18 gal
        07-07-2023 21:41:42, Duration: 0:00:00, Estimated Fuel: 13.18 gal
        06-29-2023 17:47:51, Duration: 9:07:10, Estimated Fuel: 13.17 gal
        06-28-2023 15:00:27, Duration: 0:00:23, Estimated Fuel: 0.00 gal
        06-28-2023 07:32:37, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-28-2023 07:32:36, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-28-2023 07:32:35, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-28-2023 07:32:33, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-28-2023 07:32:32, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-28-2023 07:32:30, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-28-2023 07:32:29, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-28-2023 07:32:28, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-23-2023 09:12:57, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-23-2023 09:12:56, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-23-2023 09:12:55, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-23-2023 09:12:54, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-23-2023 09:12:52, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-23-2023 09:12:51, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-23-2023 09:12:49, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-23-2023 09:12:48, Duration: 0:00:00, Estimated Fuel: 0.02 gal
        06-20-2023 21:36:51, Duration: 0:01:58, Estimated Fuel: 0.00 gal
        06-20-2023 20:05:30, Duration: 0:00:21, Estimated Fuel: 0.00 gal
        05-12-2023 14:33:50, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 14:33:49, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 14:33:48, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 14:33:46, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 14:33:45, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 14:33:44, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 14:33:42, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 14:33:41, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 12:49:18, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 12:49:17, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 12:49:16, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 12:49:15, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 12:49:13, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 12:49:12, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 12:49:10, Duration: 0:00:00, Estimated Fuel: 0.00 gal
        05-12-2023 12:49:09, Duration: 0:00:00, Estimated Fuel: 0.00 gal

Thank you,
Tim Kaser

el0552t, 2023-07-25:
Notifications from

jgyates, 2023-07-25:
You can get rid of some (if not all) of the random emails by enabling the "Ignore Unknown Values" setting on the Advanced settings page. Access the Advanced Settings page by double clicking the gear icon in the top right of the web interface. With Evolution 2.0 controllers, the controller responds with some random data periodically. Genmon reports changes in the generator status so these random values can cause false notifications. Enabling the setting mentioned above will add additional checks that attempt to filter out the garbage data that the Evolution 2.0 controller is returning.

I am going to close this issue for now but fee free to post any follow up questions or data to this thread.

el0552t, 2023-07-25:
Will do. Thank you for the information.

Thanks again

closed 2023-07-31 by jgyates