jgyates/genmon #931: Question on GPIO possibilities

Schlats, 2023-08-03:
My pi which genmon is installed, gets blasted by the sun and even with a vented enclosure and heat sinks w/ small fan the space just heat soaks. I'd like to add a small enclosure fan to get some airflow through it.

Is it possible to use the GPIO to run a fan or relay when the pi cpu temp gets to a setpoint either directly on the genmon or via mqtt?
I looked over the gengpio and customgpio addons, seems possible, then again i haven't a clue what it all means.

My setup is home assistant with mqtt comms to genmon, so if direct on genmon is not possible, I could have HA send an mqtt command to turn on a gpio output based on cpu temp which i monitor. My other thought is to use a esp device to actuate fan via mqtt, I just don't know if it could handle the distance to the WAP.

Rock on,

jgyates, 2023-08-03:
Genmon does not currently support setting GPIO pins externally, however I could add an option to gengpio.py that would set a gpio pin if the pi CPU temp reach a certain threshold. If you like I can add this to the feature request list.

Schlats, 2023-08-03:
That would be awesome. Reading through gengpoi doc I think I understand it and seems doable for a novice like myself. I assume you would add the cpu temp to the list?

skipfire, 2023-08-03:
Another option is to add a python script and set it up as a service, that can be done without interfering with GenMon.  I have that pre-installed on the preloaded units I sell, but I've got the source publicly available.

Script to control the fan: https://github.com/skipfire/GenMonExtension/blob/main/PiSetup/PintSizeFanManager.py
Service file: https://github.com/skipfire/GenMonExtension/blob/main/PiSetup/PintSizeFanManager.service - you will have to update paths to account for any difference in usernames.

I don't have a dedicated setup file for just the fan, but you could use the fan setup section of mine, just adjust it for wherever you put the files.
Setup: https://github.com/skipfire/GenMonExtension/blob/main/PiSetup/Setup.sh

jgyates, 2023-08-03:
looks like @skipfire already has a solution. I took a few minutes and added this functionality to the gengpio.py add on.

the gengpio.conf file has the details on the settings:

https://github.com/jgyates/genmon/blob/master/conf/gengpio.conf

To use this you must to the following:

1) Update genmon to the latest version on the about page
2) log into your pi via ssh and copy the new conf file from /home/pi/genmon/conf to /etc/genmon

        cd genmon
        sudo cp ./conf/gengpio.conf /etc/genmon
3) Edit the file /etc/genmon/gengpio.conf to suit your need
4) restart genmon

In my testing I set ER_HIGH_TEMP = 33 to ER_HIGH_TEMP = 0 to disable the use of that GPIO by the addon, then I set ER_PITEMP to 33. Then I set the CPU_THRESHOLD_TEMP to the appropriate value (C or F) based on the Use Metric setting.

Let me know if you have any questions or issues.

I am going to close this issue for now but feel free to post any follow up questions or status you have for me or @skipfire to this thread.

closed 2023-08-03 by jgyates

Schlats, 2023-08-03:
@jgyates @skipfire  Thank you gentlemen. I'll play with it this weekend, appreciate the help!

Schlats, 2023-08-05:
@jgyates thanks for the mods, somehow it makes complete sense after reading/reviewing file as I do not know code/electronics. I felt very accomplished I was able to get this to work to be transparent. 
@skipfire I didn't think to mention in my original post, I do have your hat v2.1f. If I am using jgyates updates and reviewing the documentation for the board, it seems there is no easy access to GPIO 13 (pin 33) built into the board? To confirm what I assume, I must solder into that header pin (and connect/solder to ground)?

So appreciative of your building and help. Looking forward to the day I have 50% of your ability.

Schlats, 2023-08-05:
@skipfire I just noticed in the doc/inst for hat installs last item "Install the PintSizeFanService if you are using a fan on the HAT", I installed a fan but never used fanservice, the fan just runs all the time once power is applied to the hat.

skipfire, 2023-08-05:
@Schlats the fan control was not available on the HATs until v2.3, prior to that the fan pins are always on.

jgyates, 2023-08-05:
@Schlats, you may be doing this already, but using jumper wires is recommended rather than soldering to the pi: 

https://www.amazon.com/Elegoo-EL-CP-004-Multicolored-Breadboard-arduino/dp/B01EV70C78/ref=asc_df_B01EV70C78/?th=1

Schlats, 2023-08-05:
@jgyates @skipfire apologies gents, I think I did not communicate my request/setup well originally, due to my limited basic understanding of the software, pi, HAT and them working together. I believe I have a better understanding now, and believe I can still complete it. Can you confirm my understanding and implementation?

I have pi 3A+ with v2.1 HAT. HAT is setup for basic modbus use, barrel power with a fan that always runs. I am content with that portion, however I'd like to add a second fan to the enclosure to bring in cool air and expel hot air. I'd like fan2 to be controlled by gengpio based on pi cpu temp. The HAT has a 5 pin header to access ground & gpio 27, 22, 17 (as inputs as are used in gengpioin). These gpio are general purpose, so I could re-purpose one of them to power fan2. With jgyates’ update to gengpio now able to utilize picputemp, I can set ER_PITEMP = 11 (gpio17), because I am not using gengpioin, there should be no gpio conflict, but will set INPUT_STOP = 0 in gengpioin so there is no inadvertent issues possible in future.

I first thought I had to access gpio33 to utilize ER_PITEMP, which is why I asked about soldering, since there is no access to it without using a portsplus type board.

jgyates, 2023-08-05:
I can not tell based on the info you supplied but you probably can not power the fan from the GPIO pin as it does not supply that much current.

https://embeddedcomputing.com/technology/processing/compute-modules/raspberry-pi-os-fan-control#:~:text=The%20Raspberry%20Pi's%20GPIO%20pins,for%20lower%20speed%20and%20n

You could use a relay, controlled by the GPIO pin, or you could just run the fan continuously and not use a GPIO.

The GPIO pin assignment you describe above appears to be good to me.

Schlats, 2023-08-05:
Awesome, I'm learning thank you gentlemen!! I was mostly concerned with the gpio setup. The rest should be easier for me.

Schlats, 2023-08-11:
I received a relay to power fan and a 1 to 2 gpio extender to access gpio today for use in setting up the enclosure fan.

@jgyates , unfortunately I couldn't seem to get the update to work implementing as you described above. Got latest update. Set temp=0 so gpio could be used in cputemp=33, move conf file, turned on the gpio outputs add on. The pin 33 would never change state. I turned debug on, but was unable to figure out where the log file would be. Thank you for putting something together so quick. I have to imagine its something I'm doing wrong, just can't figure it out.

@skipfire I was able to get your files to work and accomplish the outcome I was looking for. Thank you for your assistance.

jgyates, 2023-08-11:
Thanks for the feedback. I am glad you got it working with @skipfire's solution.