jgyates/genmon #932: Start/Stop Generator from a MQTT server

salazaram1966, 2023-08-04:
Hello, good afternoon. First of all, I'd like to thank jgyates for this awesome application. I'm new here, so please forgive me if this question has been asked before. I've set up Genmon on a Raspberry Pi 4 along with my Generac generator and Evolution 2.0 Controller. I also have a local MQTT server using Mosquitto, which allows me to monitor all the variables.

However, I'm wondering if it's possible to control the generator, including the transfer option, using MQTT. I've configured the Add-On in Genmon and provided the IP address of my Mosquitto server, but I'm unable to figure out how to send MQTT messages to Genmon. Can you confirm if this is even possible? I would really appreciate any help you can provide. Thank you in advance for your responses

- Generator Model: GENERAC 22KW Controller Evolution 2.0
- Genmon Version: V1.18.16

jgyates, 2023-08-04:
Hi @salazaram1966 

Thanks for thee kind words.

This link:

 https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmqttpy-optional

has info in the MQTT add on (genmqtt.py) and there is a section on how to publish commands. The commands to start, stop and start with transfer are:

       "setremote=start"
       "setremote=stop"
       "setremote=starttransfer"

These commands will trigger actions that are internal to the controller so all genmon is doing is issing the command start, stop, starttransfer to the controller. This is noteworthy as the stop command listed above is only good for stopping one of the other two commands. The stop command will not stop a generator that has been activated by a power loss in Auto mode.

Let me know if you have any other questions.

salazaram1966, 2023-08-05:
First and more importantly thank you for your prompt response. It helps me
a lot. I will try first thing in the morning.

Alexander

jgyates, 2023-08-05:
I am going to close this issue, but feel free to post to this thread if you have any follow up questions or feedback.

closed 2023-08-05 by jgyates

salazaram1966, 2023-08-05:
Thank you, jgyates. The information you provided worked as expected. I can now turn my generator on and off via MQTT (Thank you so much for that!). As I mentioned before, I'm using Mosquitto, but the only topics I am receiving from the generator are as follows:

pi@raspberrypi:~ $ mosquitto_sub -h localhost -t generator/# -F %t:%p
generator/client_status:Online
generator/Status/Engine/Battery_Voltage/value:12.8
generator/Status/Engine/Battery_Voltage/value:12.7
generator/Status/Line/Utility_Voltage/value:220
generator/Outage/Utility_Voltage/value:220
generator/Status/Engine/Battery_Voltage/value:12.8
generator/Status/Line/Utility_Voltage/value:219
generator/Outage/Utility_Voltage/value:219
generator/Status/Line/Utility_Voltage/value:220
generator/Outage/Utility_Voltage/value:220 .........................

I would also like to access Maintenance information, for example, to view the same details I see on the website, such as Service A, Service B, Battery Check, etc., as well as the Monitor information. 

Do I need to send a command to Genmon to retrieve that information? If so, is there a guide available for the commands that can be sent to Genmon? Thanks again

jgyates, 2023-08-05:
Hi @salazaram1966 
No other command is needed so send data.

Look at the settings on the Add On page for the MQTT plug in. There are two settings that would effect this behavior. The first is the Blacklist Filter setting. If a value in this is contains in the topic of a specific MQTT value, it will be ignored an not sent, so make sure you don't have 'Maintenance' or possibly remove all of the values from this entry. The second setting is the Flush Interval settings. If this is zero then MQTT topics will only be sent if they change (they are sent once at the beginning, but only after changing after that). You can set the Flush Interval to a non zero positive integer and it will resend all values once the number of seconds specified has expired. 

If either of these two settings do not resolve you issue then you can send your logs via the about page. The log for the MQTT plug in is /var/log/genmqtt.log if you want to view the log for errors.

salazaram1966, 2023-08-05:
Thanks again, the issue was the Blacklist Filter setting. It is working now!. :)

jgyates, 2023-08-05:
excellent! I am glad it is working for you now.

salazaram1966, 2023-08-06:
I just sent you a donation from my other email salazars.alexander@gmail.com as a token of appreciation for your excellent job