jgyates/genmon #934: Serialtest is not working

notms, 2023-08-16:
Hello Genmon community. Been following for quite a while and have gone through all the testing forum pages that I can think off. I can’t even pass a test when I jump 8 and 10. Something is odd with my pi zero 2W unless I messed something up. Any help would greatly appreciate.

### Expected Behavior

Successful test

### Actual Behavior

failed test

### Steps to Reproduce (including precondition)

Ran Serialtest.py and get, Failed: Sent data does not match receive. Received 0 bytes.

### Your Environment

- Generator Model: 22kw
- Genmon Logs: Submit via the About page
- Genmon Version: latest 18

jgyates, 2023-08-16:
Hi @notms 

Did you run the installation script?

   ./genmonmaint.sh -i

Also, what is the output of this command

   cd OtherApps
    python3 serialconfig.py -c

The install script should run serialconfig during the install process. This program (serialconfig.py) will setup the serial port on the raspberry pi.

Also, photos of your cabling / loopback may help clarify any issues with the pinout.

skipfire, 2023-08-16:
I believe it will fail if you don't stop GenMon before running it, GenMon "owns" the serial port when it is running.

notms, 2023-08-16:
> Hi @notms
>
> Did you run the installation script?
>
> ./genmonmaint.sh -i
>
> Also, what is the output of this command
>
> cd OtherApps python3 serialconfig.py -c
>
> The install script should run serialconfig [...]

Thank you for your quick reply I got all OK on the serialconfig.py

[image]

notms, 2023-08-16:
> I believe it will fail if you don't stop GenMon before running it, GenMon "owns" the serial port when it is running.

I had stopped the Genmon service

skipfire, 2023-08-16:
If your clips are on the pins you are trying to short, you are going at the wrong pins.  It's physical numbers not logical numbers.  If you look at this image, it's the 4th and 5th pins on the right side https://www.etechnophiles.com/wp-content/uploads/2021/01/R-Pi-4-GPIO-Pinout-1.jpg?ezimgfmt=ng:webp/ngcb40.

notms, 2023-08-16:
Ohh man thank you, that totally makes sense I was wondering why I would be runnning a test on different pins than the once that I connect to RX and TX it passed now

notms, 2023-08-16:
Here is my next challenge , 0 bytes failed
[image]

skipfire, 2023-08-16:
Red and black I think are backwards.  The TXD has arrows making it look like it's TX from the GEN which needs to go to RX on the Pi.

notms, 2023-08-16:
Just switched same result, btw thank you for the quick responses :-)

skipfire, 2023-08-16:
Any jumpers or anything on the other side of the board?  Any possibility the board is bad?  About the only things that can be wrong is a bad board, or TX & RX backwards on the Pi side if the other side has RX and TX shorted.

notms, 2023-08-16:
No jumpers, anyways for me to test this board that you know of? I have another board that I can try as well

notms, 2023-08-17:
@skipfire it turns out to be a bad board glad I had ordered 3, now when I run the test both TX and RX are lighting up, thank you for the quick reaponses

closed 2023-08-17 by notms

skipfire, 2023-08-17:
@notms I'm glad you got it working, that type of difficulty is why I started selling the HATs.