jgyates/genmon #935: Missing connector on Powerpact 7.5Kw controller

alfredosc8, 2023-08-18:
Hi,
I need and advice from you.
I have Powerpact 7.5kw 2018 and looking the controller, as you describe in your instructions, I cannot see the accessory connector.
The cover has the sign to install the mobile link.
Do you think the connector is covered and I have to make the hole to expose it?
Or that connector is in different place?hec
Checking the same controller on eBay, the pictures show the connector as you describe.
So Can you give me  an advice, probably other users faces the same issue i'm facing.

Thanks and congratulation , you did great solution.
Alfredo

jgyates, 2023-08-18:
Genmon uses the same connector as mobile link. I have not found anyone who had to make a hole to expose the connector. It should be on the back of the controller.

Is your generator like the one mentioned in this thread: https://github.com/jgyates/genmon/issues/888

A picture of the generator and a picture of the sign referencing mobile link may help.

alfredosc8, 2023-08-18:
The model is the 6998 stand by generator at natural gas. tomorrow I will open it and I will give more details.

grsthegreat, 2023-08-20:
im almost positive the powerpact controller does not have an accessory port. i do know for a fact that the addon mobile link wifi accessory will not work on the powerpact (7.5 kw) generator.

jgyates, 2023-08-20:
I don't know if it does now, but i certainly used to support it because genmon has been tested on a PowerPact 7.5w. Also I have two people who have submitted logs from a PowerPact 7.5w. There is a modbus register that is used to identify the Evolution variant and this register has a unique value for PowerPact.

grsthegreat, 2023-08-20:
Very odd, i dont recall ever seeing an accessory port on a powerpack controller. But then again i hardly ever see that small of unit. I service about 300 generators, and ill bet 95% are 16 kw and larger. The port may be covered with a tape seal. When i look at the accessory lte device we sale for mobillink, it specifically says it wont work with powerpack. These devices plug into accessory port also.

mhornet27, 2023-08-23:
Its 8 pin molex and under the controller.  You may have to peel a sticker off it.  I have a PowerPact and just set up my monitor.  The only thing the system doesn't monitor is the power output and voltage output when the generator is on from a utility outage.  just had a 2 hr + outage last night.

liltux, 2023-08-23:
[image]

alfredosc8, 2023-08-23:
thanks youSent from my iPhone

closed 2023-08-23 by jgyates